Koenkk/zigbee2mqtt #282: Clarification: firmware change can leave all devices out?

subzero79, 2018-08-13:
Just yesterday i attempted to flash the sniffer firmware into the dongle to some traffic capture, I don't if the firmware wasn't the correct one or i was just using the wrong software so it didn't work properly. Anyway after I gave up, i flashed the coordinator firmware, latest by pulling the repo so it wasn't the same original i was using with before i flashed the sniffer one.

Once i started zigbee2mqtt again, all devices where not recognised any more. I had to repair them again one by one. Tomorrow i have to do the plugs in the attic.

Is this expected behaviour? Whats the issue here? transport keys? should it be warned in the wiki maybe?

For know i learn, and I've already ordered another dongle for this kind of testing.

cmille34, 2018-08-13:
I learned this the hard way as well. I reflashed my CC2531 to fix an "unknown 17" error. I was told if you reflash the usb dongle, you'll have to repair the devices.

Koenkk, 2018-08-13:
Reflashing = clearing memory of CC2531 -> repair all devices.

subzero79, 2018-08-13:
Thanks for the info. Is it possible to backup the current loaded memory with the cc-debugger? 
Also what does it get lost by flashing? The transport keys ?

tb-killa, 2018-08-14:
@Koenkk  @ptvoinfo
I don´t know if this could help but this seems to be a good way:
https://sunmaysky.blogspot.com/2016/01/how-to-clone-existing-z-stack.html?m=1

Another Way (maybe..??)

Maybe we have to prebuild our firmware with options 
``NV_INIT
NV_RESTORE
`` 
to get them also working ?
But attention: http://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/t/685544?CC2531-NV-RESTORE-and-Flash-Life

Koenkk, 2018-08-14:
@tb-killa that looks good! We could use cc-znp library to read and write.

matchett808, 2018-08-20:
Is that now on the new firmware?

tb-killa, 2018-08-20:
Actually no work. We are all busy, if i got some free time next week i could take a Look at this.

lolorc, 2018-10-17:
the same way, I've replaced my cc2531 by a cc2530+cc2590 without repairing all devices. Just need to unplugged routers and shield the coordinator so that it doesn't detect the chan/panid is already used.

closed 2018-12-23 by stale[bot]