Koenkk/zigbee2mqtt #284: Raspbee Module

L00000813, 2018-08-14:
I want the capability of zigbee2mqtt without the need for an external dongle attached to my Raspberry Pi.

How can I accomplish this?

Would the following device solve my problem https://www.dresden-elektronik.de/raspbee/  ?

Any other suggestions would be appreciated.

Thanks..

tb-killa, 2018-08-14:
Actually only ti modules are compatible.

You could build your own via cc2530 with serial Header and use rpi onboard serial.
Build your own simple pcb and you got the same Module size ;)

L00000813, 2018-08-15:
Thanks for the reply. Maybe you could provide more clarity on how exactly I can do this. Provide links to products etc..

tb-killa, 2018-08-15:
If i got time (next two weeks im really busy) i tell more because working on this too.

L00000813, 2018-08-15:
Ok, no worries.

I will wait to see how you implement this. If it is a success I will try after.

Thanks.

k1ngf15h3r, 2018-11-13:
I am also intressted to get support for the raspbee module. I already have the module installed on my Raspberry so it would be nice to keep it running like it is. But I Need to upgrade my system with zigbee2mqtt so really looking forward to get it combined

mailuminatti, 2019-01-17:
Hey @tb-killa by any chance was this moved forward? Quite interested in see this happening.

sorryusernameisalreadytaken, 2019-02-03:
> 
> 
> If i got time (next two weeks im really busy) i tell more because working on this too.

*BUMP*
A raspbee support would be really nice.

nukleuz80, 2019-02-05:
I‘m interested too in RaspBee ...

sehraf, 2019-02-05:
https://koenkk.github.io/zigbee2mqtt/information/FAQ.html
RaspBee won't happen.

Koenkk, 2019-04-22:
RaspBee indeed won't happen.

closed 2019-04-22 by Koenkk