Koenkk/zigbee2mqtt #285: clean install: devices.js not latest version

kimonm, 2018-08-14:
On clean install, devices.js is an older version than on zigbee-sheperd-converters github site. Is there a proper way to update (or is there an error in my installation)?

Also, latest npm version (for my system) is 6.3.0. Does zigbee2mqtt work with the new version, or should I stay on 5.6.x?

Thanks

Koenkk, 2018-08-14:
Are you on the master or dev branch?

kimonm, 2018-08-14:
I followed the instructions on the wiki, so I assume I'm on master branch:
sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt

In the unlikely event that it's relevant, the supported devices wiki pages hasn't been updated in 18 days.

Koenkk, 2018-08-14:
The supported devices pages will be updated once dev is merged back into master, I try to do this every +- 2 weeks (but currently due to moving houses I have less time).

ciotlosm, 2018-09-03:
@kimonm use the version mentioned on the site for npm and node. Other versions are known to cause issues.

You can get involved and help with testing new versions if you have the knowledge to fix things as well and help the community.

closed 2018-09-12 by kimonm