Koenkk/zigbee2mqtt #286: Device with modelID ‘� �P’

idenny1, 2018-08-14:
Hi, I am trying to add ecosmart bulb to zigbee2mqtt and receive “Device with modelID ‘� �P’ is not supported” is it still possible to still add it to code to node_modules/zigbee-shepherd-converters/devices.js ? Even with non displayable chars in the modelID ‘� �P’ ?

Koenkk, 2018-08-15:
Can you try to?

idenny1, 2018-08-16:
Will try next week when I have some time thanks.

tb-killa, 2018-08-20:
We have to check if there is some fault on utf or ansi Code converting or something else....

tb-killa, 2018-08-30:
@idenny1 Could you please provide your ``database.db`` entry(s) for this device

tb-killa, 2018-08-31:
You could use ``cat`` and copy and paste the marked text.

idenny1, 2018-09-01:
{"id":3,"type":"Router","ieeeAddr":"0xdd16abfff1a19365","nwkAddr":2513,"manufId":4445,"manufName":"2.1.0_16d_YF05","powerSource":"Unknown","modelId":"\u0000\u0002\u0000\u0004� �P\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u000e","epList":[1],"status":"online","joinTime":1535742570,"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"2.1.0_16d_YF05","modelId":"\u0000\u0002\u0000\u0004� �P\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u000e","dateCode":"","powerSource":0,"locationDesc":" Light","physicalEnv":0,"deviceEnabled":1}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":128}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":128,"lastCfgBy":"0xffffffffffffffff"}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":1}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":255,"remainingTime":0,"onOffTransitionTime":3,"onLevel":255,"onTransitionTime":0,"offTransitionTime":0,"defaultMoveRate":255}},"genOta":{"dir":{"value":2},"attrs":{}}}}},"_id":"SUX7lcjTuwggtjKv"}

tb-killa, 2018-09-01:
@idenny1 Would you please provide one full run with ``DEBUG=* npm start`` and public via pastebin or another service.
I´v got some idea where i think this (maybe) fault generate.

idenny1, 2018-09-02:
I using zigbee2mqtt in hass.io for the debug information you require "DEBUG=* npm start". Do I have more than update the configuration.yaml with "advanced:    log_level: debug:?

tb-killa, 2018-09-02:
Sadly we need the debug messages from zigbee shepherd themselve ...
Will wait for debug messages from another issue .. maybe with this informations we could find the problem too ... wil post if something new shows up ..
thanks :)

idenny1, 2018-09-02:
I will look at setting up zigbee2mqtt environment without hass.io and get the log debug information as time permits. Thanks

idenny1, 2018-09-02:
[debug1.log.tar.gz](https://github.com/Koenkk/zigbee2mqtt/files/2343820/debug1.log.tar.gz)
[debug2.log.tar.gz](https://github.com/Koenkk/zigbee2mqtt/files/2343821/debug2.log.tar.gz)

debug 1 - is the debug output from modelID not supported
debug 2 - is the debug with my modified devices.js

tb-killa, 2018-09-03:
@idenny1 Thanks for the logfiles. please format them before post because they include all type of terminal code .. if i manually remove them i broke the ansicode output from what i need ..

lets do something other .. start again via debug=* npm start and let them run ..
i think the needing part is this:

``
[36;1mzigbee-shepherd [0mIdentified Device: { manufacturer: 2.1.0_16d_YF05, product:   � �P                        } 
``
please search for this and copy only this part to something like pastebin or other service ..
we need to know if the ansicode transport correct formated by cc-znp or by saving to object database.

closed 2018-11-09 by stale[bot]

dcshoes23, 2018-11-09:
I am having the same issue with my EcoSmart bulb, I tried adding the model to devices.js but it seems to keep changing.

boojew, 2018-12-18:
I have this bulb and issue as well. If you need any info, please just let me know