Koenkk/zigbee2mqtt #288: Syslog logging from Zigbee2mqtt

SargonofAssyria, 2018-08-16:
Zigbee2mqtt is working fine and logging to a file, BUT it is also logging to syslog a lot.
How can I prevent logging to syslog (rule?)
Now it has a lot of lines like:

Aug 16 11:06:33 raspberrypitest npm[27156]: 2018-8-16 11:06:33 #033[32mINFO#033[39m MQTT publish, topic: 'zigbee2mqtt/R...

Koenkk, 2018-08-16:
logging can be reduced using the `log_level` attribute: https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration

SargonofAssyria, 2018-08-16:
I found a solotion in creating a file 'zigbee.conf' and put that in the directory '/etc/rsyslog.d' with the content:

if $programname == 'npm' then /dev/null
& stop

That did the job. A loglevel change has no use for double logging in 2 files. I want to have my syslog file clean from a lot of zigbee messages, which are already logged in the zigbee2mqtt log.

closed 2018-08-16 by SargonofAssyria

JsBergbau, 2021-03-27:
There is a much cleaner solution. In  /etc/systemd/system/zigbee2mqtt.service

use 
```
StandardOutput=null
```

instead of 
```
StandardOutput=inherit
```

I've created a pull request. Hopefully this is soon in the official docs.