Koenkk/zigbee2mqtt #294: Question: Xiaomi Aqara Smart Curtain Motor

sapkra, 2018-08-19:
I've just one small question:
Has someone already tried to use the Xioami Aqara Smart Curtain Motor with zigbee2mqtt and can maybe share the experiences?

tb-killa, 2018-08-19:
If you doesnt found this Information on the device supported Page we havent implement this.
If you in use of this device please follow the Wiki Page ;)

holeandro, 2018-08-20:
This is how far i got:
in the node_modules/zigbee-shepherd-converters/devices.js add these lines

{ 
    zigbeeModel: ['lumi.curtain'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZNCLDJ11LM', // Vendor model number, look on the device for a model number
    vendor: 'Aqara', // Vendor of the device (only used for wiki and startup logging)
    description: 'Aqara Curtain Motor ', // Description of the device, copy from vendor site. (only used for wiki and $
    supports: 'Curtain Motor', // Actions this device supports (only used for wiki)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
   }, 

But now I am struggeling to convert these types:

2018-8-20 22:20:21 DEBUG Recieved zigbee message with data {"cid":"genAnalogOutput","data":{"61440":0,"presentValue":0}}
2018-8-20 22:20:21 WARN No converter available for 'ZNCLDJ11LM' with cid 'genAnalogOutput' and type 'attReport'
2018-8-20 22:20:21 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-8-20 22:20:21 DEBUG Recieved zigbee message with data {"cid":"genAnalogOutput","data":{"61440":0,"presentValue":0}}
2018-8-20 22:20:21 WARN No converter available for 'ZNCLDJ11LM' with cid 'genAnalogOutput' and type 'devChange'
2018-8-20 22:20:21 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
2018-8-20 22:20:48 DEBUG Check online 0x000b57fffe4ab8de

holeandro, 2018-08-20:
Found someone who looked closer on this:
https://github.com/TedTolboom/com.xiaomi-mi-zigbee/blob/master/drivers/curtain/device.js

Koenkk, 2018-08-21:
Could you post the output when running with [`log_level`: `debug`](https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration) (please post on https://pastebin.com)

marcgarciamarti, 2018-08-21:
I didn't know Xiaomi had this product. I was sort of monitoring a kickstarter project but if this one runs over zigbee, I'm sold! :)

holeandro, 2018-08-21:
Thank you:
https://pastebin.com/bsdXYPpd

Koenkk, 2018-08-24:
@holeandro do you know what the `presentValue` means, could this be how much the curtains are closed/open?

```
  zigbee2mqtt:warn 2018-8-21 23:39:22 No converter available for 'ZNCLDJ11LM' with cid 'genAnalogOutput', type 'devChange' and data '{"cid":"genAnalogOutput","data":{"61440":0,"presentValue":100}}'
```

holeandro, 2018-08-25:
@Koenkk  thank you to look over it, yes you are right presentValue":100 says the percentage of the curtain. Today I wantet to test if 100 is closed or opened and if the "61440":0 :0 changes as well. But sadly my CC2531 USB sniffer disconnects everythim when i fireup zigbee2mqtt

marcgarciamarti, 2018-08-25:
@holeandro, can you confirm the usb device the dongled is maped to, matches what you have configured?

holeandro, 2018-08-25:
@Koenkk 100 means closed, 0 means opened

@marcgarciamarti thank you for your input. After reboot, trying other ports e.t.c. LED turned off after around 2 Min. and stopped working. After reflashing the firmware the stick is now running for more 20Min. and i was able to get the log. Thanks.

`Recieved zigbee message with data {"cid":"genAnalogOutput","data":{"61440":0,"presentValue":0}}
Recieved zigbee message with data {"cid":"genAnalogOutput","data":{"61440":0,"presentValue":0}}
Recieved zigbee message with data {"cid":"genAnalogOutput","data":{"61440":52272896,"presentValue":44}}
Recieved zigbee message with data {"cid":"genAnalogOutput","data":{"presentValue":44}}
Recieved zigbee message with data {"cid":"genAnalogOutput","data":{"61440":52272896,"presentValue":78}}
Recieved zigbee message with data {"cid":"genAnalogOutput","data":{"61440":52272896,"presentValue":78}}`

Koenkk, 2018-08-30:
@holeandro looks great, could you try following: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices ?

tb-killa, 2018-08-30:
Could you please provide database.db entrys.
Does the presentvalue reportated constant if Closed or opened Action run.
We need open, close and stop converter. We could show what we could Integrate.

tb-killa, 2018-08-31:
@holeandro  Please test this:
``devices.js`` entry:

```
{
zigbeeModel: ['lumi.curtain'],
model: 'ZNCLDJ11LM',
vendor: 'Aqara',
description: 'Aqara Curtain Motor ',
supports: 'Curtain Motor',
fromZigbee: [fz.ignore_genAnalogOutput_change,fz.curtain_PositionLiftPercentage],
toZigbee: [],
},
```

``fromZigbee.js`` Entrys:

```
curtain_PositionLiftPercentage: {
        cid: 'genAnalogInput',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            return {currentPositionLiftPercentage: precisionRound(msg.data.data['presentValue'], 2)};
        },
    },
```

```
ignore_genAnalogOutput_change: {
        cid: 'genAnalogOutput',
        type: 'devChange',
        convert: (model, msg, publish, options) => null,
},
```

If we see your ``database.db`` entrys we could check if we could use the ``genonOff`` cluster to turn on / off (like do stop).

Please report back :)

holeandro, 2018-09-02:
@tb-killa 
Hi Tb-killa, i hope you can work with that:
https://pastebin.com/ZNe29pxQ

Cheers

dzungpv, 2018-09-07:
I have this device too and fully connected it to zigbee2mqtt, can get 'presentValue'. But this not auto setup in hass, how to set a config for this so i can control the curtain?
Here the database file content: https://pastebin.com/AgtkrjUS

dzungpv, 2018-09-07:
I read the guide here: https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure
and try to send many command and payload but could not control the curtain.
I find someone do in smartthing. 
https://community.smartthings.com/t/xiaomi-curtain-dth-partially-openable/126972.

I also use Ubiqua and capture form the original Xiaomi hub.

For the open/close/pause/slide they use write "multi state ouput"/"analog ouput".

The detail sniffer information here, https://pastebin.com/LPiqHzH3  I censored the MAC.

dzungpv, 2018-09-08:
@Koenkk I have make it work like open, close, stop. But for the partly open (example 50%) it not work.
A guy on Smarthing forum could make it work by **write** command, but i am not know much about javascript and zigbee-shepherd. 

They do like this with smarthing:  https://raw.githubusercontent.com/shin4299/XiaomiSJ/master/devicetypes/shinjjang/xiaomi-curtain-sj.src/xiaomi-curtain-sj.groovy

```
        log.debug "Set Level: ${level}%"
        String hex = Integer.toHexString(Float.floatToIntBits(level)).toUpperCase()
        zigbee.writeAttribute(0x000d, 0x0055, 0x39, hex)
```

0x000d is an Analog Output cluster id. Attribute id 0x0055 is present value of that cluster. 0x39 is datatype, it is float point, single decision, the hex is the value of the percent open.

I see the api **zcl-packet write**:

```
        write { attrId, dataType, attrData }
```
How to produce command like smarthing api in toZigbee?

I also try:

```
	curtain_dim_test: {
        key: 'dim',
        attr: ['presentValue'],
        convert: (value, message) => {
            return {
                cid: 'genAnalogOutput',
				cmd: '0x02',
                zclData: {
					presentValue: value,
				},
            };
        },
    },
```

But for genAnalogOutput i am not see any command so it could write some value to the curtain but not correctly and it does not work.

dzungpv, 2018-09-10:
@kirovilya Can you help me about this https://github.com/Koenkk/zigbee2mqtt/issues/294#issuecomment-419636303

dzungpv, 2018-09-10:
@Koenkk How to use https://github.com/zigbeer/zigbee-shepherd/wiki#API_write in **toZigbee.js**?

dzungpv, 2018-09-10:
Try this code but not work:
```
		configure: (ieeeAddr, shepherd, coordinator, callback) => {
					const device = shepherd.find(ieeeAddr, 1);
					const actions = [
						(cb) => device.write('genAnalogOutput', 'presentValue', value.toString(16), function (err, data) {
								if (!err)
										console.log(data);    // 'value'
								}),
					];

					execute(device, actions, callback);
				},
```

kirovilya, 2018-09-10:
@dzungpv I do not undestand what can i do? :)

dzungpv, 2018-09-10:
@kirovilya It is hacking, so not easy. I am the first one make it work with Zigbee2mqtt. But i learn much from homey addon and smarthing. If can set partly open, it will perfect.

tb-killa, 2018-09-17:
@dzungpv Shouldn´t we use the official zigbee cluster for get them integrate ?
As i see we actually use the ``genAnalogInput`` but we also got the right ``closuresWindowCovering`` cluster.
From my point of view we should ignore ``genAnalogInput`` and use only ``closuresWindowCovering`` cluster. 
Maybe we need a bind configure part to get them work and reporting.?

What do other people think about this?

dzungpv, 2018-09-20:
@tb-killa Yes i use it like this:
```
    curtain_windowcoverings: {
        key: 'state',
        attr: ['windowcoverings_state'],
        convert: (value, message) => {// value: 'upOpen','stop','downClose'
            return {
                cid: 'closuresWindowCovering',
                cmd: value,
                zclData: {},
            };
        },
    },
``` 

But original hub they don't use standard closuresWindowCovering for partly open, they use genAnalogOutput write. What we can do when manufacturer don't follow the standard? We must do like they do. I am waiting the track for final test the code before submit PR. 

Many features not work like original app like set positive/reverse, clear the stroke save in memory, disable/enable manual control, all they use write to genBasic.

Now i can write some device like this https://github.com/Koenkk/zigbee-shepherd-converters/pull/74/commits/107fbe21b2c7ed3fa4a48822559da228722c039f#diff-3e0c275bd41af699153bf59d8dda4a2aR117 but not for this curtain, problem with the datatype: 0x42.
I only decode the raw hex data, but zigbee2mqtt now not support use raw data with write command, must use hex2string convert and when it convert to data but not keep the same format like "0x0000010100080007 " become 10100080007 so it not work.

So we must add function to write raw data to any device like smarting do.

Anyone can help to solve this problem? @kirovilya @Koenkk

holeandro, 2018-11-04:
@dzungpv If i remebre corretly, you were able to open and close the curtain, what was the magic, but you are waiting for raw functionality to configure the curtain as with the original APP?

poldim, 2018-11-05:
Does the open close function work as a ‘go to XX %’ command or is it really
just open and close with no mid controls?

greynoO, 2018-12-14:
any updates on this?

poldim, 2018-12-15:
I've given up and am using the aqara hub as I don't have the programing skills to make any additions.

Evildime, 2018-12-21:
@dzungpv Thanks to your PR I managed to get the curtain motor working. I can use "open, "close", "stop" and set the track percentage.

I had to make some changes to make it work in the latest version. I created PR with the curtain motor support based on your PR. Ideally you could merge my changes into your PR and I'll close mine.

See: https://github.com/Koenkk/zigbee-shepherd-converters/pull/170

holeandro, 2018-12-24:
@Evildime @dzungpv 
Thank you very much for your work.
A kind of a chrismas present for me.

Koenkk, 2018-12-27:
The device is supported now in the dev branch.

For Home Assistant users: this device also supports position, something which is not yet discovered by zigbee2mqtt (as I don't have the device I cannot test this). 

Could somebody make a PR for this? [This](https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L200) has to be extended based on [this](https://www.home-assistant.io/components/cover.mqtt/).

Evildime, 2018-12-28:
@Koenkk I spend the last two hours trying to integrate the position in Home Assistent but was not able to figure out how to exactly modify the homeassistant.js.

The Curtain is discovered in Home Assistant without issues and I can open, close and stop it's movement without any configuration. Also the position is correctly reported.

I tried adding the lines 
`            
position_topic: true,
set_position_topic: true,
set_position_template: '{{ value_json.position }}'
`
within the cover configuration but that does not seem to work. As I'm very unfamiliar with Home Assistant I'm happy to try suggestions but will stick to my FHEM installation anyway.

Koenkk, 2018-12-28:
@evildime no problem, somebody else in this thread probably has more experience. thanks again!

sapkra, 2019-02-04:
FYI: Xiaomi will release a new Curtain Motor with a battery.
https://de.aliexpress.com/af/aqara-B1.html?SearchText=aqara+B1

greynoO, 2019-02-24:
the cover.set_cover_position service in Home Assistant does not seem to work with this?

Koenkk, 2019-02-24:
Can you check with the latest dev branch? Note that after updating and starting zigbee2mqtt you need to restart home assistant. The position slider should also show up in home assistant now.

greynoO, 2019-02-24:
I would love to but I get this for the edge addon:

```
  zigbee2mqtt:info 2/24/2019, 10:47:45 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 2/24/2019, 10:48:45 PM Starting zigbee-shepherd
  zigbee2mqtt:error 2/24/2019, 10:48:55 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2/24/2019, 10:48:55 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2/24/2019, 10:48:55 PM Failed to start
	{"message":"Timed out after 10000 ms","stack":"Error: Timed out after 10000 ms\n    at Timeout._onTimeout (/app/node_modules/q/q.js:1846:21)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)","code":"ETIMEDOUT"}
  zigbee2mqtt:error 2/24/2019, 10:48:55 PM Exiting...
npm
 
ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.1.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-02-24T22_48_55_846Z-debug.log
2019-02-24T22:49:00: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-02-24T22:49:00: PM2 log: App [npm:0] starting in -fork mode-
2019-02-24T22:49:00: PM2 log: App [npm:0] online
> zigbee2mqtt@1.1.1 start /app
> node index.js
```

Same configuration as in the stable addon which is working fine..

greynoO, 2019-02-24:
I would love to but I get this error when using the edge-version addon:

```
  zigbee2mqtt:info 2/24/2019, 11:03:30 PM MQTT publish: topic 'homeassistant/sensor/0x00158d0002752f2d/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d0002752f2d","json_attributes_topic":"zigbee2mqtt/0x00158d0002752f2d","name":"0x00158d0002752f2d_linkquality","unique_id":"0x00158d0002752f2d_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002752f2d","name":"0x00158d0002752f2d","sw_version":"Zigbee2mqtt 1.1.1","model":"Aqara door & window contact sensor (MCCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
/app/lib/extension/homeassistant.js:593
            const topic = `${config.type}/${entityID}/${config.object_id}/config`;
                                    ^
TypeError: Cannot read property 'type' of undefined
    at mapping.(anonymous function).forEach (/app/lib/extension/homeassistant.js:593:37)
    at Array.forEach (<anonymous>)
    at HomeAssistant.discover (/app/lib/extension/homeassistant.js:592:36)
    at zigbee.getAllClients.forEach (/app/lib/extension/homeassistant.js:573:22)
    at Array.forEach (<anonymous>)
    at HomeAssistant.onMQTTConnected (/app/lib/extension/homeassistant.js:570:37)
    at extensions.filter.forEach (/app/lib/controller.js:83:75)
    at Array.forEach (<anonymous>)
    at Controller.onMQTTConnected (/app/lib/controller.js:83:58)
    at MqttClient.client.on (/app/lib/mqtt.js:44:13)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.1.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.1.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

Koenkk, 2019-02-25:
@greynoO sorry, made a typo, should be fixed now.

greynoO, 2019-02-25:
> @greynoO sorry, made a typo, should be fixed now.

still the same error `TypeError: Cannot read property 'type' of undefined` unfortunately!

Koenkk, 2019-02-25:
Are you sure you are running the latest -edge?

greynoO, 2019-02-28:
It is working now, after I installed it again (maybe you did some changes in the meantime? Didn't check it!)

I noticed that the arrows are disabled now, if the cover is opened / closed. But this is not reliable (at least for me), so whenever a curtain is closed, sometimes the arrow is disabled, sometimes it's not. I even had the situation that a curtain is closed and I cannot reopen it with the arrow because it's disabled.

BUT I noticed that everything is fine if I open/close it with the position slider. So my suggestion would be to use the set position functionality for fully opening (100) and closing (0) the cover?

Koenkk, 2019-03-01:
Using the position slider to open/close is not something that can be influenced by zigbee2mqtt. Could you find out how to reproduce it?

greynoO, 2019-03-13:
I mean you could map the open / close to directly set the position in the code instead of using the other code snippet that uses the "real" open/close operation? Now that this functionality is ported to the main branch I have this problem constantly :(

Koenkk, 2019-03-13:
I could map the arrow button to set the position instead of close/open, however when these are disabled you cannot use them?

Could you provide the logging when you run in a problematic situation?

greynoO, 2019-03-18:
No, not map position to the arrows, only position(0) and position(100) as this is the only way to reliably get the correct state as it seems!

Logging from the Hass.io addon?

Koenkk, 2019-03-19:
Can you provide the complete zigbee2mqtt logging when setting the position and having the deactivated arrow. (with `log_level: debug`, http://www.zigbee2mqtt.io/configuration/configuration.html)

greynoO, 2019-03-19:
Just using the down arrow (where the slider is not updated properly and arrow status is not toggled correctly) gives me this log:

```
zigbee2mqtt:debug 3/19/2019, 7:49:42 PM Received MQTT message on 'zigbee2mqtt/0x00158d0002b09378/set' with data 'CLOSE'
  zigbee2mqtt:info 3/19/2019, 7:49:42 PM Zigbee publish to device '0x00158d0002b09378', closuresWindowCovering - downClose - {} - {"manufSpec":0,"disDefaultRsp":0} - null
  zigbee2mqtt:debug 3/19/2019, 7:49:43 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"1028":2}}' of device 'lumi.curtain' (0x00158d0002b09378)
  zigbee2mqtt:debug 3/19/2019, 7:49:43 PM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"1028":2}}' of device 'lumi.curtain' (0x00158d0002b09378)
  zigbee2mqtt:debug 3/19/2019, 7:49:49 PM Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"1028":4}}' of device 'lumi.curtain' (0x00158d0002b09378)
  zigbee2mqtt:debug 3/19/2019, 7:49:49 PM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"1028":4}}' of device 'lumi.curtain' (0x00158d0002b09378)
  zigbee2mqtt:debug 3/19/2019, 7:49:49 PM Received zigbee message of type 'attReport' with data '{"cid":"closuresWindowCovering","data":{"currentPositionLiftPercentage":14}}' of device 'lumi.curtain' (0x00158d0002b09378)
  zigbee2mqtt:debug 3/19/2019, 7:49:49 PM Received zigbee message of type 'devChange' with data '{"cid":"closuresWindowCovering","data":{"currentPositionLiftPercentage":14}}' of device 'lumi.curtain' (0x00158d0002b09378)
```

Now closing with the position slider (other curtain, setting it to 0), where the arrows get updated correctly:

```
zigbee2mqtt:debug 3/19/2019, 7:55:27 PM Received zigbee message of type 'attReport' with data '{"cid":"genAnalogOutput","data":{"61440":0,"presentValue":0}}' of device 'lumi.curtain' (0x00158d0002b0b41b)
  zigbee2mqtt:info 3/19/2019, 7:55:27 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002b0b41b', payload '{"state":"CLOSE","position":0,"running":false,"linkquality":68}'
  zigbee2mqtt:debug 3/19/2019, 7:55:27 PM Received zigbee message of type 'devChange' with data '{"cid":"genAnalogOutput","data":{"61440":0}}' of device 'lumi.curtain' (0x00158d0002b0b41b)
  zigbee2mqtt:info 3/19/2019, 7:55:27 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002b0b41b', payload '{"state":"CLOSE","position":null,"running":false,"linkquality":68}'
```

Koenkk, 2019-03-21:
@greynoO thanks for the detailed logging! the position converter was missing, added that now.

Can you check with the latest edge? (wait 3 hours before trying this)

greynoO, 2019-03-21:
Thank you very much, it's working fine now! Although I would still suggest to use position(0) and position(100) for downClose and openUp. As at least for me  the motors do not always report 0 or 100 back initially (I have four motors). Eventually they do, if I click on the arrow again for 2-3 times.

Koenkk, 2019-03-22:
@greynoO done! please check latest dev branch (or edge in +- 2 hours).

closed 2019-04-20 by sapkra

Koenkk, 2019-06-27:
Could somebody with the original gateway sniff the packets when setting the track direction? Your help is greatly appreciated. (See https://github.com/Koenkk/zigbee2mqtt/issues/1639#issuecomment-502443362).

ruslanruslanruslan, 2020-02-07:
can someone help me please to integrate ZNCLDJ11LM in HA? I'm using ZNCLDJ11LM cover.curtain_158d0004117c61 in lovelace, I see a state of curtain but nothing happens when i click on arrow 
HA log:
ERROR (SyncWorker_4) [xiaomi_gateway] Got error element in data {"error":"Invalid key"}