Koenkk/zigbee2mqtt #295: Xiaomi Zigbee (vibration, tilt, fall) sensor Model: DJT11LM

ben423423n32j14e, 2018-08-20:
New Xiaomi Zigbee sensor just released (not yet supported), I've ordered a couple.

https://www.gearbest.com/robot-vacuum-accessories/pp_009899185513.html?wid=1433363

I should be able to include support, created ticket in-case anyone beats me to it. ;)

Very exciting and unique sensor with many applications.

subzero79, 2018-08-20:
Interesting....could use for the cloth dryer machine notification. The xiaomi plug cannot handle the amp load, and i am not keen in using a sonoff POW also.

ben423423n32j14e, 2018-08-20:
I'm planning to attach one of mine to the cat's automatic kitty litter so I can track their kitty litter usage. 🐈

So many possibilities!

jboeddeker, 2018-08-31:
i just got mine. 

the model is: "lumi.vibration.aq1"

SargonofAssyria, 2018-08-31:
Nice, but should not there be any return in the converter?

jboeddeker, 2018-08-31:
This is just a start, these converter do nothing yet. I do not own a Xiaomi Hub to analyze the messages or the functions of this sensor.

tb-killa, 2018-08-31:
Please provide your ``database.db`` entry(s).
Please provide full log with DEBUG without your own defined ``fromZigbee`` converters.

ben423423n32j14e, 2018-09-01:
This sensor should also have an input (ability to set the sensitivity). Question is, is it set on the Sensor via Zigbee (most likely) or virtually on the Xiaomi hub.

Look at the description on Gearbest, it shows a picture of it being set.

https://www.gearbest.com/robot-vacuum-accessories/pp_009899185513.html

artcrime, 2018-09-01:
I got one too, but I cant pair ist with my zigbee stick. Anyone else here got a pairing?

This is the message I got out of the log:
zigbee2mqtt:error 2018-9-1 16:30:43 Cannot get the Node Descriptor of the Device: 0x00158d0002abb211 (Error: Timed out after 10000 ms)

SargonofAssyria, 2018-09-01:
I managed to get it paired. (See log above) After the 5 sec push on the button and the flashing of the blue led. I had to repeatedly (10-20 times) push the button with 1-2 sec interval until it paired.

artcrime, 2018-09-02:
I was able to pair my sensor exactly the same way, I just had to go near the zigbee stick. Now we have to wait for a code update.

paolotremadio, 2018-09-04:
Has anyone paired the sensor with an Aqara hub? I'm working on this [PR for deCONS ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/748) but we can't figure out some fo the values... can someone attach some screenshots of the Aqara / Mi home app? I'm trying to understand if the app exposes any of the values somehow.

Koenkk, 2018-09-05:
@ben423423n32j14e would you mind making a PR?

ben423423n32j14e, 2018-09-05:
@Koenkk, sure but whats the purpose?

Btw check the wiki, I added a control panel for Node-Red. :)

Koenkk, 2018-09-05:
@ben423423n32j14e as this device is not supported (DJT11LM), it would be nice if we could add support for it. 

I saw your wiki update, great! :smile:

ben423423n32j14e, 2018-09-07:
Ordered on the 20th of August from Gearbest, I'm still waiting for mine to be delivered. :(

Tim-HD, 2018-09-09:
Hello guys,

There is another thread where  [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/748](url) @veeceeoh, @oltman, @TedTolboom, @paolotremadio and some others discussed the interpretation of the transmitted payload.

I used that input to create a (partly?) working converter for the specified messages, please find the changes below:

[devices_changes.txt](https://github.com/Koenkk/zigbee2mqtt/files/2364517/devices_changes.txt)
[fromZigbee_changes.txt](https://github.com/Koenkk/zigbee2mqtt/files/2364518/fromZigbee_changes.txt)

It would be great if some of you might give feedback on it - we can then create a PR later on if it is confirmed to be working.

Interpretation:
- action [0x55]: vibration, rotation, freefall (from @paolotremadio)
- rotationAngle [0x0503] : absolute angle from last position
- unknownParam [0x0505]: no idea what this does/indicates (maybe someone can help here, just passing it out for easier analysis)
- xAngle, yAngle and zAngle [0x0508]: "how far from horizontal are the X,Y,Z axis?" (from @oltman)

Offset from "reference" position [derived from 0x0505]:
Base position to understand the described directions is: "button in front, battery on bottom."
- aX: offset on x-axis, 0 = right side, 90 = bottom side ("battery"), 180 = left side (from @TedTolboom)
- aX: offset on y-axis, 0 = front side, 90 = bottom side ("battery"), 180 = back side  (from @TedTolboom)

Hope that helps.

Best,

Tim

tb-killa, 2018-09-09:
Hi i´m actually sitting in front of this type of device too and work on some rudimental implementation.

This is what i actually do:

``devices.js``
```
{
    zigbeeModel: ['lumi.vibration.aq1'],
    model: 'DJT11LM',
    vendor: 'Xiaomi', 
    description: 'Aqara Movement detection sticker', 
    supports: 'vibration, linkage, remote push. adjustable sensitivity', 
    fromZigbee: [fz.xiaomi_battery_3v,fz.ignore_basic_change,fz.ignore_basic_report,fz.DJT11LM_closuresDoorLock_attReport,fz.ignore_closuresDoorLock_change], 
    toZigbee: [],
},
```
``fromzigbee.js``
```
ignore_closuresDoorLock_change: {
		cid: 'closuresDoorLock',
		type: 'devChange',
		convert: (model, msg, publish, options) => null,
},
```
```
DJT11LM_closuresDoorLock_attReport

Code need to develop

```

From my understanding we need to implement timer for reset vibration return value ...

As i could read you changes .. look good .. will test later :)

dzungpv, 2018-09-10:
@tb-killa I purchased this sensor too and it is on the way.
I do the same like the post above, sniff the packet from Device with Xiaomi gateway when control with app.
But when get the data, it so hard to re-produce with Zigbee2mqtt (Zigate have tool to send raw data to test).
You can see my relate problem here: https://github.com/Koenkk/zigbee2mqtt/issues/294#issuecomment-419636303

dzungpv, 2018-09-19:
@Tim-HD I use your code and support change sensitivity in this PR https://github.com/Koenkk/zigbee-shepherd-converters/pull/74 this should close now. Note: change sensitivity must press button on the sensor to wake it up

adamomg, 2018-09-20:
So when can we expect this to be released? I just got my vibration sensor today and can't wait to use it. 

I'm using the docker version of zigbee2mqtt if that helps.

dzungpv, 2018-09-20:
@adamomg Use edge version off https://github.com/danielwelch/hassio-zigbee2mqtt/tree/master/zigbee2mqtt-edge

kimonm, 2018-09-20:
This is great, but I have a question. Since many use the configuration option "cache_state: true," the last event state is included in the mqtt message when the sensor is, e.g., just updating the battery status (i.e., when no event occurs).

To avoid this, should the code be something like:
result.event = vibrationLookup[data] ? vibrationLookup[data] : null;

Or it could return, e.g., 'none'

ben423423n32j14e, 2018-10-04:
So when will this be merged? I've got 6 of these sensors sitting on my desk ready to be setup. :)

palitu, 2018-10-11:
> So when will this be merged? I've got 6 of these sensors sitting on my desk ready to be setup. :)

yep me too!  just waiting on the merge, so that i can update my docker image :)

adamomg, 2018-10-11:
Same here. My dryer automation is waiting!

lolorc, 2018-10-11:
is support for this sensor finished ?
I get an action: "vibration", shouldn't we have an action:"no_more_vibration" when it's finished ?
I also get "unknown_data":1835008 from my vibration sensor
looks like it needs more love :-)

Koenkk, 2018-10-11:
I'm waiting for mine to arrive, will also check the implementation then.

PlayedIn, 2018-10-28:
I created a [post here](https://community.home-assistant.io/t/zigbee2mqtt-getting-rid-of-your-proprietary-zigbee-bridges-xiaomi-hue-tradfri/52108/789) describing a problem I'm having with the Vibration Sensor.

Noob question:  Am I correct in assuming that I need to shelve these sensors until the issue discussed in this thread is resolved or is the issue discussed here regarding some more advanced features of the vib sensor?

brunohorta82, 2018-10-31:
@PlayedIn  i think it's because you can set the threshold value on sensor, if you use the app Mi Home you can configure the sensibility to Vibration Sensor. I think this type of configuration is not possible with zigbee2mqtt or i'm wrong?

carekapt, 2018-10-31:
> 
> 
> @PlayedIn i think it's because you can set the threshold value on sensor, if you use the app Mi Home you can configure the sensibility to Vibration Sensor. I think this type of configuration is not possible with zigbee2mqtt or i'm wrong?

I think that the payload retrived from the sensor turns vibration into a boolean (with some kind of treshold) instead of a integer (like vibration intensity)

kabturek, 2018-12-15:
So - are we there yet ? ;) Have one sitting in my iron (decased) and would be great to use it.

ben423423n32j14e, 2018-12-15:
It's been supported for about the last month, it's working fine for me (using the dev branch).

kabturek, 2018-12-15:
thanks for the update @ben423423n32j14e, guess i need to switch to the dev branch then.

Koenkk, 2018-12-24:
AFAIK this device is fully supported in 1.0.1 and also the sensitivity can be set using https://koenkk.github.io/zigbee2mqtt/information/mqtt_topics_and_message_structure.html#xiaomi-aqara-vibration-sensor-djt11lm

icouper, 2018-12-25:
Awesome thanks! I'll take a look at the sensitivity settings. 

Is the unknown data something useful? Seems to still be there after the update to 1.0.1.

Koenkk, 2018-12-25:
Can you post logging of this?

icouper, 2018-12-26:
This is the line in the log for the vibration sensor. I've highlighted the unknown_data part.

`  zigbee2mqtt:info 2018-12-26 11:43:26 MQTT publish: topic 'zigbee2mqtt/0x00158d0002af959b', payload '{"angle_x":6,"angle_y":-77,"angle_z":12,"angle_x_absolute":84,"angle_y_absolute":167,"linkquality":89,**"unknown_data":393216**,"battery":100,"voltage":3155}'`

Koenkk, 2018-12-27:
Thanks, removed.

lolorc, 2018-12-28:
it reports vibration, does it report something like no_more_vibration so we could use it as a binary_sensor in home assistant ?
Or do we have to use automation to achieve this ?

miljume, 2019-01-01:
I have received a bunch of these sensors but haven't been able to pair it. I get `zigbee2mqtt:error 1/1/2019, 2:54:23 PM Cannot get the Node Descriptor of the Device: 0x00158d0002b2381d (Error: Timed out after 10000 ms)` and I have tried to repeatedly press reset button after initial press, removing and inserting battery but no success. I am running on the dev branch, any clues?

miljume, 2019-01-01:
After some tries I have successfully paired my device but it is not recognized `Jan 01 15:31:38 raspberrypi npm[10409]:   zigbee2mqtt:warn 1/1/2019, 3:31:38 PM Device with modelID 'undefined' is not supported.
Jan 01 15:31:38 raspberrypi npm[10409]:   zigbee2mqtt:warn 1/1/2019, 3:31:38 PM Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html
`
I am running on the dev branch and the implementations seems to be in the "zigbee-shepherd-converters" library but it is still not recognized. Any clues?

vogler, 2019-01-10:
How sensitive is this supposed to be?
According to [this comment](https://www.reddit.com/r/homeautomation/comments/9ihl6w/the_aqara_vibration_sensor/e6jtfi5) it should react to tapping - mine doesn't, I really have to move/shake it. Tilt is reasonably sensitive, but vibration not at all.
I used `mosquitto_pub -t zigbee2mqtt/Aqara_Vibration/set -m '{"sensitivity": "high"}'` to increase sensitivity but didn't notice a difference. (I pressed its button before, but maybe it still didn't work, sadly there's no `"sensitivity": "read"` to check.)
Does 'high' correspond to a threshold that could be lowered further?

Koenkk, 2019-01-11:
@vogler are you sure that the command succeeded? This is shown in the logging. Note that the device has to be awake while this command is executed (which can be done by pressing a button on the device).

vogler, 2019-01-11:
Tried again and got this:
~~~
 zigbee2mqtt:info 1/11/2019, 9:08:07 AM Zigbee publish to device '0x00158d0002b127f4', genBasic - write - [{"attrId":65293,"dataType":32,"attrData":1}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null
~~~
Still behaves the same - I have to hit the table really hard to trigger it.

ben423423n32j14e, 2019-01-11:
I've had a vibration sensor monitoring my two cats kitty litter for around a couple of months now.

Thanks to being able to see the x y z values, and combining timing data, I've been able to build it such that there is zero chance of false positives. :)

[image]

icouper, 2019-01-11:
Can you share how you're using those values to accomplish that?

ben423423n32j14e, 2019-01-13:
@icouper, here is the Node-Red flow I'm using (licensed as GNU 3)

`[{"id":"530d5d9f.5093b4","type":"mqtt in","z":"f962b96d.7044c8","name":"zigbee2mqtt/out","topic":"zigbee2mqtt/tiltsensorkittylitter","qos":"2","broker":"6e165f21.f7645","x":340,"y":520,"wires":[["3e7b496c.47da76"]]},{"id":"3e7b496c.47da76","type":"json","z":"f962b96d.7044c8","name":"","property":"payload","action":"","pretty":false,"x":490,"y":520,"wires":[["9c2b4c34.56b38"]]},{"id":"50f44b51.8bf294","type":"mqtt out","z":"f962b96d.7044c8","name":"","topic":"domoticz/in","qos":"","retain":"","broker":"cd51966f.aa7e38","x":790,"y":520,"wires":[]},{"id":"9c2b4c34.56b38","type":"function","z":"f962b96d.7044c8","name":"Format Data","func":"// Variables\nvar anglex = msg.payload.angle_x;\nvar angley = msg.payload.angle_y;\nvar anglez = msg.payload.angle_z;\nvar anglexabsolute = msg.payload.angle_x_absolute;\nvar angleyabsolute = msg.payload.angle_y_absolute;\nvar rssi = msg.payload.linkquality;\nvar batt = msg.payload.battery;\nvar time = Math.round((new Date()).getTime() / 1000);\n\n// Store battery level in context if received\nif (typeof batt !== 'undefined' && batt !== null && batt >= 1){\n    context.set(\"storedbatterylevel\", batt);\n}\n\n// Get data from context\nvar previousposition = context.get(\"position\");\nvar lastseen = context.get(\"lastseen\");\nvar battery = context.get(\"storedbatterylevel\");\n\n// Variable\nvar secondssincelastseen = (time - lastseen);\n\n// Store position events\nif(angleyabsolute >= 155) {\n    context.set(\"position\", 'Down');\n    var currentposition = 'Down';\n}else if(angleyabsolute < 155) {\n    context.set(\"position\", 'Up');\n    var currentposition = 'Up';\n}\n\n// Conditions for part trigger\nif(currentposition !== previousposition && angleyabsolute > 135 && angleyabsolute < 180 && anglexabsolute > 75 && anglexabsolute < 90) {    \n\n// Store time last seen\ncontext.set(\"lastseen\", time);\n\n// Clear unwanted objects\nmsg.payload = {}\n\n// Conditions to send to Domoticz\nif(secondssincelastseen >= 17 && secondssincelastseen <= 23){\n\ncontext.set(\"lastseen\", time);\nmsg.payload.idx = 1275 ;\nmsg.payload.svalue = 'On';\nmsg.payload.nvalue = 1;\nif (typeof battery !== 'undefined' && battery !== null){msg.payload.Battery = Math.round(battery)}\nmsg.payload.position = currentposition;\nrssi = Math.round(rssi / 11);\nif (rssi > 11) {rssi = 11;}\nmsg.payload.RSSI = rssi;\nreturn msg;\n}\n}","outputs":1,"noerr":0,"x":630,"y":520,"wires":[["50f44b51.8bf294"]]},{"id":"6e165f21.f7645","type":"mqtt-broker","z":"","name":"MQTT","broker":"127.0.0.1","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"domoticz/bridge/state","birthQos":"0","birthPayload":"online","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""},{"id":"cd51966f.aa7e38","type":"mqtt-broker","z":"","name":"MQTT","broker":"127.0.0.1","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","closeTopic":"","closeQos":"0","closePayload":"","willTopic":"","willQos":"0","willPayload":""}]`

adrianmihalko, 2019-02-05:
@vogler do you succeed in setting sensitivy?

vogler, 2019-02-05:
@adrianmihalko According to the log yes, but I didn't notice any change.
Sent `mosquitto_pub -t zigbee2mqtt/Aqara_Vibration/set -m '{"sensitivity": "high"}'`
Got `zigbee2mqtt:info 1/11/2019, 9:08:07 AM Zigbee publish to device '0x00158d0002b127f4', genBasic - write - [{"attrId":65293,"dataType":32,"attrData":1}] - {"manufSpec":1,"disDefaultRsp":1,"manufCode":4447} - null`

tweaker, 2019-02-07:
V 1.1.1
For some reason, the model identifier is not determined by me and therefore the sensor is not added.

> 2019-2-7 16:53:21 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}' 
> 2019-2-7 16:53:21 - info: Device incoming... 
> 2019-2-7 16:53:21 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}' 
> 2019-2-7 16:53:21 - info: New device with address 0x00158d0002af70bd connected! 
> 2019-2-7 16:53:21 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0002af70bd"}' 
> 2019-2-7 16:53:21 - warn: Device with modelID 'undefined' is not supported. 
> 2019-2-7 16:53:21 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html 
> 2019-2-7 16:53:21 - warn: Device with modelID 'undefined' is not supported. 
> 2019-2-7 16:53:21 - warn: Please see: https://koenkk.github.io/zigbee2mqtt/how_tos/how_to_support_new_devices.html

tweaker, 2019-02-07:
After rollback to version 0.2.0, the device was successfully added

> 2019-2-7 18:25:40 - info: New device with address 0x00158d0002af70bd connected!
> 2019-2-7 18:25:40 - info: MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00158d0002af70bd"}'
> 2019-2-7 18:25:51 - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002af70bd', payload: '{"action":"vibration","linkquality":134}'
> 2019-2-7 18:25:54 - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002af70bd', payload: '{"action":"tilt","angle":13,"linkquality":136}'
> 2019-2-7 18:25:59 - info: MQTT publish, topic: 'zigbee2mqtt/0x00158d0002af70bd', payload: '{"angle_x":-2,"angle_y":-5,"angle_z":84,"angle_x_absolute":92,"angle_y_absolute":95,"linkquality":136}'

Why is that?

icouper, 2019-02-08:
@tweaker 
I've had a rough time getting these to pair successfully. Usually I have to be very close to the hub, press and hold the sync button for 5-15 seconds, then press it every second after that for 20 seconds+. Even then I get modelID "undefined" errors often.

After 5+ tries I finally got it to sync. I've never had that much trouble with other sensors.

Also changing the battery sometimes helps.

tweaker, 2019-02-08:
@icouper
When rollback to 0.2.0, the sensor without problems immediately connected

icouper, 2019-02-12:
I've been trying to get this sensor to work on a cat door - using tilt to tell if it's opening or closing. However it doesn't seem to be sensitive enough (even on high setting) to detect the tilt unless I hold the door open much longer than it would be when the cat goes through.

Starting to think this sensor won't work for anything that I want to use it for.

Zilenzioo, 2019-02-18:
I have successfully paired vibration sensor and I can see all the data, but could someone share code that shows how to use the sensor in automation. I would like to turn light on when vibration is detected.

ben423423n32j14e, 2019-02-18:
@icouper use node red, the sensitivity you are seeing may not be the raw output from the sensor.

icouper, 2019-02-18:
@ben423423n32j14e I do use node red, but how would that improve the sensor's sensitivity?

CMDR-Sloma, 2019-02-27:
> 
> 
> AFAIK this device is fully supported in 1.0.1 and also the sensitivity can be set using https://koenkk.github.io/zigbee2mqtt/information/mqtt_topics_and_message_structure.html#xiaomi-aqara-vibration-sensor-djt11lm

Hi, I'm trying to set the sensitivity but the link is no longer available.

dsbaars, 2019-03-09:
@CMDR-Sloma If you visit https://koenkk.github.io/zigbee2mqtt/ you can see it has moved to another domain. It is still available at https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#xiaomi-aqara-vibration-sensor-djt11lm

I don't expect github pages support a smart 301 permanently moved mechanism :)

closed 2019-05-16 by stale[bot]

coolguymatt, 2019-09-06:
> @adrianmihalko According to the log yes, but I didn't notice any change.
> Sent `mosquitto_pub -t zigbee2mqtt/Aqara_Vibration/set -m '{"sensitivity": "high"}'`
> Got `zigbee2mqtt:info 1/11/2019, 9:08:07 [...]

Were you ever able to get the sensor to report vibration more accurately? I attached it to my dryer and it only responds when I open and close the door.