Koenkk/zigbee2mqtt #296: Philips Hue bulbs Pairing

SirSheik, 2018-08-21:
Hi.
I have the system running in Hass.io with zigbee2mqtt add-on v0.1.3 with some Xiaomi devices, but I have some problem to pairing philips hue bulb.
The lights bulbs models are 9290012573
I tried reset the bulb with Ikea TRÅDFRI remote control or reset the bulb with Philps API following this manual:
http://thingsthataresmart.wiki/index.php?title=How_to_Reset_Philips_Hue_Bulbs

But in both cases I dont see the lights bulbs after enable "permit_join" and turn off and on the lights bulb.
Can someone help me by telling me what is the correct process to pairing the lights bulbs?
Thanks you so much

Koenkk, 2018-08-21:
Have you tried following: https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#philips-hue (especially the youtube video).

SirSheik, 2018-08-21:
Hello.
Thanks for your response
Sorry but I haven't Philips Dimmer Switch.
I tried the alternative option:
"This is also possible with the Tradfri Remote Control by pressing and holding the reset button on the bottom of the remote (next to the battery)."
Regards

Koenkk, 2018-08-21:
Do you see the flashing of the bulb after the reset (like in the video)?

SirSheik, 2018-08-22:
Hi.
I've recorded a video so you can see the steps I'm taking to try to link the light bulbs.
http://www.youtube.com/watch?v=rez1IVx9pFg

Any suggestions? I am doing something wrong?
Regards

kiwinol, 2018-08-23:
It looks like you are resetting the bulb correctly.  Something I noticed is that it can take up to a minute to pair.

SirSheik, 2018-08-23:
Hi. 
I tried again a couple of times waiting 5 minutes after the factory reset but nothing happened.
These lights used channel 11 on the philips bridge before unpairing it. Could it be the problem?
Thx

kiwinol, 2018-08-23:
One possibility (I think this happened to me) is that your light was found and added to zigbee2mqtt but then you reset it.  If this is the case you have to identify it by a process of elimination and remove it using the mqtt remove payload.

Also after resetting the light (it goes coloured then white) don't turn it off, leave it (I thought I saw you turn it off an back on again after)

SirSheik, 2018-08-24:
Hello
Thanks for the comment, I did the same test with another bulb that I had not yet tested trying to pairing without turning off the bulb as you told me.
But the result has been the same.
The strange thing is that nothing appears in the log at any time, no error, or the typical message that appears when you pair other devices: **"spinlock() false()"**
Regards

SirSheik, 2018-08-29:
Hi.
I got a philips remote control and it worked fine. It worked on the first try.
I think it would be a good idea to remove the option with the TRÅDFRI remote control from the wiki because It does not work, not for me.
You can close this post
Regards

Koenkk, 2018-08-30:
I've updated the wiki with a reference to this issue (as it did work in my case).

closed 2018-08-30 by Koenkk

Dams-38, 2018-12-09:
Same problem than SirSheik … the bulb flash, pair with the Ikea remote but never move in permit mode

LuigiTheHunter, 2019-03-09:
@Koenkk May I ask how you got the bulb to reset? Mine is flashing when I hold the reset button but I cant discover it, afterwards like the other guys already tried it. Any Ideas on what you did?

TinusScheppers, 2019-03-09:
> 
> 
> @Koenkk May I ask how you got the bulb to reset? Mine is flashing when I hold the reset button but I cant discover it, afterwards like the other guys already tried it. Any Ideas on what you did?

I've got exactly the same issue with the Tradfri remote, it flashes and seems to reset, but it does not pair.
However I did get this error in the logfile the first time trying to connect:
` error: Cannot get the Node Descriptor of the Device: 0x000b57fffe12f208 (Error: Timed out after 10000 ms)`

And this one:

` info:  Error: request timeout
    at CcZnp.<anonymous> (/zigbee2mqtt-1.1.1/node_modules/cc-znp/lib/ccznp.js:255:22)
    at Object.onceWrapper (events.js:273:13)
    at CcZnp.emit (events.js:182:13)
    at Timeout.<anonymous> (/zigbee2mqtt-1.1.1/node_modules/cc-znp/lib/ccznp.js:234:18)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
`

But that was about it, afterwards none of these came around and the bulb was still not paired...

Koenkk, 2019-03-09:
I've just tried to re-pair my hue bulb, but also failed to do this using the TRADFRI remote control, somehow it doesn't reset it correctly (sometimes??).

I guess the best way to reset it is using the Hue dimmer switch.

TinusScheppers, 2019-03-09:
But doesn't the above log show some activity?
Given that the lamp does reset maybe something else go's wrong when pairing, in some cases.
Strange indeed...

TinusScheppers, 2019-03-19:
Can confirm that resetting using the Philips dimmer works, however in my case using the Tradfri sadly didn't.

LuigiTheHunter, 2019-03-19:
Same thing for me

Koenkk, 2019-11-23:
An alternative options has been introduced to factory reset hue bulbs, see https://github.com/Koenkk/zigbee2mqtt/issues/2396

JsBergbau, 2021-02-21:
Hello,

I had a quite similiar issue and I think this helps people.

I have philips living colors bloom and paired it with Zigbee2MQTT. I have a remote control 3rd generation (click wheel). Zigbee2MQTT is on channel 26, which is no light link channel (11, 15, 20, or 25 are light link channels, see https://www.zigbee2mqtt.io/information/configuration.html )
 
I've resetted remote control, pressed the on switch for a long time directly i front of the lamp. However I haven't been able to pair the bulb again with its original remote.

As the bulb was still connected to Zigbee2MQTT I wanted to try via tochlink and send `zigbee2mqtt/bridge/request/touchlink/scan` however result was only
```
{
  "data": {
    "found": []
  },
  "status": "ok"
}
```

Luckily I had a Ike Tradfri Remote and tried to resetting with it. On the first blink of the lamp, I released reset button. After that I still couldn't pair orignal remote control.
I've tried a second time, hold the reset button much longer, letting the lamp blink a lot of times. Then I released reset button and within 3 seconds I took the original remote, pressed the on button et voilà lamp was paired again with original remote control.

kris10an, 2021-08-22:
I had a similar issue. Hue bulb no matter how many resets I did with the Hue dimmer. Bringing it closer didn't help. 
This was in the process of moving all bulbs from a hue hub to zigbee2mqtt. I had therefore deleted it from my hue hub before, so it might have joined at some point, but didn't show up in devices and I didn't see any debug log entries.

In my case, since I still have the hue hub, so I added it to the hue hub (found and joined in less than 10 seconds). I deleted it from the hue hub again and it immediately joined zigbee2mqtt.

turboproc, 2022-09-03:
Interesting, had to move a HUE bulb from one zigbee2mqtt network to the other zigbee2mqtt network. Procedure with Ikea Tradfri remote control works like a charm. First time it failed but that was because the HUE was part of the original network. After removing the HUE from that network and applying the procedure again it was done within seconds

chiro79, 2023-02-20:
Just reset a Philips Hue Bulb with the IKEA control. I had to do it twice. I think the trick is to put Zigbee2MQTT in ”Permit all” mode and the control close to the bulb. Then press and hold the “reset” button on the IKEA control. When the light start to flash, take the control far away from the bulb. At least, that worked for me.