Koenkk/zigbee2mqtt #297: Disable led of QBCZ11LM

jasimancas, 2018-08-21:
Its possible disable the led of the QBCZ11LM?

MiHome have this option, is possible via MQTT?

Koenkk, 2018-08-21:
Not yet, we have to see what zigbee command is used for this.

jasimancas, 2018-08-22:
If you need help with this, tell me.

ryanbeaton, 2018-08-22:
I hope it’s the same for the ZNCZ02LM. I’d like to see that

ryanbeaton, 2018-08-23:
Does @subzero79 info help in https://github.com/Koenkk/zigbee2mqtt/issues/201#issuecomment-415231510

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

jasimancas, 2018-08-23:
Great! :D

subzero79, 2018-08-25:
Also worth to mention i did notice that once the attribute is set via gateway, and then repairing with zigbee2mqtt the attribute stays, the same happens with restore power state. So if you want to disable the led just pair them with the gw, set it via app, then repair again. with zigbee2mqtt

ryanbeaton, 2018-08-25:
Thanks @subzero79 That’s good to know. I’ve never had a bridge as started my purchasing after starting with Zigbee2MQTT, hmmmm

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2018-11-09 by stale[bot]

YES-Lee, 2025-02-28:
> If you need help with this, tell me.

Hi, how do you resolve this?