Koenkk/zigbee2mqtt #300: Problem connecting to MQTT from Docker

clockbrain, 2018-08-21:
Trying to move zigbee2mqtt from Pi to Docker on Synology NAS but once there it doesn't want to connect to MQTT.

Current setup which works ok:
- zigbee2mqtt 0.1.3 with latest firmware on a Raspberry Pi (non-docker).
- MQTT in Docker on Synology NAS
- Home Assistant in Docker on Synology NAS

Desired setup - everything on the same Synology NAS in separate Docker containers.

Here's the log file error I get (repeated).
```
error: Not connected to MQTT server!
```

Log file
https://pastebin.com/jQeGPkVv

I don't think MQTT broker itself is the problem as I also have another client app on a separate Pi messaging it without problems.

I've followed the zigbee2mqtt docker instructions in the wiki but am trying to configure using the Synology Docker gui rather than via command line.

I'm running the zigbee2mqtt container with high privilege so it can access usb port. Tried both host network and bridge network for the docker container although I'm not confident about the bridge network settings.

Does anyone have zigbee2mqtt running in Docker on Synology and talking to MQTT in Docker on the same NAS?

Koenkk, 2018-08-21:
From inside the zigbee2mqtt container, can you ping the MQTT broker ip adress?

clockbrain, 2018-08-22:
Thank you @Koenkk 

That was the hint I needed. Works ok now.

The trick was to specify in configuration.yaml that MQTT is to be found locally.

```
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://127.0.0.1:1883'
```

The reason for moving zigbee2mqtt to docker on Synology is that becomes my production system for stable devices and it frees up the Pi to use as a development system for experimenting with pairing other devices. Yes I have 2 CC2531 sticks, one for each system.

closed 2018-08-22 by clockbrain

Sfieszaq, 2019-06-10:
Hi,

where I can find configuration.yaml on my NAS?