Koenkk/zigbee2mqtt #302: Cannot use CC2530 with RFX2401

Ton1965, 2018-08-23:
Hi Guys,

I am a complete noob in chips and firmware flashing but would like to do my first steps in it.
I have some Xiaomi zigbee devices that are currently paired with Xiaomi gateway but I want to get rid of the gateway so I plan to use zigbee2mqtt.
I have bought CC2530 with RFX2401 that looks exactly like the last one in [supported sniffer devices table](https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-sniffer-devices) and connected pins 04, 06 and 20 to the ground.
I managed to flash it using the [method with Arduino instead of CCDebugger](https://github.com/Koenkk/zigbee2mqtt/wiki/Alternative-firmware-flashing-methods). The firmware [advised in wiki](https://github.com/kirovilya/files/blob/master/CC2530ZNP-Test_for_CC2591.hex) was used. The only difference: I had to remove the line before the last one from the firmware since it contained the record type (05) that could not be handled by CCLib (found this advice somewhere on the net). The flash process completed successfully.
The problem is I cannot use it with zigbee2mqtt. It fails to start, reports an error, tries to fix it and then just fails with a message about timeout reached.
What I have already tried:

- change baudrate in zigbee2mqtt
- set rtscts to both true and false
- changed log level but no additional information appeared

Could someone help me please? Is there any way to make sure I have not broken it while soldering the pins?

Koenkk, 2018-08-24:
Maybe off topic, but for the sake of simplicity I would start with a CC2531.

Ton1965, 2018-08-25:
I probably would agree to that. I just hoped to get better coverage with this device. I have seen some posts on HA forum where people complained about the coverage with CC2531 being too small. @kirovilya said CC2530 with RFX2401 provided a longer range.

BTW, @Koenkk and @kirovilya, probably some one of you could very easily answer a question that I have. I discovered that the ping 04, 06 and 20 that I soldered to the ground pin are currently connected with the metal shield that covers the chip. As I said, I am just at a very basic level and the question might be silly but anyway: is this connection ok? I assume it is since this shield should probably be also grounded, i.e. connected to the ground pin, but I am not sure.

Thanks for the reply.

kirovilya, 2018-08-26:
@Ton1965 hello. If you look at the picture, then these contacts do not connected with case.
https://github.com/Koenkk/zigbee2mqtt/issues/52#issuecomment-391055372

Ton1965, 2018-08-26:
Guys,

I would like to share my joy! I managed to start it!

First of all I double checked the absence of any short-circuited pins. I have measured once more the resistance between the soldered pin and the chip shield and it was zero as before. BTW, I have received the same resistance between the chip shield and USB shield.

Then I inserted the chip into USB and tried to run the bridge and it gave the same errors as before. After struggling a bit more with the settings in configuration.yaml I discovered that settint 'rtscts' to false allowed the bridge to start!

I have already tried this before and got no positive results. The key difference is **I killed the 'node' process** each time I changed the settings. I noticed that if the bridge is killed with Ctrl-C after it had given that error message the 'node' process is still left running. Probably this background process interfered somehow with the restart with a different set of settings.

Thanks everyone for their replies! I will try now to integrate the bridge into my HA and move all the devices to it.

BTW, the 'node' process does not remain running if the bridge is killed during its normal operation.

123trs, 2018-08-26:
Hello Toni1965,

I have the same problem with the CC2530 stick. I dont know the 'rtscts' setting in the configuration.yaml.
Please show yours.

Ton1965, 2018-08-26:
Hi @123trs,
It is documented in [Configuration](https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration) section of the Wiki. See the last item in the list.

123trs, 2018-08-26:
Ok, found and works☺️

Ton1965, 2018-08-27:
Great to hear!

BTW, probably this information could be added to the Wiki.

dzungpv, 2018-08-27:
@Ton1965 Do you have it schematic? You can try to adapt my firmware to it, but if they use the same port to control RFX2401, it will work, CC2591 they use p0.7, p1.4 and p1.1 (you can find TI ref design about cc2591 to know how they wire it).
Or more simple, just buy CC2530+CC2591 and flash the firmware https://github.com/dzungpv/Z-Stack-firmware/tree/master/coordinator/CC2530_CC2591

ciotlosm, 2018-09-03:
I guess this issue can be closed once the docs have been updates?

Ton1965, 2018-09-04:
Yes, I think so. This sticks works very stable after 'rtscts' is set to 'false'

I have not had a chance to check another firmware suggested by @dzungpv (will try after my vacation probably:-)) but this is a different issue (or not an issue at all).

talondnb, 2018-10-25:
I'm having trouble with this. I've flashed the stick using cc-tool, have the pins soldered correctly but I'm using the hassio addon.

Flashing:

 MacBook-Air:cc-tool$ sudo ./cc-tool -e -w CC2530ZNP-Test_for_CC2591.hex 
  Programmer: CC Debugger
  Target: CC2530
  Erasing flash...
  Completed       
  Writing flash (220 KB)...
  Completed (15.51 s.)

Here is my hassio addon config:
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": false,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://homeassistant",
  "serial_port": "/dev/ttyUSB0",
  "rtscts": false
}

And here are the logs from the addon:

[Info] Configuration file found. Will overwrite configurable               fields with values from add-on configuration
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2018-10-26T06:44:29: PM2 log: Launching in no daemon mode
2018-10-26T06:44:30: PM2 log: App [npm:0] starting in -fork mode-
2018-10-26T06:44:30: PM2 log: App [npm:0] online
> zigbee2mqtt@0.1.8 start /zigbee2mqtt-0.1.8
> node index.js
  zigbee2mqtt:info 2018-10-26 06:44:34 Logging to directory: '/share/zigbee2mqtt/log/2018-10-26.06-44-34'
  zigbee2mqtt:info 2018-10-26 06:44:35 Starting zigbee2mqtt version 0.1.8 (commit #unknown)
  zigbee2mqtt:info 2018-10-26 06:44:35 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-10-26 06:44:38 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
  zigbee2mqtt:info 2018-10-26 06:45:38 Starting zigbee-shepherd
  zigbee2mqtt:error 2018-10-26 06:45:41 Error while starting zigbee-shepherd!
  zigbee2mqtt:error 2018-10-26 06:45:41 Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 2018-10-26 06:45:41 Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-0.1.8/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/zigbee2mqtt-0.1.8/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:498:11)\n    at tryOnTimeout (timers.js:323:5)\n    at Timer.listOnTimeout (timers.js:290:5)"}

Any suggestions on what to try next?

talondnb, 2018-10-25:
BTW, are the LEDs on the stick supposed to light at all?

Ton1965, 2018-10-26:
@talondnb, are you using the device shown at the picture next to RFX2401 at [wiki](https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-sniffer-devices)? I am using the one and it does not have any LEDs (at least none I know about).

I have been using it for several months already and it looks very stable. The only thing I changed in the configuration was that rtscts setting. If you have a different stick then you should probably change it back to true or change something else. I got the same set of messages with the wrong rtscts but I believe these messages would occur in many different cases.

talondnb, 2018-10-26:
> @talondnb, are you using the device shown at the picture next to RFX2401 at [wiki](https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-sniffer-devices)? I am using the one and it does not have any [...]

Yes, using the black dongle. There are LEDs on the PCB but don’t seem to light.

[image]

Can you reconfirm the firmware hex and method you used to flash?

ryanbeaton, 2018-10-26:
Can you show the soldered to pins better I doesn't look quite right to me.  The "d2" side should be the 4th pin from d2 and yours looks like the third

https://github.com/Koenkk/zigbee2mqtt/issues/52#issuecomment-391099828

ryanbeaton, 2018-10-26:
PS: Ive never gotten around to soldering mine but I plugged it in to see if I could help you today. I get the same output as you.  "ttyUSB0" didn't look right to me but I checked and it is

```
~ # ls -l /dev/serial/by-id
lrwxrwxrwx    1 root     root            13 Oct 26 09:28 usb-Silicon_Labs_CP2102_USB_to_UART_Bridge_Controller_0001-if00-port0 -> ../../ttyUSB0
```

Ton1965, 2018-10-26:
@talondnb , nice to hear your problem is solved! BTW, I did not even notice those LEDs :smile:
Do you observe any flashing? I never noticed anything but my stick is a little bit out of sight so probably I just missed it.

talondnb, 2018-10-26:
> @talondnb , nice to hear your problem is solved! BTW, I did not even notice those LEDs 😄
> Do you observe any flashing? I never noticed anything but my stick is a little bit out of sight so probably I just missed it.

I've not seen them flashing, no biggie though as long as it's working! A definite improvement in signal strength I feel too.

closed 2019-01-01 by stale[bot]