Koenkk/zigbee2mqtt #304: Xiaomi devices and their batteries

ciB89, 2018-08-24:
Hey,

quick question for everyone buying/owning xiaomi devices. So far EVERY xiaomi device I bought was unable to pair with the dongle unless I swapped the batteries. I now got a window sensor and a temperature/humidity sensor and the dongle wouldn't even read any signals from them (was the same for the other devices before swapping the batteries). Unfortunately I have no more batteries here atm, but I am curious whether I am the only one experiencing this. Because you cannot tell that it's the battery - it blinks 3 times like it should, but then the coordinator does not read anything. I hope this time its again only the battery.

kiwinol, 2018-08-24:
I have about 12 Xiaomi sensors and switches connected and didnt have to swap the battery on any of them.

cjsmns, 2018-08-24:
I'm just starting out and had no problems. I've just added a WSDCGQ11LM Aqara Temperature sensor that was delivered today. Worked fine out of the box. Had to wait a few minutes for it's to actually post any readings. Just make sure you give it 5 minutes or so after you pair it.

ciB89, 2018-08-24:
Mine won't even start the paring process. It's like their not sending any signals. I fear that this time it's not a battery problem though ... :(

closed 2018-08-28 by ciB89