Koenkk/zigbee2mqtt #305: Error Starting Zigbee2mqtt on Docker (Synology).

aptonline, 2018-08-24:
Unsupported device
First check: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

Bug report
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

I have a newly flashed CC253X (unsure of firmware as was suppled to me) and initially the drive did pickup a Tradfri bulb using the latest docker on my Synology but now I am seeing the following error:

```
Using '/app/data' as data directory
zigbee2mqtt@0.1.3 start /app
node index.js
zigbee2mqtt:info 2018-8-24 16:04:34 Logging to directory: '/app/data/log/2018-08-24.16:04:33'
zigbee2mqtt:info 2018-8-24 16:04:43 Starting zigbee2mqtt version 0.1.3 (commit #c8c8417)
zigbee2mqtt:info 2018-8-24 16:04:43 Starting zigbee-shepherd
zigbee2mqtt:info 2018-8-24 16:04:46 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
zigbee2mqtt:info 2018-8-24 16:05:46 Starting zigbee-shepherd
zigbee2mqtt:error 2018-8-24 16:05:49 Error while starting zigbee-shepherd!
zigbee2mqtt:error 2018-8-24 16:05:49 Press the reset button on the stick (the one closest to the USB) and start again
zigbee2mqtt:error 2018-8-24 16:05:49 Failed to start
{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/cc
```

I've been led to believe this is a firmware issue on the stick and it needs updating.  Can some one confirm this is the case? as I don't have the debugger and cables I need to look at getting this updated from my source or purchased the relevant hardware but would like to know before hand.

This is my config my Zigbee2mqtt:

```
# Home Assistant integration (MQTT discovery)
homeassistant: true

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: homeassistant
  # MQTT server URL
  server: 'mqtt://192.168.1.12'
  # MQTT server authentication, uncomment if required:
  user: homeassistant
  password: Submarine72

# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/ttyACM0
```

Koenkk, 2018-08-24:
What is the output of `ls /dev | grep ACM`?

aptonline, 2018-08-24:
I get ttyACM0 from the terminal connected on my Synology which is where the stick is connected.

aptonline, 2018-08-24:
for reference this is the docker command I want initially which includes the device attachment:

```
docker run \
   -it \
   -v /volume1/docker/zigbee2mqtt:/app/data \
   --device=/dev/ttyACM0 \
   koenkk/zigbee2mqtt:latest
```

Koenkk, 2018-08-26:
if it worked before, can you try: replug usb -> press reset button (one closest to usb) -> start zigbee2mqgt

philhawthorne, 2018-08-27:
@aptonline might also help to try the `--privileged` flag as well, to give the container higher access to the device. May or may not help

aptonline, 2018-08-27:
Thanks @Koenkk and @philhawthorne both both suggestions didn't help. I thought it may also have to do with my having USB copy installed and configured but still no joy.  Going to try another reboot of the NAS then I think its about getting the stick flashed again.

19Homer73, 2018-08-30:
You have to change the starting-settings (or priority...) for the docker container to high, before you start it!
Then the USB-stick is mapped correctly and it will work!
I had the same problem...

closed 2018-11-09 by stale[bot]