Koenkk/zigbee2mqtt #306: Sending config msg works only with default topic

littleyoda, 2018-08-24:
```
mqtt:
  base_topic: IO/Device/Controll/zigbee2mqtt
  server: 'mqtt://localhost'
```

e.g. I want to change the debug level. I receive the following error in the log file:

```
zigbee2mqtt:debug 2018-8-24 20:38:14 Recieved mqtt message on topic 'IO/Device/Controll/zigbee2mqtt/bridge/config/log_level' with data 'debug'
zigbee2mqtt:warn 2018-8-24 20:38:14 Cannot handle MQTT config option 'zigbee2mqtt' with message 'debug'
```

I assume the following line in controller.js is the problem:
```
    handleMQTTMessageConfig(topic, message) {
        const option = topic.split('/')[3];
```
Instead of the fourth element, the second last element would have to be used.

littleyoda, 2018-08-24:
The following line is also wrong:
```
   handleMQTTMessageDevice(topic, message, withPrefix) {
        const friendlyName = topic.split('/')[1];
```

Koenkk, 2018-08-24:
You are right! Would you mind making a PR for this?

closed 2018-09-07 by littleyoda