Koenkk/zigbee2mqtt #307: CC2530 arduino uno flashing help

pleasanthill17, 2018-08-25:
I am trying to get arduino uno and CC2530 wired to flash CC2530. 

CC2530 purchased from the supported device and looks like the device in the link.  https://www.aliexpress.com/item/CC2530-Zigbee-UART-Wireless-Core-Board-Development-Board-CC2530F256-Serial-Port-Wireless-Module-24MHz/32767470382.html 

Wired Arduino as shown in the wiki https://github.com/Koenkk/zigbee2mqtt/wiki/Alternative-firmware-flashing-methods and used the arduino sketch. 

I do not cc-debugger, I connected the arduino pins to the CC2530 directly as shown in the image below:

GND-GND
RST-RESET
DC-P2.2
DD-P2.1
3.3V-VCC
On CC2530 GND to P2.0

[image]

When I connect arduino with CCLib, I get the error message chip/device not found.

In the http://ptvo.info/how-to-select-and-flash-cc2530-144/ it says to connect Pin#2 to 3.3V. Since I do not have the debugger, what arudino pin do I connect to which pin on the CC2530 board to accomplish that target voltage sense connection ?

(I know arduino sketch and wires are fine, since I followed the wiki for alternative flashing and was able to flash CC2531 successfully.)

kirovilya, 2018-08-26:
As described, you need to connect 3 data contacts + power + gnd. It's enough.
Check that you are connected to the correct pins on arduino.

ciotlosm, 2018-09-03:
@pleasanthill17 please close this issue if you got it resolved.