Koenkk/zigbee2mqtt #308: Can't pair more than 16 Xiaomi devices

blakeblackshear, 2018-08-26:
I am trying to move away from the Xiaomi hub, and I seem to be hitting a limit of 16 xiaomi devices. Once I have paired 16 devices, I stop seeing any messages related to pairing. I tried restarting the docker container and the RPi3. I re-flashed an extra CC2531 usb stick, started fresh, and ran into the exact same issue after 16 devices again. I am running the latest docker build on a RPi3. The 16 devices that are paired work great.

I also tried removing one of the 16 devices and pairing one of the devices that I was unable to pair. This did not work either, but I was able to pair the device I had removed to get back to 16 again. It is as though once it hits 16 devices, it cannot see messages from any new devices. I have 6 more devices, and I have tried them all without success. Also, the first time I tried to pair all of them, I paired in a different order, and I was able to pair some of the 6 that will not join, until hitting 16 devices.

Logs: https://pastebin.com/vS2Si1M7

Koenkk, 2018-08-26:
As you only have end devices, you need to add some routers to be able to increase the network size (e.g. a xiaomi power plug)

blakeblackshear, 2018-08-26:
Good to know I am not going insane I guess. Based on [this wiki page](https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network#network-size), I was expecting 35. I didn't realize the device types would introduce additional limitations.

Koenkk, 2018-08-26:
Sorry, the wiki was outdated