Koenkk/zigbee2mqtt #309: zigbee2mqtt stop reveive data and green is turned off

pixiandreas, 2018-08-27:
Hello
First of all thanks for the software works very well but i have a strange problem.

When i run the software for a time the green led on the CC2531 turn off and no more data is received.
I can't find any reason why this occur, sometime is happen only 30 minutes after i restart it but it could also works for days with out any problem?

I'm running if from docker (latest)
And using the last firmware for CC2531.

The only thing that is not so "normal" is that i run the host (ubuntu 18.04) on Vmware and use USB  pass throw to the host .

If i restart the docker container everything is ok again, so i don't need to unplug the CC2531.

Here is the debug log.
https://pastebin.com/GRKrA4Ag

And here is the debug log just after the restart.
https://pastebin.com/x2G6Biv2

Regards
Andreas

Koenkk, 2018-08-27:
Does this also happen when running without vmware? (on a pi for example)

pixiandreas, 2018-08-27:
Hi again.
I install it on a pi (2) direct after i open the issue
It have no run on the pi for 1 hours and i have restart it and pull out the cable and it always start with now issue.

And everything feels quicker than before.

On Vmvare i have an another problem (not reported) that when the system has been idle for around 10 minutes the first button press was not recognized.  
I see the blue led on the enddevice but nothing happen in the log, the next press everything was ok again. 
When i run in on a pi the issue is also gone.

Maybe you could do some settings in Vmware for the USB pass throw but i haven't  found any settings for this.
And i don't want to change to much because i use other usb devices in my HASS server.

ciB89, 2018-08-28:
I just recently had the same issue on Hassio with my raspi 3. Needed to acutally unplug the raspi in order to get the stick to work again properly. I first thought that my stick broke, but now its working fine again.

ciotlosm, 2018-09-03:
@pixiandreas so the problem remaining is just with VMWare setup?

pixiandreas, 2018-09-03:
Hi.
I have run it now on on docker in a Ubuntu 18 server for 7 days and no problem at all.
Also a several restart and reboot. All fine.

It works better that i thought.
Thanks for the software and work.

closed 2018-11-09 by stale[bot]