Koenkk/zigbee2mqtt #310: Philips HUE motion sensor - No converters missing message

TheCellMC, 2018-08-27:
I'm trying to connect my Philips HUE motion sensor to zigbee2mqtt, however I'm having trouble doing so.

I was easily able to pair the device with zigbee2mqtt and I added my device to `node_modules/zigbee-shepherd-converters/devices.js`.

Here is my config:
```
{
    zigbeeModel: ['SML001'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '9290012607', // Vendor model number, look on the device for a model number
    vendor: 'Philips Hue', // Vendor of the device (only used for wiki and startup logging)
    description: 'Philips Hue motion sensor ', // Description of the device, copy from vendor site. (only used for wiki and start$
    supports: 'temperature, light and humidity', // Actions this device supports (only used for wiki)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},
```

However, once I saved this and restarted the bridge, I didn't get any message for converters missing, even with debug enabled. I've tried resetting my motion sensor and restarting the bridge. However the messages never appear when I move my hand over the sensor.

I do get this in the log when initially connecting the sensor after a factory reset of the sensor.
```
  zigbee2mqtt:debug 2018-8-27 17:45:33 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:info 2018-8-27 17:45:33 Device incoming...
  zigbee2mqtt:info 2018-8-27 17:45:33 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:info 2018-8-27 17:45:33 New device with address 0x001788010212dcd0 connected!
  zigbee2mqtt:info 2018-8-27 17:45:33 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x001788010212dcd0"}'
```

TheCellMC, 2018-08-27:
Here is my full log after a restart of zigbee2mqtt:

```
> zigbee2mqtt@0.1.3 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2018-8-27 17:54:56 Logging to directory: '/opt/zigbee2mqtt/data/log/2018-08-27.17:54:56'
  zigbee2mqtt:debug 2018-8-27 17:54:57 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"channelList":[11]},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2018-8-27 17:54:57 Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 2018-8-27 17:54:57 Starting zigbee2mqtt version 0.1.3 (commit #c8c8417)
  zigbee2mqtt:info 2018-8-27 17:54:57 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-8-27 17:54:57 zigbee-shepherd started
  zigbee2mqtt:info 2018-8-27 17:54:57 Coordinator firmware version: '20180507'
  zigbee2mqtt:debug 2018-8-27 17:54:57 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0014d9d6ca","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180507},"startTime":1535392497,"joinTimeLeft":0}
  zigbee2mqtt:info 2018-8-27 17:54:57 Currently 1 devices are joined:
  zigbee2mqtt:info 2018-8-27 17:54:57 0x001788010212dcd0 (0x001788010212dcd0): 9290012607 - Philips Hue Philips Hue motion sensor  (EndDevice)
  zigbee2mqtt:info 2018-8-27 17:54:57 Connecting to MQTT server at mqtt://192.168.1.25
  zigbee2mqtt:info 2018-8-27 17:54:58 zigbee-shepherd ready
  zigbee2mqtt:info 2018-8-27 17:54:58 Connected to MQTT server
  zigbee2mqtt:info 2018-8-27 17:54:58 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:debug 2018-8-27 17:55:02 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:03 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:03 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:04 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:04 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:05 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:05 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:07 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:08 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:09 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:10 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:11 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:12 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:13 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:debug 2018-8-27 17:55:13 Recieved zigbee message with data "0x001788010212dcd0"
  zigbee2mqtt:info 2018-8-27 17:55:13 Device incoming...
  zigbee2mqtt:info 2018-8-27 17:55:13 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:debug 2018-8-27 17:55:13 Recieved zigbee message with data "online"
```

ryanbeaton, 2018-08-28:
Linking to the old issue
https://github.com/Koenkk/zigbee2mqtt/issues/54

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

TheCellMC, 2018-08-28:
I'm making some progress.
I believe that this issue is very similar to #36 

I've added the following to `/opt/zigbee2mqtt/lib/zigbee.js` under the `handleReady` function:
```
        // TEMP: Bind Hue dimmer to coordinator.
        let huesensor = '0x001788010212dcd0';
        huesensor = this.shepherd.find(huesensor, 1);
        let coordinator = this.shepherd.list().find((d) => d.type === 'Coordinator');
        coordinator = this.shepherd.find(coordinator.ieeeAddr, 1);

        huesensor.bind('genOnOff', coordinator, (err) => {
            if (err) {
                console.log('FAILED TO BIND :(');
            } else {
                console.log('GOOD!');
            }
        });
```

I then started `zigbee2mqtt` with `DEBUG=zigbee-shepherd* npm start` and waited to receive the `GOOD` message.

```
Now I get this when I wave my hand over the sensor:
zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +17s 
zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +12ms
```

I then  started `zigbee2mqtt` with `DEBUG=* npm start`.

```
Here is what I got when I triggered the sensor:

  serialport:poller received "readable" +5s
  serialport:bindings read +5s
  serialport:unixRead Starting read +5s
  serialport:unixRead Finished read 33 bytes +1ms
  serialport:main binding.read finished +5s
  cc-znp { sof: 254,
  cc-znp   len: 28,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload: 
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 6,
  cc-znp      srcaddr: 4515,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 78,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 6012053,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 8,
  cc-znp      data: <Buffer 01 07 42 00 b8 0b 00 00> },
  cc-znp   fcs: 1,
  cc-znp   csum: 1 } +5s
  serialport:main _read reading +14ms
  serialport:bindings read +16ms
  serialport:unixRead Starting read +16ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 6, srcaddr: 4515, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 78, securityuse: 0, timestamp: 6012053, transseqnumber: 0, len: 8, data: <Buffer 01 07 42 00 b8 0b 00 00> } +10ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +13ms
  serialport:unixRead waiting for readable because of code: EAGAIN +16ms
  serialport:poller Polling for "readable" +35ms
```

tb-killa, 2018-08-28:
Please remove your test-code under the ``handleReady`` function and use this test configure snippet in the ``devices.js`` file:

```
{
    zigbeeModel: ['SML001'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '9290012607', // Vendor model number, look on the device for a model number
    vendor: 'Philips Hue', // Vendor of the device (only used for wiki and startup logging)
    description: 'Philips Hue motion sensor ', // Description of the device, copy from vendor site. (only used for wiki and start$
    supports: 'temperature, light and humidity', // Actions this device supports (only used for wiki)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
    const device = shepherd.find(ieeeAddr, 1);
    const actions = [
        (cb) => device.bind('genOnOff', coordinator, cb),
     ];
    execute(device, actions, callback);
    },
},
```
And post again log vial DEBUG (bigger to pastebin or simulare).

TheCellMC, 2018-08-28:
Ok.
So here is the full log when I startup the service: https://pastebin.com/ZZF4pzc3
And here is what I get when I trigger the sensor: 
```
  serialport:poller received "readable" +25s
  serialport:bindings read +25s
  serialport:unixRead Starting read +25s
  serialport:unixRead Finished read 33 bytes +0ms
  serialport:main binding.read finished +25s
  cc-znp { sof: 254,
  cc-znp   len: 28,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload: 
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 6,
  cc-znp      srcaddr: 4515,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 68,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 1058168,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 8,
  cc-znp      data: <Buffer 01 09 42 00 b8 0b 00 00> },
  cc-znp   fcs: 58,
  cc-znp   csum: 58 } +25s
  serialport:main _read reading +8ms
  serialport:bindings read +10ms
  serialport:unixRead Starting read +10ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 6, srcaddr: 4515, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 68, securityuse: 0, timestamp: 1058168, transseqnumber: 0, len: 8, data: <Buffer 01 09 42 00 b8 0b 00 00> } +4ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +11ms
  serialport:unixRead waiting for readable because of code: EAGAIN +14ms
  serialport:poller Polling for "readable" +25ms
```

tb-killa, 2018-08-28:
Great please provide content of database.db.
Maybe there are multiple entries for same device.

TheCellMC, 2018-08-28:
Here are the contents of the database.db file:
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014d9d6ca","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1535476894,"endpoints":{"1":{"profId":260,"epId":1,"devId"$
{"id":2,"type":"EndDevice","ieeeAddr":"0x001788010212dcd0","nwkAddr":4515,"manufId":4107,"manufName":"Philips","powerSource":"Battery","modelId":"SML001","epList":[1,2],"status":"online","joinTim$
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0014d9d6ca","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1535477846,"endpoints":{"1":{"profId":260,"epId":1,"devId"$
{"id":2,"type":"EndDevice","ieeeAddr":"0x001788010212dcd0","nwkAddr":4515,"manufId":4107,"manufName":"Philips","powerSource":"Battery","modelId":"SML001","epList":[1,2],"status":"offline","joinTi$
{"id":2,"type":"EndDevice","ieeeAddr":"0x001788010212dcd0","nwkAddr":4515,"manufId":4107,"manufName":"Philips","powerSource":"Battery","modelId":"SML001","epList":[1,2],"status":"online","joinTim$
{"id":2,"type":"EndDevice","ieeeAddr":"0x001788010212dcd0","nwkAddr":4515,"manufId":4107,"manufName":"Philips","powerSource":"Battery","modelId":"SML001","epList":[1,2],"status":"online","joinTim$
```

tb-killa, 2018-08-28:
seems to be not complete .. please copy raw content.

TheCellMC, 2018-08-28:
This is the raw content which I got with `nano`. 
I've tried deleting the contents from this file and my sensor from the configuration file but the same database is generated.

tb-killa, 2018-08-28:
hmm strange because it seems to be snipped because the last word isnt finished like ``joinTim$`` which should be ``"joinTime":`` and so on ... maybe use ``cat`` to file and copy raw content from screen.

TheCellMC, 2018-08-28:
Sorry, you were correct. The nano editor was cutting off the end of the lines.
Here is the full database.db: https://pastebin.com/mpzLCJsM

tb-killa, 2018-08-28:
great thanks ... now we could check ...

TheCellMC, 2018-08-28:
Sure thing.
So here is the full log when I startup the service: https://pastebin.com/j53N0wSy
And here is what I get when I trigger the sensor: https://pastebin.com/8mxe9Vy9
It seems that it is now asking for a converter to be written.

TheCellMC, 2018-08-28:
I'm also getting these from time to time:
```
  zigbee2mqtt:debug 2018-8-28 18:50:00 Recieved zigbee message with data {"cid":"msTemperatureMeasurement","data":{"measuredValue":2262}}
  zigbee2mqtt:warn 2018-8-28 18:50:00 No converter available for '9290012607' with cid 'msTemperatureMeasurement', type 'devChange' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2262}}'
  zigbee2mqtt:warn 2018-8-28 18:50:00 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-8-28 18:50:36 Saving state to file /opt/zigbee2mqtt/data/state.json
```

tb-killa, 2018-08-28:
For this 
``
zigbee2mqtt:warn 2018-8-28 18:45:43 No converter available for '9290012607' with cid 'genPowerCfg', type 'attReport' and data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":200}}'
``
we could use actually this converter ``fz._324131092621_power`` -> converter should be renamed to something generally.

For this
``
zigbee2mqtt:warn 2018-8-28 18:45:45 No converter available for '9290012607' with cid 'msOccupancySensing', type 'attReport' and data '{"cid":"msOccupancySensing","data":{"occupancy":0}}'
``
we could check with this converter ``fz.xiaomi_occupancy`` maybe this could work here too ;) ??

For this
``
zigbee2mqtt:warn 2018-8-28 18:45:46 No converter available for '9290012607' with cid 'msTemperatureMeasurement', type 'devChange' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2248}}'
``
we could check with this converter ``fz.xiaomi_temperature`` ..

so finally please replace the line with this 
``
fromZigbee: [fz._324131092621_power,fz.xiaomi_occupancy,fz.xiaomi_occupancy],
``

and report with debug again ...

TheCellMC, 2018-08-28:
Start log: https://pastebin.com/wm2X5DUd
On motion: https://pastebin.com/75sr2Y0M

TheCellMC, 2018-08-28:
I'm still getting this also:
```
  zigbee2mqtt:debug 2018-8-28 19:19:55 Recieved zigbee message with data {"cid":"msTemperatureMeasurement","data":{"measuredValue":2262}}
  zigbee2mqtt:warn 2018-8-28 19:19:55 No converter available for '9290012607' with cid 'msTemperatureMeasurement', type 'devChange' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2262}}'
  zigbee2mqtt:warn 2018-8-28 19:19:55 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
```

TheCellMC, 2018-08-28:
Start log: https://pastebin.com/SZxNQx4N
On motion:  https://pastebin.com/BHJg9xdH
Still no lux reporting:
`zigbee2mqtt/0x001788010212dcd0 {"occupancy":true,"linkquality":126,"battery":100}`

tb-killa, 2018-08-28:
Do you use my edited fromzigbee part ?
I see you get the ``devChange`` Messages .. this should be removed with ``ignore_occupancy_change`` ..

Because of lux reporting we have to check what we could do on reporting ..
Maybe @Koenkk could help faster ...

TheCellMC, 2018-08-28:
Oh, temperature is now coming in:
`zigbee2mqtt/0x001788010212dcd0 {"occupancy":false,"linkquality":115,"battery":100,"temperature":22.76}`

Only lux reporting is left to be implemented.

On another note, how will I use my sensor with HA as it publishes TRUE every time it is triggered but it never publishes FALSE.

TheCellMC, 2018-08-28:
I'm getting this error on motion:
```
  zigbee2mqtt:warn 2018-8-28 19:41:59 Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices.
  zigbee2mqtt:debug 2018-8-28 19:42:00 Recieved zigbee message with data {"cid":"msOccupancySensing","data":{"occupancy":1}}
```

tb-killa, 2018-08-28:
I don´t use HA so we have to ask somebody who know this.
Do you see in log the return value ``FALSE`` ?
We reset the state after 
`` const occupancyTimeout = 60; // In seconds ``

We could do some test for lux
please add 
``
(cb) => device.report('msIlluminanceMeasurement', 'measuredValue', 0, 600, null, cb),
``
before ``];`` and restart via DEBUG ..

Maybe we find your mentioned error too ;)

TheCellMC, 2018-08-28:
I added `occupancy_timeout: 15` to fix the FALSE issue.
So this is what I have in my configuration.yaml:
```
devices:
  '0x001788010212dcd0':
    friendly_name: 'Kitchen Motion Sensor'
    retain: false
    occupancy_timeout: 15

```
Here's the full log: https://pastebin.com/Tck2rDS9
Here's the log on motion: https://pastebin.com/tGE86kXG

I have to add that the led on the sensor flashes red every time motion sensor. Normally, with the Hue hub, that led never turns on when motion is detected unless it is unable to connect to the bridge.

tb-killa, 2018-08-28:
Okay great ..
now please change again with this:
```
fromZigbee: [fz._324131092621_power,fz.xiaomi_occupancy,fz.xiaomi_temperature,fz.ignore_occupancy_change,fz.xiaomi_illuminance,fz.ignore_illuminance_change],
```
and please provide full debug logging ..
maybe we finish it ;) ?

TheCellMC, 2018-08-28:
Here's the full log: https://pastebin.com/Jkn4gWA4

TheCellMC, 2018-08-28:
I triggered the sensor again and I just got an illuminance report: https://pastebin.com/KRqY1hv2

tb-killa, 2018-08-28:
Okay see some more converter messages .. fix this via:
```
fromZigbee: [fz._324131092621_power,fz.xiaomi_occupancy,fz.xiaomi_temperature,fz.ignore_occupancy_change,fz.xiaomi_illuminance,fz.ignore_illuminance_change,fz.ignore_temperature_change],
```

TheCellMC, 2018-08-28:
Philips says that the red led indicates that :

> Sensor cannot reach bridge (if this happens often, please consider a Zigbee channel change using the Philips Hue app (SettingsHue bridgesZigbee)

tb-killa, 2018-08-28:
log seems to be good .. doesn´t see any error or warning messages ...

maybe the red led comes up because you doesn´t use original hub .. 
i would say .. check with actually config .. if everything seems fine .. let it be ..
if there are some errors or no motion is triggered .. we should show again ..

I'm leaving, tomorrow is another stressful day :)

TheCellMC, 2018-08-28:
Thanks for your help. I really appreciate the time you spent getting this to work.
I'll let this run overnight to see if any more errors arise.
Do you want to submit the PR or should I do it ?

closed 2018-08-28 by TheCellMC

tb-killa, 2018-08-28:
If you would .. please provide PR .. if somethings doesn´t work .. i could do it .. let me know tomorrow if everything is great :)

TheCellMC, 2018-08-29:
Small question: the sensor normally comes with a setting to adjust the sensitivity of the PIR sensor. Should this is implemented before submitting the PR ?

tb-killa, 2018-08-29:
Hi we could check if we got this too.
Need to Figure out how we could do this...
You could provide base PR and we provide extended too later if this extension work...