Koenkk/zigbee2mqtt #311: serial_port attribute no accepting devices passed as /dev/serial/by-id/

marcgarciamarti, 2018-08-27:
When attempting to limit how hassio (the host actually) handles USB devices, I tried to follow the suggestion found here: https://github.com/home-assistant/hassio/issues/261#issuecomment-416268012 
Unfortunately the add-on won’t accept it:
```
zigbee2mqtt:error 2018-8-27 21:31:04 Failed to start
	{"message":"Error: No such file or directory, cannot open /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0012022476-if00","stack":"Error: Error: No such file or directory, cannot open /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0012022476-if00"} 
```

This is what I see when I log into the host:

```
~ # cd /dev/serial/by-id/
/dev/serial/by-id # ls -l
lrwxrwxrwx    1 root     root            13 Aug 26 21:11 usb-0658_0200-if00 -> ../../ttyACM0
lrwxrwxrwx    1 root     root            13 Aug 26 21:34 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0012022476-if00 -> ../../ttyACM1
```

Could you please check if the add-on is limit the syntax of the parameter "serial_port" so that /dev/serial/by-id/ is not accepted?

Thanks

Koenkk, 2018-08-28:
What happens if you execute the `ls` command in the container?

marcgarciamarti, 2018-08-28:
Hi there @Koenkk not sure what do you mean as the code I reported earlier does show an `ls -l` output. I may be confused with some concepts so please try to be more specific on what I should do to help in the troubleshooting process

epeome5, 2018-08-28:
@Koenkk and @marcgarciamarti I think it only works when you SSH into the Host, not when you SSH into the container. At lest I just tried SSHd into the container and it answered "No such file or directory".

I have in my hassios configuration.yaml file added:
```
zwave: 
  usb_path: /dev/serial/by-id/usb-0658_0200-if00
  network_key: !secret zwave_network_key

rflink:
  port: /dev/serial/by-id/usb-Arduino__www.arduino.cc__0042_557353233353513080A1-if00
```

Which works great, and as far as I know that's inside of the container as well. Maybe it's some functionality in Hassio which makes it possible for the container to see the serial/by-id address from the host?

Koenkk, 2018-08-29:
@marcgarciamarti what you need to do is:

```
docker ps -a # to find out the ID of the zigbee2mqtt containers
docker exec -it [ZIGBEE2MQTT_CONTAINER_ID]
ls -l /dev/serial/by-id
```

marcgarciamarti, 2018-08-29:
thanks @Koenkk and sorry for being a bit newbe with containers and stuff. Your command results in this:

```
~ # docker exec -it ea5c97a93ad2 /bin/ls -l /dev/serial/by-id
ls: /dev/serial/by-id: No such file or directory
```
however, if I go back one level, this is what I see 

```
~ # docker ps -a
CONTAINER ID        IMAGE                                         COMMAND                  CREATED             STATUS                    PORTS                                                                NAMES
4fa118bce970        homeassistant/aarch64-hassio-supervisor       "python3 -m hassio"      6 hours ago         Up 6 hours                                                                                     hassos_supervisor
ea5c97a93ad2        dwelch2101/zigbee2mqtt-aarch64                "./run.sh"               2 days ago          Up 2 days                                                                                      addon_7ad98f9c_zigbee2mqtt
253df2a3f200        homeassistant/aarch64-addon-ssh               "/run.sh"                2 days ago          Up 2 days                 0.0.0.0:22->22/tcp                                                   addon_core_ssh
8a655e889d8d        homeassistant/raspberrypi3-64-homeassistant   "/bin/entry.sh pytho…"   2 days ago          Up 2 days                                                                                      homeassistant
3df813ed1720        homeassistant/aarch64-addon-mariadb           "/usr/bin/entry.sh /…"   2 days ago          Up 2 days                 0.0.0.0:3306->3306/tcp                                               addon_core_mariadb
f62f1df2829d        homeassistant/aarch64-addon-mosquitto         "/run.sh"                2 days ago          Up 2 days                 0.0.0.0:1883-1884->1883-1884/tcp, 0.0.0.0:8883-8884->8883-8884/tcp   addon_core_mosquitto
a6647fd3fc70        homeassistant/aarch64-addon-letsencrypt       "/usr/bin/entry.sh /…"   5 days ago          Exited (1) 5 days ago                                                                          addon_core_letsencrypt
fc8c23a4890b        hassioaddons/node-red-aarch64                 "/init"                  7 days ago          Exited (129) 7 days ago                                                                        addon_a0d7b954_nodered
~ # docker exec -it ea5c97a93ad2 /bin/ls -l /dev/
total 466
lrwxrwxrwx    1 root     root            13 Aug 27 21:47 fd -> /proc/self/fd
crw-rw-rw-    1 root     root        1,   7 Aug 27 21:47 full
-rwxr-xr-x    1 root     root        476800 Jul 16 00:55 init
drwxrwxrwt    2 root     root            40 Aug 27 21:47 mqueue
crw-rw-rw-    1 root     root        1,   3 Aug 27 21:47 null
lrwxrwxrwx    1 root     root             8 Aug 27 21:47 ptmx -> pts/ptmx
drwxr-xr-x    2 root     root             0 Aug 27 21:47 pts
crw-rw-rw-    1 root     root        1,   8 Aug 27 21:47 random
drwxrwxrwt    2 root     root            40 Aug 27 21:47 shm
lrwxrwxrwx    1 root     root            15 Aug 27 21:47 stderr -> /proc/self/fd/2
lrwxrwxrwx    1 root     root            15 Aug 27 21:47 stdin -> /proc/self/fd/0
lrwxrwxrwx    1 root     root            15 Aug 27 21:47 stdout -> /proc/self/fd/1
crw-rw-rw-    1 root     root        5,   0 Aug 27 21:47 tty
crw-rw----    1 root     audio     166,   0 Aug 27 21:47 ttyACM0
crw-rw----    1 root     audio     166,   1 Aug 29 21:38 ttyACM1
crw-rw----    1 root     audio     204,  64 Aug 27 21:47 ttyAMA0
crw-rw-rw-    1 root     root        1,   9 Aug 27 21:47 urandom
crw-rw-rw-    1 root     root        1,   5 Aug 27 21:47 zero

```
apparently the docker can't see the devices available in the host but rather the ttyACM* right?

Big shoutout to @epeome5 for stepping in and for providing some additional insights. The big difference between your scenario and ours is that you are using HA native components, while zigbee2mqtt is an add-on produced by Koenkk and the z2mqtt team.

Koenkk, 2018-08-30:
So thats the problem. @danielwelch could it be possible to also map this with the hassio addon?

ciotlosm, 2018-09-03:
@marcgarciamarti can you please open an issue against https://github.com/danielwelch/hassio-zigbee2mqtt/issues ?

marcgarciamarti, 2018-09-05:
done @ciotlosm

closed 2018-12-22 by stale[bot]