Koenkk/zigbee2mqtt #315: Aqara cube MFKZQ01LM not supported

Skysurfer-14, 2018-08-29:
I tried to join my new Aqara cube to my Hassio with zigbee2mqtt. It joined, but is not supported:
`Device with modelID 'lumi.sensor_cube.aqgl01' is not supported.`
I have the newest versions of zigbee2mqtt (V 0.1.3) and the newest Firmware on the CC2531 Stick.
The Aqara Cube MFKZQ01LM is on the list of supported devices, but did not work at me.

tb-killa, 2018-08-30:
This seems to be new lumi version (2017), we supported actually only the older aquara version.
If you´v access to your files you could check if it would help if you modify
``node_modules/zigbee-shepherd-converters/devices.js`` with the entry 
``zigbeeModel: ['lumi.sensor_cube'],`` to 
`` zigbeeModel: ['lumi.sensor_cube', 'lumi.sensor_cube.aqgl01'], ``
restart your instances and show .. if failure ... please provide log-files (Debug).

Skysurfer-14, 2018-08-30:
Thank you for your help! But I don't know where zigbee-shepherd-converters is installed or running. I have Hass.io on Raspi installed with zigbee2mqtt as addon. When I ssh on the Raspi, I can't find any file or directory called zigbee-shepherd-converters. 
Is it possible to install zigbee-shepherd-converters on docker of my synology nas? And how can I transfer the modification on my Raspi?

AnhDuc85, 2018-09-19:
Hi,
I replaced the zigbeeModel as you mentioned above, I paired the cube after restart, please check the logs: https://pastebin.com/raw/FaCXav7E

AnhDuc85, 2018-09-19:
@ciotlosm Just flashed firmware 20180815 to my CC2531 and even the "old" cube doesn't work anymore. The cube worked fine under 20180705 no matter which docker images has been used. How can it be?

HKC001, 2018-10-01:
Hi, I think I've the new model as well, but the error message is slightly different:

```
Device with modelID '0x656275635f726f736e65732e696d756c' is not supported
```

There's no way for a Hass.IO user to edit the .js files right? If there is, let me know. I'd like to try and help get this added to the list of supported devices.

AnhDuc85, 2018-10-01:
please check #388

loglux, 2018-10-31:
I hope you find a way to provide a support for Aqara Cube

closed 2019-01-07 by stale[bot]

csloz, 2019-03-27:
Having issues getting the aqara cube to pass on messages. 
Successfully paired the device with the cc2531 (flashed with latest firmware)
Detected as a hex id, have edited configuration.yaml to map it to the right device in devices.js (lumi.sensor_cube)

devices:  
  '0x00158d000275dc3c':
    friendly_name: 'lumi.sensor_cube'
    retain: true

Logs giving me errors about no converter available.
If I check the converters I can see that it should be supported.
i.e. in devices.js it uses the following converters for lumi.sensor_cube

fromZigbee: [
            fz.xiaomi_battery_3v, fz.MFKZQ01LM_action_multistate, fz.MFKZQ01LM_action_analog,
            fz.ignore_analog_change, fz.ignore_multistate_change, fz.ignore_basic_change,
        ],

In 
https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js
I can see that  fz.MFKZQ01LM_action_multistate should match for this - 

 zigbee2mqtt:debug 3/27/2019, 4:01:40 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":2}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)

...but it isn't.

Any suggestions?

Logs - 

zigbee2mqtt:info 3/27/2019, 4:01:11 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 3/27/2019, 4:01:40 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":2}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)
  zigbee2mqtt:warn 3/27/2019, 4:01:40 PM No converter available for 'MFKZQ01LM' with cid 'genMultistateInput', type 'attReport' and data '{"cid":"genMultistateInput","data":{"presentValue":2}}'
  zigbee2mqtt:warn 3/27/2019, 4:01:40 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 3/27/2019, 4:01:41 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":133}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)
  zigbee2mqtt:warn 3/27/2019, 4:01:41 PM No converter available for 'MFKZQ01LM' with cid 'genMultistateInput', type 'attReport' and data '{"cid":"genMultistateInput","data":{"presentValue":133}}'
  zigbee2mqtt:warn 3/27/2019, 4:01:41 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 3/27/2019, 4:01:41 PM Received zigbee message of type 'devChange' with data '{"cid":"genMultistateInput","data":{"presentValue":133}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)
  zigbee2mqtt:debug 3/27/2019, 4:01:58 PM Saving state to file /app/data/state.json
  zigbee2mqtt:debug 3/27/2019, 4:02:10 PM Received zigbee message of type 'attReport' with data '{"cid":"genAnalogInput","data":{"65285":500,"presentValue":15.029999732971191}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)
  zigbee2mqtt:warn 3/27/2019, 4:02:10 PM No converter available for 'MFKZQ01LM' with cid 'genAnalogInput', type 'attReport' and data '{"cid":"genAnalogInput","data":{"65285":500,"presentValue":15.029999732971191}}'
  zigbee2mqtt:warn 3/27/2019, 4:02:10 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 3/27/2019, 4:02:10 PM Received zigbee message of type 'devChange' with data '{"cid":"genAnalogInput","data":{"presentValue":15.029999732971191}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)
  zigbee2mqtt:debug 3/27/2019, 4:06:58 PM Saving state to file /app/data/state.json