Koenkk/zigbee2mqtt #317: Zigbee-shepherd stops and fails to restart after working for a while

umar-kh, 2018-08-29:
I have CC2531 which picks up my Xiaomi Wireless Switch. After a few hours however, zigbee2mqtt stops working (and the LED on the CC2531 turns off). To fix this I have to unplug the USB and restart zigbee2mqtt. 

I have zigbee2mqtt running in a docker container. Everything runs fine except for this seemingly random and bizarre behaviour. It tends to after a couple of hours or overnight. I am on the latest releases for both zigbee2mqtt (0.1.3) and the CC2531 firmware.

This is towards the end of my log.

` zigbee2mqtt:info 2018-8-29 16:59:18 MQTT publish, topic: 'zigbee2mqtt/Xiaomi Switch', payload: '{"battery":"100.00","voltage":3082,"linkquality":97}',
  zigbee2mqtt:warn 2018-8-29 17:59:25 Soft resetted zigbee,
  zigbee2mqtt:info 2018-8-29 17:59:25 zigbee-shepherd ready,
  zigbee2mqtt:info 2018-8-29 17:59:31 MQTT publish, topic: 'zigbee2mqtt/Xiaomi Switch', payload: '{"battery":"100.00","voltage":3082,"linkquality":94}',
  zigbee2mqtt:warn 2018-8-29 18:59:36 Soft resetted zigbee,
  zigbee2mqtt:info 2018-8-29 18:59:36 zigbee-shepherd ready,
  zigbee2mqtt:info 2018-8-29 18:59:42 MQTT publish, topic: 'zigbee2mqtt/Xiaomi Switch', payload: '{"battery":"100.00","voltage":3082,"linkquality":94}',
  zigbee2mqtt:warn 2018-8-29 19:59:47 Soft resetted zigbee,
  zigbee2mqtt:info 2018-8-29 19:59:47 zigbee-shepherd ready,
  zigbee2mqtt:info 2018-8-29 19:59:56 MQTT publish, topic: 'zigbee2mqtt/Xiaomi Switch', payload: '{"battery":"100.00","voltage":3082,"linkquality":94}',
  zigbee2mqtt:warn 2018-8-29 21:00:04 Soft reset error,
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:255:22)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at Timeout.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:234:18)\n    at ontimeout (timers.js:475:11)\n    at tryOnTimeout (timers.js:310:5)\n    at Timer.listOnTimeout (timers.js:270:5)"},
  zigbee2mqtt:info 2018-8-29 21:00:04 zigbee-shepherd stopped,
  zigbee2mqtt:warn 2018-8-29 21:00:04 Zigbee stopped,
  zigbee2mqtt:info 2018-8-29 21:00:04 Starting zigbee-shepherd,
  zigbee2mqtt:info 2018-8-29 21:00:04 Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds),
  zigbee2mqtt:info 2018-8-29 21:01:04 Starting zigbee-shepherd,
  zigbee2mqtt:error 2018-8-29 21:01:10 Error while starting zigbee-shepherd!,
  zigbee2mqtt:error 2018-8-29 21:01:10 Press the reset button on the stick (the one closest to the USB) and start again,
  zigbee2mqtt:error 2018-8-29 21:01:10 Failed to restart!`

It looks like every hour the zigbee-shepherd is restarted then eventually doesn't restart. I've so far tried running the zigbee2mqtt container as root to no avail.

Koenkk, 2018-08-30:
On what system are you running zigbee2mqtt and are there any other containers also accessing serial ports?

umar-kh, 2018-08-30:
I'm running Lubuntu 18.04, there aren't any other containers accessing serial ports.

Koenkk, 2018-08-30:
Can you check if this also happens when running without docker?

umar-kh, 2018-08-30:
Sure, I'll test it out and update on it soon.

umar-kh, 2018-08-31:
It looks like running zigbee2mqtt without docker doesn't cause "Failed to restart zigbee-shepherd" error. 

It would be nice to have it running in Docker. I'm guessing there might be a problem in how I've set up zigbee2mqtt in Docker? In the instance running outside of Docker, I see it has it's own "node_modules" folder, whereas the Docker one doesn't have this. The error above seems to be it timing out after "trying" to run something from the node_modules folder which might not exist where it thinks it is? I can't seem to find an /app/node_modules directory for the Docker one (or on my computer).

I'm not incredibly knowledgeable with Docker so apologies if I've messed something in understanding it.

pixiandreas, 2018-09-03:
I'm running exact the same setup, and i hove not this error. 
Are you using the docker run command from the wiki ?

umar-kh, 2018-09-06:
I've since set up zigbee2mqtt and it has been running for a while now without stopping. I also added,

```
    stdin_open: true
    tty: true
```

to my docker-compose file. These correspond to the "-it" in the docker run command in the wiki. I don't what stopped the the software from stopping and whether these made a difference, but it's working...for now.

umar-kh, 2018-09-09:
Unfortunately it seems like this problem is still prevalent. I'm still getting the soft reset error.

Koenkk, 2018-09-11:
This seems to be related to #274, could you try the fix from: https://github.com/Koenkk/zigbee2mqtt/issues/274#issuecomment-412591653

umar-kh, 2018-09-15:
@Koenkk thank you, I will try that as soon as I can.

umar-kh, 2018-09-25:
@Koenkk I applied this fix to my docker container and since starting the container I haven't needed to restart it at all!

closed 2018-12-01 by stale[bot]