Koenkk/zigbee2mqtt #321: strange behavior light with transitions

11hp72, 2018-08-30:
I noticed a some strange behavior on my zigbee lights after switching from the original hubs to zigbee2mqtt. everything works fine. But when i switch on a light with a transition value, the light first goes on to the old value and than fades to the new value. This is especially annoying when a light is turned on at night and it directly goes to full brightness and than starts to slowly fades to a nice dimmed brightness.

second thing i noticed, but this raises far less acceptation issues in the household, is that the transition value is not respected while turning off a light. Regardless of the specified transition value the light is turned off directly.

I see this behavior on any light of my lights that are a mix of Philips Hue, Ikea Tradfri and Innr lights.

Here is a full log with debug turned on
[https://pastebin.com/BZ3M5Azc](https://pastebin.com/BZ3M5Azc)

And these are the relevant mqqt messages:
zigbee2mqtt/0x00158d00020a61fb/set {"state": "ON", "transition": 2, "brightness": 200}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":77}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":77}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":89}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":144}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":200}
zigbee2mqtt/0x00158d00020a61fb/set {"state": "OFF", "transition": 2}
zigbee2mqtt/0x00158d00020a61fb {"state":"OFF","brightness":200}
zigbee2mqtt/0x00158d00020a61fb {"state":"OFF","brightness":200}
zigbee2mqtt/0x00158d00020a61fb/set {"state": "ON", "transition": 5, "brightness": 20}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":200}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":200}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":192}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":156}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":120}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":84}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":45}
zigbee2mqtt/0x00158d00020a61fb {"state":"ON","brightness":20}

ciotlosm, 2018-09-03:
@11hp72 can you try with caching turned off?

https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration

https://github.com/Koenkk/zigbee2mqtt/commit/9396bde1f3b022e0f634487d1a37d2a5127c8cb3#diff-f68567477d803b49930337bf7fe1556bR16

11hp72, 2018-09-03:
@ciotlosm I've tested your suggestion, and maybe i was to fast with my conclusion i had this problem with all lights. After setting up the zigbee network form scratch (because of some issues with end devices that stop working after a couple of days) the Ikea and Hue lights work ok. with cache_state true or false. the Innr (RB145 E14) lights however still show the same behavior. so it seems to be a problem specific for these lights.

closed 2018-09-29 by Koenkk