Koenkk/zigbee2mqtt #326: Feature Request: Add mqtt_topic to per device config

mihalski, 2018-08-31:
First of all I LOVE this project and I wish everything has a mqtt bridge as robust as this (Zwave anyone?). Thank you for your efforts!

I've run into all manner of problems with using the device entries in zigbee2mqtt to name or at least help identify devices in Home Assistant and I've concluded that it's a LOT less painful to use the entity_registry via the Home Assistant web UI. 

However, if you leave entity_id and friendly_name as the MAC of the device then the mqtt topic follows. It would be very helpful if it were possible to set an mqtt_topic per device so that rather than:

```yaml
    topic: 'zigbee2mqtt/0x00158d00019dee01'
```
I could use:
```yaml
    topic: 'zigbee2mqtt/spare_bedroom_lamp'
```
I realise that friendly_name can do this now but the MANY hours I wasted today trying to get that to work properly across multiple HASS instances and some possibly dodgy mqtt discovery (unique id's being ignored) has led me to believe that there must be a better way, and leaving naming to the entity_registry seems to be it.

Ton1965, 2018-08-31:
zigbee2mqtt already has what you want if I understand you suggestion correctly. Please have a look at the [documentation](https://github.com/Koenkk/zigbee2mqtt/wiki/Device-specific-configuration). This is configuration on zigbee2mqtt side, not on HA. For example:

```
devices:
  '0x00158d00019dee01':
    friendly_name: 'spare_bedroom_lamp'
    retain: false
```
The topics will be zigbee2mqtt/spare_bedroom_lamp just like you want.

mihalski, 2018-08-31:
As I acknowledged in the original post I know this.. But it caused 5 hours of troubleshooting to get back to a working config.

In the next few days get ready for dozens of screenshots and config entries to demonstrate the problems that changing friendly_name causes in conjunction with the entity_registry.

mihalski, 2018-09-01:
Furthermore it seems no matter what hoops you jump through, ie. restarting zigbee2mqtt, restarting HASS, turning off MQTT discovery restarting and then turning back on MQTT discovery and restarting, the only way to get a PARTIAL fix is to remove existing zigbee2mqtt entries from the entity_registry (which is no longer intended to be editable by hand) and then restarting HASS again.

This is because friendly_name in zigbee2mqtt is not just the topic, but also the entity_id and you can't have multiple entities with the same unique_id. But even when you do all this the new entity does not get re-added back to the entity_registry you lose the ability to edit the HASS friendly_name (ie the little cog is no longer there and you're stuck with the friendly name being derived from the entity_id as well as cumbersome entity_id's, for example:
```
spare_bedroom_booklight_switch
switch.spare_bedroom_booklight_switch
```
[image]

I've tried everything I can imagine over the last few hours including uninstalling the MQTT broker and reinstalling it but there appears to be no way to get the devices back and fully working in HASS after changing the zigbee2mqtt friendly_name.

mihalski, 2018-09-01:
And so here is the root of the problem: https://github.com/Koenkk/zigbee2mqtt/issues/138#issuecomment-399663947
@Ton1965 This is why my feature request makes sense.

mihalski, 2018-09-01:
Why not decouple the unique_id and friendly_name? The unique_id could be calculated as if the user had not set the friendly_name for every occasion. Seems like this would be the perfect solution.

ciotlosm, 2018-09-03:
@mihalski just to summarise as it took me a while to understand what you wanted to achieve. 

You want to be able to configure besides `friendly_name` another parameter that could be used similar to friendly name but only for MQTT topic.  

https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/homeassistant.js#L313 and probably a few other places.

mihalski, 2018-09-03:
Yes, but even more simply perhaps I'm happy to use friendly_name for the MQTT topic so long as unique_id is still available for the purposes of entity renaming using the home assistant web ui.

Could the unique_id not always be of the same format as it is if no friendly_name had been set rather than being coupled with friendly_name?

mihalski, 2018-09-06:
In what situation could this be a problem?
```node
        // Only set unique_id when user did not set a friendly_name yet,
        // see https://github.com/Koenkk/zigbee2mqtt/issues/138
        // if (deviceID === friendlyName) {
        payload.unique_id = `${deviceID}_${config.object_id}_${settings.get().mqtt.base_topic}`;
        // }
```
As far as I can tell this would ALWAYS be unique while still allowing friendly_name to act as the MQTT topic.

mihalski, 2018-09-08:
I have been using this solution and it works perfectly for me. https://github.com/Koenkk/zigbee2mqtt/pull/342

mihalski, 2018-09-09:
Not completely perfect I must admit.. But inexplicably sometimes I get no unique_id.. This SHOULD be impossible no? It's NOT optional.
```
zigbee2mqtt:info 2018-9-9 21:36:25 Connected to MQTT server
  zigbee2mqtt:info 2018-9-9 21:36:25 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-9-9 21:36:25 MQTT publish, topic: 'homeassistant/sensor/0x00158d000123df04/click/config', payload: '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","json_attributes":["battery","voltage","action","duration"],"force_update":true,"state_topic":"zigbee2mqtt/square_button_2","availability_topic":"zigbee2mqtt/bridge/state","name":"square_button_2"}'
  zigbee2mqtt:info 2018-9-9 21:36:25 MQTT publish, topic: 'homeassistant/sensor/0x00158d0002437899/temperature/config', payload: '{"unit_of_measurement":"°C","icon":"mdi:temperature-celsius","value_template":"{{ value_json.temperature }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/office_temp","availability_topic":"zigbee2mqtt/bridge/state","name":"office_temp_temperature"}'
  zigbee2mqtt:info 2018-9-9 21:36:25 MQTT publish, topic: 'homeassistant/sensor/0x00158d0002437899/humidity/config', payload: '{"unit_of_measurement":"%","icon":"mdi:water-percent","value_template":"{{ value_json.humidity }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/office_temp","availability_topic":"zigbee2mqtt/bridge/state","name":"office_temp_humidity"}'
  zigbee2mqtt:info 2018-9-9 21:36:25 MQTT publish, topic: 'homeassistant/sensor/0x00158d0002437899/pressure/config', payload: '{"unit_of_measurement":"Pa","icon":"mdi:speedometer","value_template":"{{ value_json.pressure }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/office_temp","availability_topic":"zigbee2mqtt/bridge/state","name":"office_temp_pressure"}'
```

This is running in the hassio add-on on an rpi3. It ALWAYS works perfectly on my Mac..

Anyone have any ideas?

mihalski, 2018-09-09:
On the Mac I get this:
```
zigbee2mqtt:info 9/9/2018, 9:47:18 PM Connected to MQTT server
  zigbee2mqtt:info 9/9/2018, 9:47:18 PM MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 9/9/2018, 9:47:18 PM MQTT publish, topic: 'homeassistant/sensor/0x00158d000123df04/click/config', payload: '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","json_attributes":["battery","voltage","action","duration"],"state_topic":"zigbee2mqtt/square_button_2","availability_topic":"zigbee2mqtt/bridge/state","name":"square_button_2","unique_id":"0x00158d000123df04_click_zigbee2mqtt"}'
  zigbee2mqtt:info 9/9/2018, 9:47:18 PM MQTT publish, topic: 'homeassistant/sensor/0x00158d0002437899/temperature/config', payload: '{"unit_of_measurement":"°C","icon":"mdi:temperature-celsius","value_template":"{{ value_json.temperature }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/office_temp","availability_topic":"zigbee2mqtt/bridge/state","name":"office_temp_temperature","unique_id":"0x00158d0002437899_temperature_zigbee2mqtt"}'
  zigbee2mqtt:info 9/9/2018, 9:47:18 PM MQTT publish, topic: 'homeassistant/sensor/0x00158d0002437899/humidity/config', payload: '{"unit_of_measurement":"%","icon":"mdi:water-percent","value_template":"{{ value_json.humidity }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/office_temp","availability_topic":"zigbee2mqtt/bridge/state","name":"office_temp_humidity","unique_id":"0x00158d0002437899_humidity_zigbee2mqtt"}'
  zigbee2mqtt:info 9/9/2018, 9:47:18 PM MQTT publish, topic: 'homeassistant/sensor/0x00158d0002437899/pressure/config', payload: '{"unit_of_measurement":"Pa","icon":"mdi:speedometer","value_template":"{{ value_json.pressure }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/office_temp","availability_topic":"zigbee2mqtt/bridge/state","name":"office_temp_pressure","unique_id":"0x00158d0002437899_pressure_zigbee2mqtt"}'
```

Koenkk, 2018-09-10:
The problem with setting the `unique_id` is that the `friendly_name` is ignored as an entity name. This leads to confusion therefore this has been disabled.

mihalski, 2018-09-10:
Could the zigbee2mqtt friendly_name therefore not be changed to mqtt_name (or similar). It feels like it has been misnamed relative to what friendly_name is within Home Assistant.

I'm curious how people are using this feature as for me it is only to change the mqtt topic to be human readable. I find the integrated Home Assistant functionality for renaming both the entity_id and friendly_name using the interface to be much more compelling than setting it via the zigbee2mqtt configuration.yaml.

mihalski, 2018-09-10:
Or maybe just this in the configuration.yaml?
```yaml
force_unique_id: true
```

Koenkk, 2018-09-11:
The flow I use is to set the zigbee2mqtt `friendly_name`  (e.g. `temperature_bedroom`) and rename the entity in homeassistant using the `customize.yaml`. (e.g. `Temperature bedroom`).

Koenkk, 2018-09-11:
@ciotlosm as you have more homeassistant knowledge, what is your view about this?

mihalski, 2018-09-11:
The problem with using customize.yaml is that you cannot change the entity_id like you can when using the entity registry. So you are stuck with entities like switch.spare_bedroom_booklight_switch.

stoinov, 2019-03-04:
@Koenkk and @ciotlosm - #470 allowed to have `/` in the `friendly_name` parameter in order to accommodate for the need of MQTT subscription based on topics (for instance distinguishing by location or room). But this is now causing unnecessary overhead when used in combination with HA discovery. Here's my example:
```
mqtt:
  base_topic: places/location
...
devices:
  '0x00158d0000000000':
    friendly_name: room/type/name
```
This allows me to filter in MQTT based on type (places, persons), location (home, villa), room, type of device (sensor, switch...).
As you can see `base_topic` only makes sense on location level, and the `friendly_name` is handling the further distinguishing.
While this is working fine, the `room/type/name` string is passed to HA during discovery and is seen in multiple places which is not great experience.

I am not sure what a good name for such property should be - the request is for `mqtt_topic` but it seems redundant so `device_topic` or just `topic` looks more descriptive and to the point.

Another change that seems logical is to stop adding `friendly_name` to the discovery parameter `name`  and leave it only for the `device\name`. My [understanding is](https://developers.home-assistant.io/docs/en/device_registry_index.html) that the entity is displayed as a child object to the device, and adding `friendly_name_` is just redundant and visually cluttering.

Koenkk, 2019-03-04:
@stoinov as far as I have seen, having the `friendly_name` is actual useful, as this allows you discover the zigbee devices with the correct name in home assistant (in the sensors) (multiple users are using it in this way).

stoinov, 2019-03-04:
@Koenkk I assumed as much, but don't actually use HA personally at the moment and can't say how hard the search would be with such change. But this was not the main request here - more like a side question. I'm OK with it either way.

But allowing for extending of the topic would really benefit the project long-term, and will be fully backward compatible for anyone that don't use the new property.

Koenkk, 2019-03-04:
@stoinov I agree, reopening this.

stoinov, 2019-03-10:
using #1030, I am able to workaround through this using similar config:
```
devices:
  '0x00158d0000000000':
    friendly_name: room/device/sensor
    retain: false
    homeassistant:
      device:
        identifiers: '0x00158d0000000000'
        name: 'Room Sensor'
      temperature:
        name: 'Room Sensor Temperature'
        unique_id: room_sensor_temperature'
```
It's still nice to have this feature as it can save a lot of typing, but at least we can set it properly now.

Fenix999, 2019-03-30:
It would be nice to add a mqtt topic per device to have a standardization on mqtt.

closed 2019-06-05 by stale[bot]