Koenkk/zigbee2mqtt #327: Xiaomi WSDCGQ01LM Update Frequence

evildad, 2018-08-31:
Hi, 

according to the Xiaomi MiHome App the Temperature and Humidity Sensor updates data

a) if the temperature exceeds 0,5°C or the humidity variation exceeds 6%
b) if the temperature and humidity varies a bit only data will be reported once an hour.

I've migrated a few WSDCGQ01LM Sensors to zigbee2mqtt and now only a) is working as expected.
If you hold the Sensor in your hand it reports the data immediately if the change in temperature is > 0,5°C
It also reports the data every hour, but this seems to be the wrong data as it represents only the data from a)

For example:
`2018-8-31 10:42:05 - info: MQTT publish, topic: 'zigbee2mqtt/living_temp', payload: '{"temperature":19.72,"linkquality":31,"humidity":53.17,"battery":"100.00","voltage":3205}'
2018-8-31 11:41:21 - info: MQTT publish, topic: 'zigbee2mqtt/living_temp', payload: '{"temperature":19.72,"linkquality":28,"humidity":53.17,"battery":"100.00","voltage":3205}'
2018-8-31 12:38:31 - info: MQTT publish, topic: 'zigbee2mqtt/living_temp', payload: '{"temperature":19.72,"linkquality":26,"humidity":53.17,"battery":"100.00","voltage":3205}''`

It looks like the temperature and humidity don't have changed > 0,5°C or > 6% but there seems to be a genBasic report which represents the actual changed data:

The Data reported from 11:41:21 looks like this:

`2018-8-31 11:41:21 - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3205,"4":5032,"5":14,"6":[0,2],"10":43665,"100":1972,"101":5501}}}
2018-8-31 11:41:21 - info: MQTT publish, topic: 'zigbee2mqtt/living_temp', payload: '{"temperature":19.72,"linkquality":28,"humidity":53.17,"battery":"100.00","voltage":3205}'
2018-8-31 11:41:21 - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,3205,null,null,5032,14,[0,2],null,null,null,43665,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,1972,5501]}}
...
...
`
The reported Temperature and Humidity is : 19.72°C and 53.17% and in genBasic 19,72°C and 55,01% ( 1972,5501)

The Data from 12:38 is the following:

`2018-8-31 12:38:31 - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3205,"4":5032,"5":14,"6":[0,2],"10":43665,"100":1956,"101":5656}}}
2018-8-31 12:38:31 - info: MQTT publish, topic: 'zigbee2mqtt/living_temp', payload: '{"temperature":19.72,"linkquality":26,"humidity":53.17,"battery":"100.00","voltage":3205}'
2018-8-31 12:38:31 - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":[null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,null,1956,5656]}}
...
...
`
The reported Temperature and Humidity is : 19.72°C and 53.17% (it seems like it didn't change) but in genBasic 19,56°C and 56,56% (1956,5656) -> humidity changed from 55,01% to 56,56%

Looking at 
/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/devices.js you're skipping those genBasic reports with fz.ignore_basic_change.

Any chance we can use those real values in the genBasic report? Otherwise the Sensors in HomeAssistant seem to be stale and show only those rough changes (0,5°C and 6%)
Looking at a Sensor still running on the Xiaomi Gateway I can confirm those genBasic values seem to be correct.
The issue also applies to WSDCGQ11LM but for the simplicity I've only shown the reports for an WSDCGQ01LM Sensor.

You can check this issue quite easily, just press the pair Button and with the Aqura Gateway the actual Data shows up on HomeAssistant with zigbee2mqtt nothing happens because of the above Problem.

Thanks in advance!

evildad, 2018-09-02:
I've digged a little bit deeper and the following **ugly** hack in from Zigbee. js seems to fix it.

`xiaomi_battery_3v: {
        cid: 'genBasic',
        type: 'attReport',
        convert: (model, msg, publish, options) => {
            let voltage = null;
                    //ugly fix to get more data
            let humidity = null;
            let temperature = null;
                    //ugly fix to get more data

            if (msg.data.data['65281']) {
                voltage = msg.data.data['65281']['1'];
                    //ugly fix to get more data
                temperature =  parseFloat(msg.data.data['65281']['100']) / 100.0;
                humidity = parseFloat(msg.data.data['65281']['101']) / 100.0;
                    //ugly fix to get more data
            } else if (msg.data.data['65282']) {
                voltage = msg.data.data['65282']['1'].elmVal;
            }

            if (voltage) {
                return {
                    battery: toPercentage(voltage, battery3V.min, battery3V.max),
                    voltage: voltage,
                    //ugly fix to get more data
                    temperature : temperature,
                    humidity : humidity
                    //ugly fix to get more data
                };
            }
        },
    },
`

Clicking on the pair button immediately pushes the changed temperature/humidity

Koenkk, 2018-09-02:
Great! Could you make a PR of this by adding a new converter?

ryanbeaton, 2018-09-02:
@evildad I can see how the ugly fix code would work with the 11:41:21 data as it included the voltage but the 12:38:31 has null as the voltage so the return block would not fire. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

evildad, 2018-09-02:
the code refers to this zigbee-messages:

```
2018-8-31 11:41:21 - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3205,"4":5032,"5":14,"6":[0,2],"10":43665,"100":1972,"101":5501}}} 
````
and

```
2018-8-31 12:38:31 - debug: Recieved zigbee message with data {"cid":"genBasic","data":{"65281":{"1":3205,"4":5032,"5":14,"6":[0,2],"10":43665,"100":1956,"101":5656}}} 
````
and both contain voltage under "1".
The part with a lot of nulls can be ignored, that was my initial thought,
it came from a genBasic devChange zigbee message but parsing didn't work for me. 

The working one is a genBasic attReport zigbee message.

lolorc, 2018-09-02:
it looks like an ugly fix indeed :-)
how could this be better fixed ? it feels like other sensors might also benefit from a proper fix.

ciotlosm, 2018-09-03:
Oh my, this fix would make a lot of people happy.

closed 2018-09-12 by Koenkk

kaelsaes, 2018-09-30:
What do I have to do to use this new converter?

Koenkk, 2018-10-01:
@kaelsaes update zigbee2mqtt to the latest version.

kaelsaes, 2018-10-04:
I updated but nothing changed. Than I realized that I use the new devices WSDCGQ11LM instead of the old devices WSDCGQ01LM.

Is there also a new converter for the WSDCGQ11LM available?

evildad, 2018-10-04:
The converter is for both devices, old and new.

mclaudiopt, 2019-05-12:
hi there. i had to re-pair my sensor...now the reading show are always the same. what can i do?
thanks