Koenkk/zigbee2mqtt #328: Philips Hue Bloom not supported although on supported device list

kiwinol, 2018-09-01:
When I connect my Philips Hue Bloom I get a notification that device modelID ‘LLC011’ is not currently supported.

It is on he supported devices list:
https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-devices

Did anyone else encounter this issue, could it be that there are multiple hardware versions of the Philips Hue Bloom?

tb-killa, 2018-09-01:
I see that this type of modelID isn´t supported on zigbee2mqtt converter, only modelID ``LLC012`` ..
If you could modify your files please replace in ``devices.js`` from zigbee-shepherd-converters
``zigbeeModel: ['LLC012'],`` with ``zigbeeModel: ['LLC012','LLC012'],``
and restart your instance .. please report back..

kiwinol, 2018-09-01:
Thanks tb-killa.

I am using hassio and the docker add-on to run zigbee2mqtt.

Would the easiest option to modify the files be for me to clone the repo, modify my fork and reinstall the add-on using my fork?

kiwinol, 2018-09-18:
Hi All,

On the last HomeAssistant podcast Paulus mentioned that over half of HomeAssistant users are using Hassio this made me think about how beneficial it would be to have a process that would allow Hassio users to contribute for Zigbee2Mqtt (Including myself).  I know this has been mentioned before but I would like to document a process

Steps followed so far
1. Enabled developer access to the hassos host
    https://developers.home-assistant.io/docs/en/hassio_debugging.html
2. Opened an interactive session
	docker exec -it <CONTAINER ID/NAME> bash
3. Edited devices.js
4. Tried to do NPM restart (Not sure if it was the right thing to do)
I then got errors about access to /dev/ttyACM0 (It looked like it was in use)
I then used docker commit to create an "updated image"
I stopped the zigbee2mqtt container and tried to start my modfied container but got the same access error to /dev/ttyACM0

Any advice would be appreciated :)

kiwinol, 2018-09-30:
This is has been fixed
https://github.com/Koenkk/zigbee-shepherd-converters/pull/79

closed 2018-09-30 by kiwinol