Koenkk/zigbee2mqtt #329: AQARA WXKG02LM (new Hardware Rev)?

k3erg, 2018-09-01:
Not sure who is faster – I first got to read into this (somehow my changes to device.js don't get picked up by my zigbee2mqtt) ... so here we go with this ticket for another reincarnation of WXKG02LM:

```
{
    zigbeeModel: ['lumi.remote.b286acn01'],
    model: 'WXKG02LM',
    vendor: 'Xiaomi',
    description: 'Aqara double key wireless wall switch',
    supports: 'left, right and both click',
    fromZigbee: [],
    toZigbee: []
}
```

It would support double left click, long left click, double right click, long right click, double both click, long both click aswell.

Logs for the events:
https://pastebin.com/qprUTe0V

k3erg, 2018-09-02:
Further down the rabbit hole, it turns out the modelID really hast Unicode chars: `lumi.remote.b286acn01\u0000\u0000\u0000`... now it shows up and I can start adding coverters :) – I leave the issue open, until I come up with a solution / pullrequest for now.

=> I got the real modelID out of the data/database.db, maybe that could help others, too, to have it in the wiki.

tb-killa, 2018-09-02:
Because of the UniCode Problem .. we got it here too 
https://github.com/Koenkk/zigbee2mqtt/issues/286

BTW: Please post your ``database.db`` content ..

k3erg, 2018-09-02:
I paired the device, took the posted ieeeAddr from the log to grep the database:
`cat /opt/zigbee2mqtt/data/database.db | grep 0x00158d00027c0d08`

```{"id":13,"type":"EndDevice","ieeeAddr":"0x00158d00027c0d08","nwkAddr":51894,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.b286acn01\u0000\u0000\u0000","epList":[1,2,3],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"_id":"F74K4PsdTv7bywzA"}```

I'm testing my the converter now... had to add a new lookup table for left, right, both (inkl. long, double). Hope to post in 1-2h.

tb-killa, 2018-09-02:
Could you please provide one full run with ``DEBUG=* npm start`` and post this log via pastebin or other ?

k3erg, 2018-09-02:
Pull request is out... do I need to name the model different than "WXKG02LM"?

Log will follow shortly... new log format for me, got the check what needs to be sanitized. If you are looking for particular entries, I can post that quickly.

k3erg, 2018-09-02:
Here is the log from DEBUG=* start
https://pastebin.com/sdP5CqRW

tb-killa, 2018-09-02:
In your logfile i doesn´t see any informations of your ``lumi`` device .. search failed .
please do following: stop zigbee2mqtt , rename database.db to something else and restart zigbee2mqtt with debug=* npm start .. after that pair again your lumi device.
please report with full log from start to stop.

If search for this: ``Identified Device: { manufacturer: %s, product: %s }'``

your posted log show actually only 
``Identified Device: { manufacturer: IKEA of Sweden, product: TRADFRI bulb E27 opal 1000lm } +4ms`` ;)

cybe, 2018-09-02:
@k3erg The previous revision allegedly only supports click events for left/right/both. So this new revision sounds pretty interesting. Is there any clue on how to spot this new revision on Gearbest/Aliexpress?

k3erg, 2018-09-03:
@tb-killa 
lol, yes, because I had my new model and converters already in place.
https://paste.ee/p/eRJnJ

@cybe 
Mi mandarin is not very good, so I stick with the numbers and signs :)
It is branded as AQARA and carries the same SKU and Model Number as noted on the old box. But if you compare the two boxes the new one has references to Apples HomeKit (not very prominent, dough) on it. The new AQARA Gateway is HomeKit compatible. This led to a change in certification Policy for compatible products so the most noticeable change on the new packages is the CE and RoHS Logo.
There are also different EAN-13 Code on the boxes:
OLD: 6 970504 210049
NEW: 6 970504 210516

Not sure if this goes with the policy to directly link the product I ordered:
http://s.click.aliexpress.com/e/cpldVy5w
I order from that seller quite often, they are quite fast.

Hope this helps.

cybe, 2018-09-03:
@k3erg Much appreciated!

emiliojose91, 2018-09-22:
Hi, is this device currently working on Home Assistant using zigbee2mqtt?

I have installed Home Assistant in a rapsberry 2 and I can not find the file "device.js" to add the code. How can I configure zigbee-shepherd-converters to add the new WXKG02LM?

Thank you.

k3erg, 2018-09-23:
The code can be usually found in
```/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/devices.js```
and
```/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/converters/fromZigbee.js```

As of writing this, the working code is not yet merged, but can be found in this PR
https://github.com/Koenkk/zigbee-shepherd-converters/pull/62

emiliojose91, 2018-09-23:
> The code can be usually found in
> `/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/devices.js`
> and
> `/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/converters/fromZigbee.js`
> 
> As of writing this, the working code is not yet merged, but can be found in this PR
> [Koenkk/zigbee-shepherd-converters#62](https://github.com/Koenkk/zigbee-shepherd-converters/pull/62)

I can not find this path. I use docker, I think this may be the problem. Is there any way to modify the docker image to modify devices.js?

cybe, 2018-09-23:
@emiliojose91 Switch inside the running container first, like `docker exec -ti zigbee2mqtt sh`

emiliojose91, 2018-09-23:
@k3erg and @cybe Thanks for the answers. I do not know docker, I try to run "docker exec -ti zigbee2mqtt sh" from command line but can not find docker. The installation of Home Assistant was done following the steps of https://www.home-assistant.io/hassio/installation/ in my Raspberry 2. Do I have to install docker to execute this?

k3erg, 2018-09-26:
Code has been merged - txs for this great project!

closed 2018-09-26 by k3erg