Koenkk/zigbee2mqtt #331: Amount of devices / Routing

Jaxon77, 2018-09-03:
Fisrt of all i want to thank you for your work to make such a project possible.
Right now i am a little bit confused. As i started to migrate to the CC2531 the amount of possible direct connected devices was 43. Now we landed at 15 devices. As i now have >60 (xiaomi)devices in my house, i am not sure if this is still a good solution for me, because it requires a lot of routers and i haven't got any use for the devices with routing function. So i'd have to use CC253X for that.
Does anyone else have such an amount of devices running without problems?

Am i right that every router extends the max devices about 10 so that i have to connect 5 additional routers to get 60 possible devices?

Would it be better to have one more router as needed (in my case 6) so that the clients can connect to another router if one dies?

Would it even be possible to plug all of these CC253X routers into one usb hub if there is no need to extend the range with it?

Is there any chance that the limitation of 15 devices gets elevated in future and does this limitation also apply to the CC2530 or other devices?

At least: is there any listing which devices act as router or is it as easy as "every devices with a permanent external power supply acts as a router"?

In hope to get some clearification,
Jack

baerengraben, 2018-09-03:
Hi
I had the same question. I'm running a system with > 25 Xiaomi Devices (Door/Window-Sensors). It needed a lot of patience to got this running. But it runs now:
https://github.com/ioBroker/ioBroker.zigbee/issues/37#issuecomment-417915880

For now I'm unsure if I would run > 60 Devices this with this solution. I would probably run > 60 Devices with several Xiaomi-Hubs and block the Xiaomi-Hub via firewall to "call home".

> Am i right that every router extends the max devices about 10 so that i have to connect 5 additional routers to get 60 possible devices?

Yes. A router extends the max devices like thad.

> Is there any chance that the limitation of 15 devices gets elevated in future and does this limitation also apply to the CC2530 or other devices?

As far as I learned, there is a Memory-Limit on those cheap devices. I wouldn't count on a lot more Devices in the future (if there will be a new firmware at all).

hope this helps

ben423423n32j14e, 2018-09-03:
Z-Stack 3* support is apparently being worked on which should massively increase the number of devices to 120 if I am not mistaken.

https://github.com/Koenkk/zigbee2mqtt/issues/211

subzero79, 2018-09-04:
It is possible apparently using zigbee-shepherd to force devices to join through a router. If you manually do that, then is possible to have more devices with the current cc2531 as coordinator. Process is described https://github.com/Koenkk/zigbee2mqtt/issues/252#issuecomment-411507339

Unfortunately zigbee2mqtt has not implemented that join process yet. So apparently will have to work the zstack 3.0.

lolorc, 2018-09-04:
https://github.com/Koenkk/zigbee-shepherd/commit/0e4f387e963e69ef160b6f44e90fef026f753354 is it supposed to fix joining through a router or does it need more from zigbee2mqtt ?

Jaxon77, 2018-09-04:
@baerengraben your apprach is to span multiple zigbee networks instead a big one with routers is a 
nice approach, but i think i'll stick with a second xiaomi gateway until there is a robust Z-Stack 3x. Thanks for the info @ben423423n32j14e.

lolorc, 2018-09-24:
@Koenkk I've seen you've reverted "Allow joining through specified device.", is there something wrong with it ?

Koenkk, 2018-09-25:
@lolorc It didn't work (caused exceptions)

closed 2018-12-01 by stale[bot]