Koenkk/zigbee2mqtt #332: Unsupported device, LST001

WagnerBeccard, 2018-09-03:
Well struggleing a little bit here to do it myself. 
Thourght it cant be so hard because the newe one is supported - thourght falsly. 
Its a Gen one Lightstrip with the device IP LST001
https://www.amazon.com/Philips-Generation-LightStrip-Dimmable-Assistant/dp/B00F0W3CI0

I started with:
 devices.js

    {
        zigbeeModel: ['LST001'],
        model: '915004463301',
        vendor: 'Philips',
        description: 'Hue ambiance LightStrip',
        supports: 'll see ' ,
        fromZigbee: [], // We will add this later
        toZigbee: generic.light_onoff_brightness_colortemp.toZigbee, // Should be empty, unless device can be controlled (e.g. lights, switches)
    },
// V2 original config 
    {
        zigbeeModel: ['LST002'],
        model: '915005106701',
        vendor: 'Philips',
        description: 'Hue white and color ambiance LightStrip plus',
        supports: generic.light_onoff_brightness_colortemp_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee,
    },

LOG:
https://hastebin.com/igaroxilof.http

Maybe its because the strip isnt a gateway - or maybe mor concret - I have no Idea what I'm doing 💃

WagnerBeccard, 2018-09-11:
Neat - So far controlable with on/off  brightness and RGB color 
x&y and colortemp are'nt working - but it could be me. 

```
Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"currentHue":170,"currentSaturation":247,"currentX":9904,"currentY":9296,"enhancedCurrentHue":43703,"colorLoopStoredEnhancedHue":43703}}

No converter available for '915004463301' with cid 'lightingColorCtrl', type 'devChange' and data '{"cid":"lightingColorCtrl","data":{"currentHue":170,"currentSaturation":247,"currentX":9904,"currentY":9296,"enhancedCurrentHue":43703,"colorLoopStoredEnhancedHue":43703}}'

```
https://hastebin.com/imizozowed.http
Unluckaly I still dont have a Idea how to go further

Koenkk, 2018-09-11:
#354 added support for it, can you check if it works? Merged into the dev branch.

WagnerBeccard, 2018-09-12:
Would love to. But I dont know how to change the branch 😢 
When I just add these 9 lines, the service will not start. When I leave out supports:  and just use '' , Its starting and working with on/off brightness and RGB & xy. 
https://hastebin.com/opicedixoy.sql

closed 2018-09-13 by WagnerBeccard