Koenkk/zigbee2mqtt #333: Can't start zigbee2mqtt with npm start

slurpeemachine, 2018-09-03:
Hi Everyone,

I've been really banging my head on this one. I've been able to run zigbee2mqtt before. But on this reinstall, I get stuck when trying to run zigbee2mqtt, here's the console output I'm getting after running "npm start":
https://pastebin.com/0UZL1ZQ4

I get a different a different console output when running "sudo npm start":
https://pastebin.com/Y3xSwudU

I have no idea what is causing the issue as my setup is identical to a previous successful install of zigbee2mqtt
Thanks for your help!

subzero79, 2018-09-04:
```
Error: EACCES: permission denied, open '/opt/zigbee2mqtt/data/database.db'```

Check permissions there top to bottom. Chown the whole zigbee2mqtt as the runtime user.

On the second log, could be password wrong or there is ACL on the topic by the broker? If is mosquitto the server enable the debug there to see why is rejecting.

slurpeemachine, 2018-09-05:
Thank you both for your help, I think Koenkk partially was on the right track.
I did not realize I had skipped step 3 as I confused the zigbee2mqtt configuration.yaml with the hassbian configuration.yaml (my plan was to just restore the configuration.yaml from the previous working install I had).

I tried fixing "/opt/zigbee2mqtt/data/configuration.yaml" by putting my mqtt login info in there, but no matter what I tried, it still wouldn't work.
Thankfully, I had made an .img backup right before installing zigbee2mqtt, reinstalled it, and, this time, did not forget to configure "/opt/zigbee2mqtt/data/configuration.yaml". Seems to work now!

Thank you both for chiming in and trying to help, and to whoever sees "Error: Connection refused: Not authorized", it's probably your mqtt login info! But most importantly, and this apply to everything, MAKE BACKUPS! :)

closed 2018-09-27 by ciotlosm