Koenkk/zigbee2mqtt #336: npm install failed -> /usr/include/c++/6/limits:1598:7: internal compiler error

aymeric106, 2018-09-05:
I'm running zigbee2mqtt on a raspberry pi (openhabian distribution)

I had a old version that was working fine.
Try to update from scratch (deleted /opt/zigbee2mqtt and install using the wiki)

During npm install, after everything is downloaded, I have a error (looks like a compilation error)

make : on entre dans le répertoire « /opt/zigbee2mqtt/node_modules/serialport/build »
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from /usr/include/c++/6/bits/uniform_int_dist.h:35:0,
                 from /usr/include/c++/6/bits/stl_algo.h:66,
                 from /usr/include/c++/6/algorithm:62,
                 from ../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/usr/include/c++/6/limits:1598:7: internal compiler error: Erreur de segmentation
       min() _GLIBCXX_USE_NOEXCEPT { return __FLT_MIN__; }

full log here -> https://pastebin.com/YcgebVH1

Any idea ?

Thank you.

ciotlosm, 2018-09-27:
@aymeric106 did you check npm and node version to match correct ones? Did you delete node_modules folder?

aymeric106, 2018-09-27:
According to log on pastebin, it's 

> node -v v8.11.4

and yes i removed the folder

> rm -rf node_modules
npm install

I dont remember where but I think i saw something about  a bug in gcc (or gcc++)
searched for 'min() _GLIBCXX_USE_NOEXCEPT { return FLT_MIN; }' on google.

I tried to update gcc on the raspberry but never succeded

closed 2018-12-03 by stale[bot]