Koenkk/zigbee2mqtt #337: linkquality attribute not always set

lolorc, 2018-09-06:
In some occasion, the link attribute is not set, I don't know why.
In this example, cc-znp reports linkquality, the first mqtt message gets it, the second one doesn't.

```
  cc-znp { sof: 254,
  cc-znp   len: 25,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 6,
  cc-znp      srcaddr: 10318,
  cc-znp      srcendpoint: 3,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 68,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 607235,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 5,
  cc-znp      data: <Buffer 18 3d 0b 00 00> },
  cc-znp   fcs: 228,
  cc-znp   csum: 228 } +32ms
  cc-znp { sof: 254,
  cc-znp   len: 27,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 6,
  cc-znp      srcaddr: 10318,
  cc-znp      srcendpoint: 3,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 68,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 607248,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 7,
  cc-znp      data: <Buffer 18 e4 0a 00 00 10 00> },
  cc-znp   fcs: 63,
  cc-znp   csum: 63 } +3ms
  serialport:main _read reading +6ms
  serialport:bindings read +8ms
  serialport:unixRead Starting read +6ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 6, srcaddr: 10318, srcendpoint: 3, dstendpoint: 1, wasbroadcast: 0, linkquality: 68, securityuse: 0, timestamp: 607235, transseqnumber: 0, len: 5, data: <Buffer 18 3d 0b 00 00> } +2ms+0ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +2ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 6, srcaddr: 10318, srcendpoint: 3, dstendpoint: 1, wasbroadcast: 0, linkquality: 68, securityuse: 0, timestamp: 607248, transseqnumber: 0, len: 7, data: <Buffer 18 e4 0a 00 00 10 00> } +0ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +3ms
  serialport:unixRead waiting for readable because of code: EAGAIN +8ms
  serialport:poller Polling for "readable" +16ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +1ms
  zigbee2mqtt:debug 2018-9-6 00:24:06 Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}
  zigbee2mqtt:info 2018-9-6 00:24:06 MQTT publish, topic: 'zigbee2mqtt/prise_01', payload: '{"state":"OFF","linkquality":68}'
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 62 +8ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 10318, destendpoint: 3, srcendpoint: 1, clusterid: 6, transid: 62, options: 48, radius: 30, len: 6, data: <Buffer 00 3e 0c 00 00 f0> } +1ms1 bytes of data +19ms
  serialport:bindings write 21 bytes +19ms
  zigbee2mqtt:debug 2018-9-6 00:24:06 Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}
  zigbee2mqtt:info 2018-9-6 00:24:06 MQTT publish, topic: 'zigbee2mqtt/prise_01', payload: '{"state":"OFF"}'
```

also looking at https://github.com/Koenkk/zigbee2mqtt/commit/2ba86a13ebfc5cf880989f0476f0c6fdd646c7ca#diff-f68567477d803b49930337bf7fe1556bR316 if quality is 0 the attribute is not set.
I think it should report linkquality==0, it's a valuable information.

Koenkk, 2018-09-10:
Did you set `cache_state: false` in your `configuration.yaml`?

lolorc, 2018-09-11:
nope, no "cache_state: false"
but I've disabled sendAllCachedStates() and // Add cached state to payload code.
why would the cache be involved by the way ? cc_znp debug messages do indicate the linkquality attribute was there for the 2 zigbee message but it fails to be set for the last the mqtt message.

lolorc, 2018-09-11:
oh I see, it looks it comes from the callback https://github.com/Koenkk/zigbee2mqtt/blob/c8c8417192c8d53000183647b02c61f5bd7678ee/lib/controller.js#L513

lolorc, 2018-09-12:
ok thanks, not a big deal, I first thought it was a really issue :-)
I'm not using the cache, even though I'm using home-assistant.

closed 2018-09-12 by lolorc