Koenkk/zigbee2mqtt #338: missing mqtt messages ?

lolorc, 2018-09-06:
I'm not quite sure, but why doesn't it sends a mqtt message ?
```
  cc-znp { sof: 254,
  cc-znp   len: 45,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 6,
  cc-znp      srcaddr: 10318,
  cc-znp      srcendpoint: 3,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 68,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 607819,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 25,
  cc-znp      data: <Buffer 18 3f 01 00 00 00 10 00 00 40 00 10 01 01 40 00 21 00 00 02 40 00 21 00 00> },
  cc-znp   fcs: 204,
  cc-znp   csum: 204 } +26ms
  serialport:main _read reading +3ms
  serialport:bindings read +5ms
  serialport:unixRead Starting read +5ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 6, srcaddr: 10318, srcendpoint: 3, dstendpoint: 1, wasbroadcast: 0, linkquality: 68, securityuse: 0, timestamp: 607819, transseqnumber: 0, len: 25, data: <Buffer 18 3f 01 00 00 00 10 00 00 40 00 10 01 01 40 00 21 00 00 02 40 00 21 00 00> } +2ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +0ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +12ms
  serialport:unixRead waiting for readable because of code: EAGAIN +12ms
  serialport:poller Polling for "readable" +18ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object] +0ms
  zigbee2mqtt:debug 2018-9-6 00:24:06 Recieved zigbee message with data {"cid":"genOnOff","data":{"onOff":0}}
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 64 +744ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 10318, destendpoint: 3, srcendpoint: 1, clusterid: 6, transid: 64, options: 48, radius: 30, len: 5, data: <Buffer 00 40 00 00 00> } +2ms

```

Koenkk, 2018-09-10:
Can you update to the latest version of the dev branch and provide the logging (I've improved the logging of zigbee messages so we can further investigate this).

lolorc, 2018-09-19:
Hi,

Ok looking at my logs, I've got 4 days of debug logs to read.
I the mean time I've come across this message, wondering why zigbee2mqtt sees 2 messages, one attReport and one devChange for 1 zigbee messages:
```
Sep 16 21:10:31   serialport:main binding.read finished +648ms                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                ┤
│Sep 16 21:10:31   cc-znp { sof: 254,                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          ┤
│Sep 16 21:10:31   cc-znp   len: 27,                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           ┤
│Sep 16 21:10:31   cc-znp   type: 'AREQ',                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                      ┤
│Sep 16 21:10:31   cc-znp   subsys: 'AF',                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                      ┤
│Sep 16 21:10:31   cc-znp   cmd: 'incomingMsg',                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                ┤
│Sep 16 21:10:31   cc-znp   payload:                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           ┤
│Sep 16 21:10:31   cc-znp    { groupid: 0,                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                     ┤
│Sep 16 21:10:31   cc-znp      clusterid: 6,                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                   ┤
│Sep 16 21:10:31   cc-znp      srcaddr: 61846,                                                                                                                                                                                                                                                                                                                                               [... cut ...]                                                                                                                                                                                                                                                           ┤
│Sep 16 21:10:39   serialport:poller received "readable" +7s                                                                                                                                                                                        
```

lolorc, 2018-09-19:
first zigbee msg is without a mqtt msg and last one is getting a mqtt msg but the zigbee msg seen as both attReport & devChange:
```
Sep 17 06:53:20   serialport:main binding.read finished +19ms                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 ┤
│Sep 17 06:53:20   cc-znp { sof: 254,                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                          ┤
│Sep 17 06:53:20   cc-znp   len: 45,                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           ┤
│Sep 17 06:53:20   cc-znp   type: 'AREQ',                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                      ┤
│Sep 17 06:53:20   cc-znp   subsys: 'AF',                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                      ┤
│Sep 17 06:53:20   cc-znp   cmd: 'incomingMsg',                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                ┤
│Sep 17 06:53:20   cc-znp   payload:                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                           ┤
│Sep 17 06:53:20   cc-znp    { groupid: 0,                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                     ┤
│Sep 17 06:53:20   cc-znp      clusterid: 6,                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                   ┤
│Sep 17 06:53:20   cc-znp      srcaddr: 10318,                                                                                                                                                                                                                                                                                                                                                                                                                                  [... cut ...]                                                                                                                                                                                                                                                                                                                                            ┤
│Sep 17 06:54:00   serialport:main _write 9 bytes of data +18s                                                                                                     
```

Koenkk, 2018-09-19:
The `genOnOff` `devChange` message is ignored due to the `ignore_onoff_change`, https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L774

lolorc, 2018-09-19:
ok. any idea why this ignore_ was set up in the first place ?

Koenkk, 2018-09-20:
After `attReport` a `devChange` is always followed, this makes it unnecessary to handle the `devChange`, it would only increase the number of MQTT messages.

lolorc, 2018-09-20:
ok, thanks for the answer.

closed 2018-09-20 by lolorc