Koenkk/zigbee2mqtt #341: Only colour temperature transitions when both temperature and brightness or stated (IKEA Tradfri)

umar-kh, 2018-09-06:
I noticed when my light (which is scheduled to turn on in the morning before my alarm) would be turned on at the correct temperature but the incorrect brightness - the brightness it had previously.

```
  zigbee2mqtt:debug 2018-9-6 21:07:43 Recieved mqtt message on topic 'zigbee2mqtt/IKEA Bulb/set' 
  with data '{"state": "ON", "transition": 10, "brightness": 255, "color_temp": 250}'
  zigbee2mqtt:info 2018-9-6 21:07:43 Zigbee publish to '0x90fd9ffffe20ae3e', genOnOff - on - {} - null
  zigbee2mqtt:info 2018-9-6 21:07:43 Zigbee publish to '0x90fd9ffffe20ae3e', genLevelCtrl - moveToLevel - {"level":255,"transtime":100} - null
  zigbee2mqtt:info 2018-9-6 21:07:43 Zigbee publish to '0x90fd9ffffe20ae3e', lightingColorCtrl - moveToColorTemp - {"colortemp":250,"transtime":100} - null
  zigbee2mqtt:info 2018-9-6 21:07:43 MQTT publish, topic: 'zigbee2mqtt/IKEA Bulb', payload: '{"state":"ON","color_temp":454}'
  zigbee2mqtt:debug 2018-9-6 21:07:44 Recieved zigbee message with data {"cid":"genLevelCtrl","data":{"remainingTime":93}}
  zigbee2mqtt:info 2018-9-6 21:07:44 MQTT publish, topic: 'zigbee2mqtt/IKEA Bulb', payload: '{"state":"ON","color_temp":454}'
  zigbee2mqtt:debug 2018-9-6 21:07:44 Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"colorTemperature":250}}
  zigbee2mqtt:info 2018-9-6 21:07:44 MQTT publish, topic: 'zigbee2mqtt/IKEA Bulb', payload: '{"state":"ON","color_temp":250}'
  zigbee2mqtt:debug 2018-9-6 21:07:44 Recieved zigbee message with data {"cid":"lightingColorCtrl","data":{"remainingTime":91,"colorTemperature":250}}
  zigbee2mqtt:info 2018-9-6 21:07:44 MQTT publish, topic: 'zigbee2mqtt/IKEA Bulb', payload: '{"state":"ON","color_temp":250}'
  zigbee2mqtt:debug 2018-9-6 21:08:03 Check online IKEA Bulb 0x90fd9ffffe20ae3e

```

As you can see, zigbee2mqtt receives an MQTT message with both brightness and temperature defined, but only changes the temperature. If I were to resend this message after the temperature had settled, the brightness would change to the correct level.

Koenkk, 2018-09-07:
What is you do the same without a transition?

umar-kh, 2018-09-07:
Without the transition, it still does the same thing. When I first push the message, it changes the temperature. Then if I push the same message again, it will change the brightness.

closed 2018-12-22 by stale[bot]

berrywhite96, 2019-02-18:
@umar-kh do you solved the problem? I paired my Tradfri GU10 the first time with zigbee2mqtt, 10 times faster than with philips bridge :D, I noticed also this problem.
What I noticed:
{"brightness": 25,"transition": 3} => Working
{"color_temp": 1555,"transition": 3} => Working
{"brightness": 25, "color_temp": 1555,"transition": 3} => Only color temp changed (like already said)
{"brightness": 25, "color_temp": 1555} => Working

Problem didnt solve for me, I am on the current version 1.1.1.

kennymc-c, 2019-03-23:
You also have this problem if you connect Tradfri lamps with the Philips Hue bridge. I have heard that the commands must be sent with a short delay in between if a transition > 0 is set. Maybe this can be integrated directly into zigbee2mqtt.