Koenkk/zigbee2mqtt #343: [QUESTION] 3D Case Design

jwillmer, 2018-09-07:
I'm working on a 3D case for a Raspberry Pi Zero combined with the CC2531 USB sniffer. Would there be an interest that I link the final project (files) in the wiki of this repository? 

Below you can see how it currently looks like. At the moment I am in the process of printing it to see if all the margins are correct.

[image]

Koenkk, 2018-09-07:
Yes! Feel free to add it to the [Useful information section](https://github.com/Koenkk/zigbee2mqtt/wiki#useful-information)

jwillmer, 2018-09-07:
Prototype print of the lower half with low quality print settings:
[image]

subzero79, 2018-09-08:
nice work @jwillmer! , any tests on coverage once the case is being closed with the top ?

jwillmer, 2018-09-08:
@subzero79 I can make a few tests when I have the final print but in general I don't expect any changes. After all the CC2531 sniffer was designed with a plastic body around.

ciotlosm, 2018-09-08:
Any chance for more 3d designs with some donation?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

jwillmer, 2018-09-09:
Sure, why not. Just send me a mail with what you have in mind.

ciotlosm, 2018-09-12:
@jwillmer something similar for RPI3B+ with support for an USB mSATA SSD or something similar for an OrangePI. I bet if you do a "support case" topic and specify costs for building each to make it worth your time, more people would be happy to contribute

jwillmer, 2018-09-12:
I actually got my inspiration from this case: [Round Raspberry Pi Case](https://www.thingiverse.com/thing:2481024). The thing with the standard Raspberry Pi is that there are already [a lot of printable cases](https://www.thingiverse.com/search?q=raspberry+pi+case). If you need the space for an HDD then this case looks quite nice: [ownCloud Raspberry Pi Case + HDD](https://www.thingiverse.com/thing:1357022)

ciotlosm, 2018-09-12:
@jwillmer I use an mSATA USB enclosure on one (aluminium) and  X850 mSATA SSD hat on the other. Your suggestions look really good and I might try one even if I get extra space around.

jwillmer, 2018-09-13:
Quick update. I finally got the parts printed in a good quality and tested the margins and lock - all working as expected. Final step is now to print them with white ABS filament.

[image]
[image]
[image]
[image]

sebastianheierhoff, 2018-09-13:
This looks awesome. A case just for the USB stick itself would be great though. Has anyone found something usable or would be willing to create the 3d models?

ciotlosm, 2018-09-13:
A case for the stick already exists.

sebastianheierhoff, 2018-09-13:
Can you tell me, where I can find the case for the stick?

jwillmer, 2018-09-13:
Have a look at [thingiverse.com](https://www.thingiverse.com/search?q=CC2531)

cybe, 2018-09-13:
> Can you tell me, where I can find the case for the stick?

- https://www.thingiverse.com/thing:2947863
- https://www.thingiverse.com/thing:2803664

jwillmer, 2018-09-14:
Quick update:
I printed the parts now with white ABS. As you can see from the pictures there are a few issues left:

- The ABS transformed a little while cooling and there is a small opening on the top of the case. I might need to add a sliding connector to prevent this.
- I printed the ventilation openings without support material so they bend a little.
- You can see the parts slightly trough the ABS filament. (This could be a feature for a Zigbee logo if someone decides to let the CC2531 LED on.) Maybe I try some black colour from the inside to prevent this - I don't want to make the walls much thicker.
 
[image]
[image]
[image]
[image]

jwillmer, 2018-09-15:
I added a new page called [3D case options](https://github.com/Koenkk/zigbee2mqtt/wiki/3D-Case-Options) to the wiki. I think we can close this question/issue now.

closed 2018-09-15 by jwillmer