Koenkk/zigbee2mqtt #344: Philips dimmer switch model: 324131137411

nunoronconcouto, 2018-09-07:
I need help please...
Can i use the swithch on Homeassistant??

`zigbee2mqtt:info 2018-9-7 22:21:34 MQTT publish, topic: 'zigbee2mqtt/0x0017880103eaf47a', payload: '{"state":"ON","brightness":178}'`

` zigbee2mqtt:info 2018-9-7 22:26:12 MQTT publish, topic: 'zigbee2mqtt/0x0017880103eaf47a', payload: '{"state":"OFF","brightness":178}'`

nunoronconcouto, 2018-09-09:
```
zigbee2mqtt:info 2018-9-9 23:06:25 0x00178801033b6f89 (0x00178801033b6f89): 8718696449691 - Philips Hue White Single bulb B22 (Router)
zigbee2mqtt:info 2018-9-9 23:06:25 0x0017880103a510d5 (0x0017880103a510d5): 324131092621 - Philips Hue dimmer switch (EndDevice)

```

clockbrain, 2018-09-10:
Yes you can.

Recommend you edit the zigbee2mqtt configuration.yaml file to add friendly names for your 2 devices 
 as per the wiki https://github.com/Koenkk/zigbee2mqtt/wiki/Device-specific-configuration

Then set up an automation in home assistant. Choose Configuration -> Automation then add a new automation.

Name: Turn on light

Triggers
Trigger type: MQTT
Topic zigbee2mqtt/_Philips dimmer_

Conditions
Condition type: Template
Value template: {{ 'on' in trigger.payload }}

Actions
Action type: Call service
Service: switch.turn_on
Service data: {"entity_id": "_switch.philips_b22_"}

Set up another similar automation for off using service switch.turn_off

nunoronconcouto, 2018-09-10:
@clockbrain Thx for you help ... i will try it ...
And can you help me with dimmer(brightness) function???

clockbrain, 2018-12-02:
@nunoronconcouto sorry for the delay but real life and my limited understanding of home assistant got in the way. I've now worked out a simple coupling between hue dimmer switch and hue light bulb via zigbee2mqtt using home assistant automations that does what you are seeking.

The 4 buttons on the hue dimmer switch function as labelled: on, increase brightness, decrease brightness, and off. On will restore the previous level of brightness.

Ignore the automation I posted earlier, it has some side effects.

You will need 3 separate automations, one for each of on, off and dimmer. Only the on and off can be set up via the HA automations front end though. The dimmer automation needs to be entered directly into automations.yaml.

Here is the turn on automation as you would enter it in home assistant automations front end.

```
Name: Turn on test light
Trigger type: State
Entity: sensor.back_dimmer
To: on
Action type: Call service
Service: light.turn_on
Service_data: {
 "entity_id": "light.test_light" 
}
```

Alternatively here is how the same turn on automation would be entered in automations.yaml. Of course your id value would be different. Create the automation in the front end then edit in automations.yaml if needed.

```
- id: '1536836281688'
  alias: Turn on test light
  trigger:
  - entity_id: sensor.back_dimmer
    platform: state
    to: 'on'
  condition: []
  action:
  - data:
      entity_id: light.test_light
    service: light.turn_on
```

Here is the corresponding turn off automation.

```
- id: '1536836697541'
  alias: Turn off test light
  trigger:
  - entity_id: sensor.back_dimmer
    platform: state
    to: 'off'
  condition: []
  action:
  - data:
      entity_id: light.test_light
    service: light.turn_off
```

Here is the dimmer automation in automations.yaml

```
- id: '1538304874860'
  alias: Set test brightness
  trigger:
  - entity_id: sensor.back_dimmer
    platform: state
  condition:
    condition: template
    value_template: >
      {% if is_state('sensor.back_dimmer','up-press') or is_state('sensor.back_dimmer','down-press') %}
        True
      {% else %}
        False
      {% endif %}
  action:
  - data_template:
      brightness: '{{ states.sensor.back_dimmer.attributes.brightness | int }}'
      entity_id: light.test_light
    service: light.turn_on
```

This was the tricky one. First note that the brightness attribute is only available when state is up-press or down-press hence the condition. Also, the data template in the action can't be input via the automations front end. Seems to be a limitation of the automations front end editor.

Hope this helps. I certainly learned heaps about home assistant automations in trying to get it working fairly succinctly.

closed 2019-02-08 by stale[bot]

marcotini, 2019-02-10:
> @nunoronconcouto
> ```
> - id: '1538304874860'
>   alias: Set test brightness
>   trigger:
>   - entity_id: sensor.back_dimmer
>     platform: state
>   condition:
>     condition: template
>     value_template: >
>       {% if is_state('sensor.back_dimmer','up-press') or is_state('sensor.back_dimmer','down-press') %}
>         True
>       {% else %}
>         False
>       {% endif %}
>   action:
>   - data_template:
>       brightness: '{{ states.sensor.back_dimmer.attributes.brightness | int }}'
>       entity_id: light.test_light
>     service: light.turn_on
> ```

This automation is not working. When I try to check automation.yaml I get (x001234_action is my switch):

`Invalid config for [automation]: invalid template (TemplateSyntaxError: expected token 'end of print statement', got 'x001234_action') for dictionary value @ data['action'][0]['data_template']['brightness']. Got None. (See /config/configuration.yaml, line 28). Please check the docs at https://home-assistant.io/components/automation/`