Koenkk/zigbee2mqtt #345: CC2630 support

aleincsharp, 2018-09-07:
Hello,

ive got this device: https://de.aliexpress.com/item/USB-to-ZigBee-wireless-module-1-6-km-transmission-CC2630-chip-far-exceeding-CC2530/32893648941.html

Can i use it with this project?

Koenkk, 2018-09-10:
At the moment not, but as this is a TI device which supports Z-Stack, I think it's a matter of compiling the correct firmware.

aleincsharp, 2018-09-10:
Thank you for your reply. Could you point me to the right direction how to build the correct firmware?

ben423423n32j14e, 2018-09-11:
I think majority of people are using the CC2531 because it's widely available and inexpensive.

That model you linked cost: AU $66.50

CC2531 cost: AU $13.12

Unless you have good reason for using CC2630, I suggest using a CC2531.

aleincsharp, 2018-09-11:
I paied "only" 30€ on local ebay, but yes it is not so cheap like the CC2531.

THanks for yout Time;)

closed 2018-09-11 by aleincsharp

hdo, 2018-09-12:
> That model you linked cost: AU $66.50
> 
> CC2531 cost: AU $13.12
> 
> Unless you have good reason for using CC2630, I suggest using a CC2531.

I think the main reason switching to the CC2630 is that it has 20kB memory instead of 8kb. With that it should be possible to have more than 15 end devices paired to it. 

I would like to see an implementation for that module.

ben423423n32j14e, 2018-09-12:
hdo, the answer to the 15 device limit issue is Z-Stack 3.0 running on the CC2531

See: https://github.com/Koenkk/zigbee2mqtt/issues/211

hdo, 2018-09-12:
> hdo, the answer to the 15 device limit issue is Z-Stack 3.0 running on the CC2531
> 
> See: #211

Yeah, i'm aware of that.

However i still prefer ARM Cortex-M3 over 8051 ;-)

Currently i'm running over 18 Devices in my network and i'm afraid to add more devices to it.

papperone, 2021-04-28:
anyone tinkering with CC2630 modules? got some but struggling to find proper firmware to load... :(