Koenkk/zigbee2mqtt #346: Device with modelID '3210-L' is not supported.

pleasanthill17, 2018-09-08:
I am running hassio and with zigbee2mqtt addon. I added this zigbee switch 3210-L. It says device 3210L is not supported.

I have read wiki "how to support new devices." However, I am unable to find devices.js file on hassio.

Can someone please help me add this device to hassio install with zigbee2mqtt addon.

Koenkk, 2018-09-10:
@danielwelch what do you think is the best way to edit this file? (using hassio).

kimonm, 2018-09-10:
Just FYI, I added support last weekend for Iris plug 3210-L to Koenkk/zigbee-shepherd-converters.

danielwelch, 2018-09-11:
As we’ve discussed in other threads, I think the best way is to attach to the container from the command line and edit the file in its place. But I don’t think this is a great solution, and still think it’d be nice to refactor along the lines of what we’ve talked about here https://github.com/Koenkk/zigbee2mqtt/issues/232

Of course, if support for this device has already been added, best thing to do would be to just use the edge version of the add-on.

Koenkk, 2018-09-11:
@danielwelch thanks for you input.

`3210-L` is now supported in the dev branch.

closed 2018-09-16 by pleasanthill17