Koenkk/zigbee2mqtt #347: Home Assistant binary_sensor does not support json_attributes

yakox, 2018-09-08:
I am using zigbee2mqtt to integrate xiaomi door sensors in Home Assistant. These are set up as mqtt binary sensors in HA and so far have been working perfectly. As I have automation that monitors battery status I noticed that the door sensors were not showing up there.

The door sensors are set up as mqtt binary sensors in HA. The documentation page "Integrating with Home Assistant" shows all binary_sensor configurations with json_attributes. Unlike the mqtt sensor platform, it looks like the [mqtt binary sensor platform](https://www.home-assistant.io/components/binary_sensor.mqtt/) does not support this attribute. 

Probably the docs ought to be updated.

dreimer1986, 2018-09-10:
Well, in this case there is a button above right to edit the site on git and push the change. Did that a while ago somewhere else and it works nicely. Direct Link: https://github.com/home-assistant/home-assistant.github.io/tree/current/source/_components/binary_sensor.mqtt.markdown

Koenkk, 2018-09-11:
I know that the MQTT binary sensor doesn't support `json_attributes`. The reason that they have been added is because I hope that homeassistant does support it in the future. 

@ciotlosm do you know more about this?

ciotlosm, 2018-09-27:
@Koenkk that is correct. I think the best approach would be to have buttons represented as something else or add for support for events in binary_sensors in home assistant.

Koenkk, 2018-11-26:
Fixed in https://github.com/Koenkk/zigbee2mqtt/issues/605

closed 2018-11-26 by Koenkk