Koenkk/zigbee2mqtt #348: The big "Bitron" Thread

tb-killa, 2018-09-08:
I´v got some different types of Bitron Devices (Telecom Branded).
Some of them are functionality (will provide zigbee-converter).

Actually i stuck on this crazy enddevice:
``Bitron AV2010/34`` with 4 Touch Points:

``database.db`` entry
``
{"id":12,"type":"EndDevice","ieeeAddr":"0x00124b000959a3ea","nwkAddr":14354,"manufId":0,"epList":[1,4],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":4,"inClusterList":[0,1,3],"outClusterList":[3,4,5],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":6,"hwVersion":0,"manufacturerName":"Bitron Video","modelId":"AV2010/34","dateCode":"20160815        ","powerSource":3,"locationDesc":"                ","physicalEnv":0,"deviceEnabled":1}},"genPowerCfg":{"dir":{"value":1},"attrs":{"batteryAlarmMask":0}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}}}},"4":{"profId":260,"epId":4,"devId":4,"inClusterList":[],"outClusterList":[25],"clusters":{"genOta":{"dir":{"value":2},"attrs":{"70":4294967295,"upgradeServerId":"0xffffffffffffffff","fileOffset":4294967295,"currentFileVersion":6,"currentZigbeeStackVersion":2,"downloadedFileVersion":4294967295,"downloadedZigbeeStackVersion":65535,"imageUpgradeStatus":0,"manufacturerId":0,"imageTypeId":0,"minimumBlockReqDelay":0,"imageStamp":0}}}}},"_id":"0UolrGDXZtyeTFJD"}
``

I got the typical device not supported message and include rudimental starter ``devices.js`` entry.
After that .. no new messages :(

As show by then database.db entry i doesn´t see the typical cluster .. 
maybe someone could give me a hint how we could work with this ?

dzungpv, 2018-09-10:
It may run Zigbee 3.0/Zigbee Pro. You can check my same problem here: https://github.com/Koenkk/zigbee-shepherd-converters/issues/7 You can also use a sniffer (cc2531) and a software like Wireshark/Ubiqua (trial) to capture the data when it connect/control by original hub.

kaelsaes, 2018-10-04:
Bitron smoke detector

Okt 04 20:37:44 Raspberry-MQTT npm[351]:   zigbee2mqtt:info 2018-10-4 20:37:44 Device incoming...
Okt 04 20:37:44 Raspberry-MQTT npm[351]:   zigbee2mqtt:info 2018-10-4 20:37:44 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
Okt 04 20:37:44 Raspberry-MQTT npm[351]:   zigbee2mqtt:info 2018-10-4 20:37:44 New device with address 0x00124b000904f83b connected!
Okt 04 20:37:44 Raspberry-MQTT npm[351]:   zigbee2mqtt:info 2018-10-4 20:37:44 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00124b000904f83b"}'
Okt 04 20:37:44 Raspberry-MQTT npm[351]:   zigbee2mqtt:warn 2018-10-4 20:37:44 Device with modelID '902010/24A' is not supported.

kaelsaes, 2018-10-04:
I don't understand anything but I hope I can learn now how to create a new converter...😀

kaelsaes, 2018-10-10:
@tb-killa 

Do you need further information?

tb-killa, 2018-10-11:
Will report if we could do more tests .. i´m actually very busy.

tb-killa, 2018-10-28:
Hi @Chrischi- 
yes i developed for them but actually i´m really busy with another project so work only on this if i got more time ;)

```
{"id":11,"type":"EndDevice","ieeeAddr":"0x000d6f000b6ab141","nwkAddr":49018,"manufId":4209,"manufName":"Bitron Home","powerSource":"Battery","modelId":"902010/32","epList":[1],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,10,32,513,516,2821],"outClusterList":[3,25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{"batteryVoltage":32,"batteryAlarmMask":1,"batteryVoltMinThres":25,"batteryAlarmState":0}},"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":0}},"genTime":{"dir":{"value":1},"attrs":{"time":0,"timeStatus":0,"timeZone":0,"dstStart":0,"dstEnd":0,"dstShift":0,"standardTime":0,"localTime":0,"lastSetTime":4294967295,"validUntilTime":4294967295}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":1},"attrs":{"checkinInterval":57600,"longPollInterval":28,"shortPollInterval":2,"fastPollTimeout":240,"checkinIntervalMin":180,"longPollIntervalMin":20,"fastPollTimeoutMax":480}},"hvacThermostat":{"dir":{"value":1},"attrs":{"localTemp":2340,"outdoorTemp":2340,"ocupancy":1,"absMinHeatSetpointLimit":700,"absMaxHeatSetpointLimit":3000,"absMinCoolSetpointLimit":1600,"absMaxCoolSetpointLimit":3200,"localTemperatureCalibration":0,"occupiedCoolingSetpoint":2600,"occupiedHeatingSetpoint":2250,"unoccupiedCoolingSetpoint":2600,"unoccupiedHeatingSetpoint":2000,"minHeatSetpointLimit":700,"maxHeatSetpointLimit":3000,"minCoolSetpointLimit":1600,"maxCoolSetpointLimit":3200,"ctrlSeqeOfOper":2,"systemMode":4,"startOfWeek":0,"numberOfWeeklyTrans":70,"numberOfDailyTrans":10,"programingOperMode":0,"runningState":0,"setpointChangeSource":0,"setpointChangeAmount":-250,"setpointChangeSourceTimeStamp":0}},"hvacUserInterfaceCfg":{"dir":{"value":1},"attrs":{"tempDisplayMode":0,"keypadLockout":0,"programmingVisibility":0}},"haDiagnostic":{"dir":{"value":1},"attrs":{"numberOfResets":4,"macTxUcastRetry":93,"macTxUcastFail":8,"aPSRxBcast":2,"aPSTxBcast":2,"aPSRxUcast":105,"aPSTxUcastSuccess":119,"aPSTxUcastRetry":1,"aPSTxUcastFail":0,"routeDiscInitiated":0,"neighborAdded":1,"neighborRemoved":0,"neighborStale":0,"joinIndication":0,"childMoved":0,"nwkFcFailure":0,"apsFcFailure":0,"apsUnauthorizedKey":0,"nwkDecryptFailures":0,"apsDecryptFailures":0,"packetBufferAllocateFailures":0,"relayedUcast":0,"phyToMacQueueLimitReached":0,"packetValidateDropCount":1}}}}},"_id":"GYWYdQgJJwhRqBUt"}
```
If i got more time i will continue.

tb-killa, 2018-10-28:
I would give you the Hint... This is a hobby project, please be aware that it could give something that doesnt will work. Dont buy only because i wrote something!

Chrischi-, 2018-10-28:
don't worry, but thanks for the advice. 
Anything else you came across while implementing?

Chrischi-, 2018-10-30:
@tb-killa 

got my Bitron Wall Thermostat today

this is all i get:

```
{
"id":23,
"type":"EndDevice",
"ieeeAddr":"0x000d6f000bf8a799",
"nwkAddr":14057,
"manufId":4209,
"manufName":"Bitron Home",
"powerSource":"Battery",
"modelId":"902010/32",
"epList":[1],
"status":"offline",
"joinTime
":null,
"endpoints":{
	"1":{
	"profId":260,
	"epId":1,
	"devId":769,
	"inClusterList":[0,1,3,10,32,513,516,2821],
	"outClusterList":[3,25],
	"clusters":{
		"genBasic":{"dir":{"value":1},"attrs":{}},
		"genPowerCfg":{"dir":{"value":1},"attrs":{"batteryVoltage":32,"batteryAlarmMask":1,"batteryVoltMinThres":25,"batteryAlarmState":0}},
		"genIdentify":{"dir":{"value":3},"attrs":{"identifyTime":129}},
		"genTime":{"dir":{"value":1},"attrs":{"time":0,"timeStatus":0,"timeZone":0,"dstStart":0,"dstEnd":0,"dstShift":0,"standardTime":0,"localTime":0,"lastSetTime":4294967295,"validUntilTime":4294967295}},
		"genOta":{"dir":{"value":2},"attrs":{}},
		"genPollCtrl":{"dir":{"value":1},"attrs":{"checkinInterval":57600,"longPollInterval":28,"shortPollInterval":2,"fastPollTimeout":240,"checkinIntervalMin":180,"longPollIntervalMin":20,"fastPollTimeoutMax":480}},
		"hvacThermostat":{"dir":{"value":1},"attrs":{}},
		"hvacUserInterfaceCfg":{"dir":{"value":1},"attrs":{}},
		"haDiagnostic":{"dir":{"value":1},"attrs":{}}
		}
	}
},
"_id":"bjXhlroKeTfYLiSo"
}

that's a lot less than you...
```

sunflowerABB, 2018-11-13:
@tb-killa: I have the same problem as @Chrischi- above - If you have some time I would appreciate if you could have a look at this as I believe I have a similar problem when trying to integrate the Bitron thermostat. This is my output from the console of zigbee2mqtt:

  zigbee2mqtt:warn 2018-10-4 11:40:23 Device with modelID '902010/32' is not supported.
    {                                                        
        zigbeeModel: ['902010/32'],                           
        model: 'AV2010/32',                                                             
        vendor: 'Bitronvideo',                                                                  
        description: 'Bitronvideo Thermostat',              
        supports: 'temperature',             
        fromZigbee: [],                    
        toZigbee: [],                                                                                           
    },

Chrischi-, 2018-11-14:
@sunflowerABB 
i got it working but it is not ready to do a PR. I also wanted to wait until the new convertert structure is released.

kaelsaes, 2018-12-04:
Any new information available?

I'm still waiting vor the Implementation of the bitron smoke sensor.

Mouky66, 2019-01-14:
Any updates on this? I try to add a bitron AV2010/24A smoke detector. The device is connectable but not yet usable :(

Chrischi-, 2019-01-14:
I implemented the bitron wall thermostat but don‘t have a smoke detector. Maybe i will get a used one everywhere.

asciidisco, 2019-02-04:
I've successfully integrated the [`AV2010/27A`](http://bvsite.smabit.eu/index.php/produkte/smart-home-produkte/temperatur-luftfeuchtigkeitssensor-mit-display/) temperature & humidity sensor & have access to nearly every Bitron device.

I'd like to submit a PR but I haven't quite found the information on how to deal with the PRs, as they're across projects (docs & info here, converters in the shephard-converter repo)...
So, my questions:

- Should I send separate PRs for each device in both projects, or should i bulk add a bunch of them at once?
- Should I first wait for the merge in shephard-converters & then add a PR here for all the docs stuff, or should I add them simultaneously & reference them?

Chrischi-, 2019-02-04:
@asciidisco 
Awesome! Maybe and i think you have access to further documentation of the Bitron devices. ;) 
It would be nice if you could also check for the missing parts of my Bitron wall thermostats implementation and for the implementation in general. Maybe I overlooked something.

asciidisco, 2019-02-04:
@Chrischi- You're maybe right about this...
Would love to help, but I'm still struggling a bit with the converter conventions in shephard (but I also have to say I'd only gave it an hour or two & explored it more exploratively than trying to read the source).

I know all (possible) clusters & Zcl attributes of those, but mapping them to shephard conventions/converters is the thing that gives me some headaches.

Also, the problem with Bitron devices is that old ones can have some really buggy firmwares that makes them quite unreliable (and report/expose less clusters than there should be).

asciidisco, 2019-02-04:
@Chrischi- I know that there’s some CDN out there that provides them. I’ll try to dig out some URLs next week.

asciidisco, 2019-02-05:
@Chrischi- My mail address is public on my profile, just shoot me one & then we can take a look at the specs.

sunflowerABB, 2019-02-20:
I finally succeeded in pairing my Bitron Thermostat with the zigbee2mqtt bridge and am reading status messages such as the following "{"running_state":0,"linkquality":84,"occupied_heating_setpoint":19,"local_temperature":27.8}". 
Any clues how to modify the "occupied_heating_setpoint" variable via MQTT publish?

Chrischi-, 2019-02-20:
```
zigbee2mqtt/your_bitron_device/set

{
    "occupied_heating_setpoint": 21.5
}
```

tb-killa, 2019-03-09:
Hi working actually to get the ``Bitron AV2010/34`` with 4 Touch Points integrated.
After searching i found the original vendor: http://www.climax.com.tw/wss-4e-zb-zb.php
It use ``ZigBee Pro HA 1.2`` what would be a great base to get them integrate.
But this seems to make them more complicated: ``Facilitate the use of 4 pre-programmed scenario conditions``
This sounds like its a End-Device who use also the Group for recall Scenes.
Does somebody got Scenes working on Zigbee2Mqtt ??

tb-killa, 2019-03-09:
@Koenkk 
How does we could parse this Informations to get them inside zigbee2mqtt and the converter????
```
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: {"groupid":0,"clusterid":5,"srcaddr":5159,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":10946846,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[17,8,5,4,0,4]}} +1m
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +1m
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":5,"srcaddr":5159,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":10946846,"transseqnumber":0,"len":6,"data":{"0":17,"1":8,"2":5,"3":4,"4":0,"5":4},"zclMsg":{"frameCntl":{"frameType":1,"manufSpec":0,"direction":0,"disDefaultRsp":1},"manufCode":0,"seqNum":8,"cmdId":"recall","payload":{"groupid":4,"sceneid":4}}} +10ms
```

tb-killa, 2019-03-09:
@Koenkk Need help by building simple converter function for this:
```
zigbee2mqtt:warn 2019-3-9 23:23:57 No converter available for 'AV2010/34' with cid 'genScenes', type 'cmdRecall' and data '{"cid":"genScenes","data":{"groupid":1,"sceneid":1}}'
```
per Button this change ``groupid":1,"sceneid":1`` from 1 to 4.

Should return the number value.

Maybe you´v got a simple and easy solution for building smart converter function ??

Koenkk, 2019-03-09:
```js
AV2010_34_click: {
        cid: 'genScenes',
        type: 'cmdRecall',
        convert: (model, msg, publish, options) => {
            return {click: msg.data.data.groupid};
        },
    },
```

tb-killa, 2019-03-10:
Damn thanks. Its so easy and i´m sitting in front of my system on this late time and doesnt got the right structure :) Work perfect. Will provide PR. Thanks a alot.

closed 2019-05-16 by stale[bot]

Hopeman101, 2020-01-18:
I have two Bitron AV2010/21A Magentic Contacts, which are not yet supported. So I followed the steps in "How to support new devices" and figured out its message structure. The MC has two "buttons": the tamper switch and the contact switch. All Messages from the MC are similar to

Received Zigbee message from 'Bitron_MC_2', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":21,"extendedstatus":0}' from endpoint 1 with groupID 0

I tried through an determined, that all states are represented by four two-digit numbers: 
Tamper pressed 16 or 17 (depending on whether the MC is open or closed)
Tamper open 20 or 21
Magnetic contact closed: 16 or 20
Magnetic contact open: 17 or 21

Or from the number's point of view:
16: Tamper = closed (or false), MC = closed
17: Tamper = closed (or false), MC = open
20: Tamper = open (or true), MC = closed
21: Tamper = open (or true), MC = open

I assume (but did not test and verify, as I do not have a low battery of AAAA type), that "extendedstatus" ist the battery alert with 0 = false (Battery OK) and 1 = true (low battery)

And here starts the trouble: I have never programmed JS and have no clue on how to start (and couldn't either learn from the other conversions, as I don't know their initial message structure). But I assume it is a kind of simple converision, if you take it from the number's point of view and use "if". 

Maybe someone can help me out here?

Best regards,
Hajo

Koenkk, 2020-01-18:
Try reusing the fromZigbee of https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L2957

Hopeman101, 2020-01-18:
Wow, this was a quick answer :-)

OK, I tried (and failed) as follows:

Added to converters.js:
**********************************************************************
    AV2010_21A_iaszone: {    // for Bitron AV 2010/21A magnetic contact
        cluster: 'ssIasZone'
        type 'commandStatusChangeNotification'
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            const extendedStatus = msg.data.extendedstatus;
            return {
                contact: !((zoneStatus & 1) > 0),
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (extendedStatus & 1<<3) > 0,
            },
        },
    },

******************************************************************************
Added to devices.js:
    {
        zigbeeModel: ['902010/21A'],
        model: 'AV2010/21A',
        vendor: 'Bitron',
        description: 'Compact Magnetic Contact',
        supports: 'contact, tamper',
        fromZigbee: [fz.AV2010_21A_iaszone],
        toZigbee: [],
    },

Leads to:
Jan 19 00:43:55 openHABianPi npm[8142]: zigbee2mqtt:error 2020-01-19 00:43:55: Failed to call 'DeviceReceive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
Jan 19 00:43:55 openHABianPi npm[8142]:     at mappedDevice.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:106:42)
Jan 19 00:43:55 openHABianPi npm[8142]:     at Array.filter (<anonymous>)
Jan 19 00:43:55 openHABianPi npm[8142]:     at DeviceReceive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/deviceReceive.js:105:52)
Jan 19 00:43:55 openHABianPi npm[8142]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:322:44)
Jan 19 00:43:55 openHABianPi npm[8142]:     at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:226:14)
Jan 19 00:43:55 openHABianPi npm[8142]:     at Zigbee.emit (events.js:198:13)
Jan 19 00:43:55 openHABianPi npm[8142]:     at Controller.herdsman.on (/opt/zigbee2mqtt/lib/zigbee.js:57:52)
Jan 19 00:43:55 openHABianPi npm[8142]:     at Controller.emit (events.js:198:13)
Jan 19 00:43:55 openHABianPi npm[8142]:     at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:446:22)
Jan 19 00:43:55 openHABianPi npm[8142]:     at Generator.next (<anonymous>))

To be fair - I didn't expect this to work yet, as I most certainly need to get the bits and digits straight. But this somehow looks like I generated more than one error? - I would have assumed to receive rubbish output.

Anyways, I will try again tomorrow. I am too tired to think straight right now anyways.

Best regards,
Hajo

Hopeman101, 2020-01-18:
P.S.: Just figured out that I need to restart Zigbee2mqtt for seeing the complete effect. So, currently, it won't start at all as I did too much wrong in the converters.js.... well, tomorrow more.

Koenkk, 2020-01-19:
Be aware to not replace the whole devices.js with that single definition, just append it to the existing list. Als you dont need to add anything to fromzigbee.js as the converter is already there.

omer1141, 2020-01-19:
Hello, I have a Bitron AV210/32 (heating controller) installed for several month already. Initially, I did not have any problems. However, a couple of weeks ago, the device lost contact with my controller/routers, all of a sudden. I have now the strange situation where I only can get a one-way communication going, i.e. I can set the temperature via mosquitto, however I don‘t receive any regular temperature updates anymore from the device. 
I have version 1.9.0 of zigbee2mqtt installed with firmware 20190619 (with source routing).
Anybody else with the same problem?

Hopeman101, 2020-01-19:
> 
> Be aware to not replace the whole devices.js with that single definition, just append it to the existing list. Als you dont need to add anything to fromzigbee.js as the converter is already there.

Aah, now I got it. ADDING it to the devices.js was obvious to me; but with my cunning NON-knowledge of Javascript I assumed I needed to edit fz.ias_contact_alarm_1 and by "try reusing" you meant "take it as a start". However: It seems to work and produces plausible output, great :-)

I'll test it more thoroughly and then read on how to do the pull request :-)

Cheers,
Hajo

Hopeman101, 2020-01-19:
@omer1141: I have the Bitron 2010/32 as well and it still works like a charm; however, I am also still at zigbee2mqtt 1.8

Hopeman101, 2020-01-20:
@omer1141 I upgraded to 1.9 and the bitron is still working and reporting.

Hopeman101, 2020-01-20:
@Koenkk : I can't figure out to open a pull request; If I go to the website and click on "new pull request" it changes to the "compare" site where the "new pull request is grayed out. I can't figure out how to upload the devices.js

Anyways: The only change is a simple adding of the following code. Lacking an empty battery I can't test the battery alarm, but everything else works as expected. 
Device: Bitron AV2010/21A https://bvsite.smabit.eu/index.php/produkte/smart-home-produkte/kompakter-magnetkontakt/

Code:

   {
        zigbeeModel: ['902010/21A'],
        model: 'AV2010/21A',
        vendor: 'Bitron',
        description: 'Compact Magnetic Contact',
        supports: 'contact, tamper',
        fromZigbee: [fz.ias_contact_alarm_1],
        toZigbee: [],
    },

Koenkk, 2020-01-20:
@Hopeman101 I've added the device, will be supported in the latest zigbee2mqtt dev branch in a few hours.

portboy, 2020-01-22:
I try to get the Bitron remot AV2010/23 running.
without success :P
The remote is similar to the AV2010/34 touch control. Just smaller ..

This is my DB entry:
{"id":8,"type":"EndDevice","ieeeAddr":"0x00124b0008b7658c","nwkAddr":43813,"manufId":0,"manufName":"Bitron Home","powerSource":"Battery","modelId":"902010/23","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":6,"inClusterList":[0,1,3],"outClusterList":[6,8],"clusters":{"genBasic":{"attributes":{"modelId":"902010/23","manufacturerName":"Bitron Home","powerSource":3,"zclVersion":1,"appVersion":19,"hwVersion":0,"dateCode":"20150518        "}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018df3311","endpointID":1}]}},"appVersion":19,"hwVersion":0,"dateCode":"20150518        ","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1579730700956}

With this config I get no response in the debug log
zigbeeModel: ['902010/23'],
model: 'AV2010/23', 
vendor: 'Bitron', 
description: '4-Touch Remote',
supports: 'click',
fromZigbee: [], 
toZigbee: [],

I tried to adapt the config of AV2010/34 in this way:
zigbeeModel: ['902010/23'],
model: 'AV2010/23',
vendor: 'Bitron',
description: '4-Touch Remote',
supports: 'click',
[fz.ignore_power_report, fz.AV2010_34_click], 
toZigbee: [],
meta: {configureKey: 1},
configure: async (device, coordinatorEndpoint) => {
const endpoint = device.getEndpoint(1);
await bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);

No success ...
Any hints what I can/mus du to get it work?

cyablo, 2020-01-26:
I've got my hands on an Bitron 902010/24 smoke detector.

Debug gives me something like this:

zigbee2mqtt:debug 2020-01-26 13:54:30: Received Zigbee message from '0x00124b0008a666c7', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":16,"extendedstatus":0}' from endpoint 1 with groupID 0

I would like to help adding support for the device, but I have no Idea what zonestatus 16 means.

Koenkk, 2020-01-26:
@cyablo try using `fz.ias_smoke_alarm_1` in the `fromZigbee` converter then it should work.

cyablo, 2020-01-28:
@Koenkk looking good. I now get smoke, tamper and battery_low. If I can get turning the alarm on/off reotely to work, I would be happy :)

Koenkk, 2020-01-28:
@cyablo good, I'm not sure if this is possible (I haven't seen it yet for other smoke sensors), is this possible via the original hub?

cyablo, 2020-01-28:
@Koenkk I don't own the original hub, but seems to be supported in deconz: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1593

Koenkk, 2020-01-28:
@cyablo try adding `toZigbee: [tz.light_alert]` and send to `zigbee2mqtt/[DEVICE_NAME]/set` `{"alert": "none"}`

cyablo, 2020-02-02:
I tried:

`mosquitto_pub -h localhost -t zigbee2mqtt/0x00124b0008a666c7/set -m '{"alert": "none"}'
`
But it did nothing. zigbee2mqtt output:

`zigbee2mqtt:debug 2020-02-02 12:17:23: Received MQTT message on 'zigbee2mqtt/0x00124b0008a666c7/set' with data '{"alert": "none"}'
zigbee2mqtt:debug 2020-02-02 12:17:23: Publishing 'set' 'alert' to 'rauchmelder_test'
`

marktroll, 2020-02-27:
I've bought a Bitron 4-Touch single click buttons (AV2010/34) and paired it successfully to z2m.
But I can do what I want.. Dont get a message from a click.

So I've tried to reconnect (pushing the button inside for arround 12 secs).

Log gives:

`warn: Device '0x00124b00096e6158' left the network
info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"xxxxx"}}'
info: Device 'Bitron 4-fach' joined
info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"Bitron 4-fach"}}'
info: Starting interview of 'Bitron 4-fach'
info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"Bitron 4-fach"}}'
info: Successfully interviewed 'Bitron 4-fach', device has successfully been paired
info: Device 'Bitron 4-fach' is supported, identified as: Bitron 4-Touch single click buttons (AV2010/34)
info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"Bitron 4-fach","model":"AV2010/34","vendor":"Bitron","description":"4-Touch single click buttons","supported":true}}'
info: Configuring 'Bitron 4-fach'
info: Successfully configured 'Bitron 4-fach'
`

So everything looks good for me. But still the same... Pushing a button don't trigger a message.
Anybody an idea/help??

lociii, 2020-03-01:
> @omer1141 I upgraded to 1.9 and the bitron is still working and reporting.

@Hopeman101 
According to https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L759 the device is not set up to report anything than the default linkquality to Home Assistant.
Did you change sth on that? Are you using Home Assistant at all?

lociii, 2020-03-03:
> > @omer1141 I upgraded to 1.9 and the bitron is still working and reporting.
> 
> @Hopeman101
> According to https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L759 the device is not set up to report anything than the default linkquality to Home Assistant.
> Did you change sth on that? Are you using Home Assistant at all?
> 
> New issue opened about HA integration #3038

Home Assistent Support added by @Koenkk in latest Dev. 👍

Hopeman101, 2020-03-03:
@lociii
Sorry for not replying earlier.

> > @Hopeman101
> > According to https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L759 the device is not set up to report anything than the default linkquality to Home Assistant.
> > Did you change sth on that? Are you using Home Assistant at all?

I am not using homeassistant and therefore also deactivated it

> > New issue opened about HA integration #3038
> 
> Home Assistent Support added by @Koenkk in latest Dev. 👍

Good to hear :-)

Cheers,
Hajo

cyablo, 2020-03-19:
Also tried to send selct and lselect to bitron smoke detector, without success:

```
Mär 19 10:41:16 cybstretch1 npm[26021]: zigbee2mqtt:debug 2020-03-19 10:41:16: Received MQTT message on 'zigbee2mqtt/0x00124b0008a666c7/set' with data '{"alert": "lselect"}'
Mär 19 10:41:16 cybstretch1 npm[26021]: zigbee2mqtt:debug 2020-03-19 10:41:16: Publishing 'set' 'alert' to 'rauchmelder_test'
Mär 19 10:41:24 cybstretch1 npm[26021]: zigbee2mqtt:debug 2020-03-19 10:41:24: Received Zigbee message from 'zigbee_router', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 8 with groupID 0
Mär 19 10:41:24 cybstretch1 npm[26021]: zigbee2mqtt:info  2020-03-19 10:41:24: MQTT publish: topic 'zigbee2mqtt/zigbee_router', payload '{"state":true,"led_state":false,"linkquality":23}'
Mär 19 10:41:25 cybstretch1 npm[26021]: zigbee2mqtt:debug 2020-03-19 10:41:25: Received MQTT message on 'zigbee2mqtt/0x00124b0008a666c7/set' with data '{"alert": "select"}'
Mär 19 10:41:25 cybstretch1 npm[26021]: zigbee2mqtt:debug 2020-03-19 10:41:25: Publishing 'set' 'alert' to 'rauchmelder_test'
```