Koenkk/zigbee2mqtt #349: Support Aqara S2 door lock

dzungpv, 2018-09-09:
I am adding support for Aqara S2 doorlock https://www.gearbest.com/access-control/pp_009777936300.html.
But it show error when the lock send report (password set):
```
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:301
        dataTypeStr = zclId.dataType(dataType).key;
                                              ^

TypeError: Cannot read property 'key' of undefined
```

Detail logs here: https://pastebin.com/G190FcjG

The detail packet when the bug occurs:
https://pastebin.com/TXdDyvPk

dzungpv, 2018-09-10:
@kirovilya Can you check this?

kirovilya, 2018-09-10:
@dzungpv Well, I'll look in the evening

dzungpv, 2018-09-10:
@kirovilya I change like below and it not crash anymore:
https://github.com/Koenkk/zcl-packet/blob/master/lib/foundation.js#L627
```
        var attrId = this.vars['attrId'];
        // special xiaomi struct-string
        if (attrId === 65281) {
            ru['xiaoMiStruct'](name)(this);
        } else {
            parsedBufLen += this.vars.len;
            this.string(name, this.vars.len);
}
```
to
```
        parsedBufLen += this.vars.len;
        this.string(name, this.vars.len);
        delete this.vars.len;
```
But i don't know you add those for what purpose?

kirovilya, 2018-09-10:
@dzungpv this means that now the xiaomi structure (heartbeat) is not parsed.
But we can now see other messages in the log :)

dzungpv, 2018-09-10:
@kirovilya I still receive the message after that change with the **msg.data.data['65281']**
But only test for this device (Aqara Lock S2).
I see your pull https://github.com/zigbeer/zcl-packet/pull/10/files?diff=split.
When debug **without change your code** by add **console.log(dataType, name)** after this line:
https://github.com/Koenkk/zcl-packet/blob/master/lib/foundation.js#L697
It show
```
strPreLenUint8 [ 'attrData' ]
```
But after that code it crash.

kirovilya, 2018-09-10:
@dzungpv add 
`console.log(dataType);`
before /opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:301 
and show logs to understand on what value fails.

kirovilya, 2018-09-10:
I thinking it will in '6' value...

dzungpv, 2018-09-10:
@kirovilya Yes, more than that:
```
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2ms
66
76
1
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:302
        dataTypeStr = zclId.dataType(dataType).key;
                                              ^

TypeError: Cannot read property 'key' of undefined
    at ensureDataTypeString (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:302:47)
    at getDataType (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:312:16)
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:698:28)
    at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)

```

dzungpv, 2018-09-10:
76 is **4c** 66 is **42** and 01, from the buffer 
<Buffer 1c 5f 11 99 0a 01 ff 42 10 4d 4c 42 06 01 23 01 00 22 02 06 06 06 06 06 06>

dzungpv, 2018-09-10:
"4d 4c 42 06 01 23 01 00 22 02 06 06 06 06 06 06" is  ASCII  code, without the xiaoMiStruct code it output correct data ""MLB\u0006\u0001#\u0002\u0000\"\u0002\u0006\u0006\u0006\u0006\u0006\u0006""

kirovilya, 2018-09-10:
buffer: `<1c 5f 11 0b 0a 01 ff 42 10 4d 4c 42 06 01 23 01 00 22 01 06 06 06 06 06 06>`
xiaomi struct part: `01 ff 42 10 4d 4c 42 06 01 23 01 00 22 01 06 06 06 06 06 06`
`01 ff` - attrId
`42` - (66) type 'charStr'
`10` - length 16 bytes
`4d` - index '77'
`4c` - (76) type 'struct' https://github.com/zigbeer/zcl-id/blob/master/definitions/common.json#L244 I've never seen a structure inside a xiaomi struct...
`42 06 01 23 01 00 22 02 06 06 06 06 06 06`
here I do not know how to parse :(

dzungpv, 2018-09-10:
@kirovilya This device they not write any Zigbee standard in the label so maybe they use private data format/structure. It is smart lock but it not follow standard for **closuresDoorLock**

kirovilya, 2018-09-10:
with the rest of the devices xiaomi it was clear what was written in the structure, and here something new.

dzungpv, 2018-09-10:
I only want to make it not crash with zigbee2mqtt, i have found the way they encode the data ML is Mijia Lock, B is code for the report
u0002 is user id in the lock u0001# is normal use password. 06 is blank array for add user and etc by capture the packer with Xiaomi router

dzungpv, 2018-09-10:
All other device they mark it is Zigbee HA 1.2 standard, but not this lock

kirovilya, 2018-09-10:
I soon expect my Smart door lock https://ru.aliexpress.com/item/Original-xiaomi-mijia-aqara-Smart-Lock-Door-Home-Security-Practical-Anti-theft-Door-Lock-Core-with/32888176191.html , maybe it will be with a similar control. Until I have not a quick solution.

dzungpv, 2018-09-10:
@kirovilya This lock. I can now get the status: open, close, wrong password, the user id open the lock, open buy password or finger print, the time enter wrong pass/finger .... So all the function they provide in the Mi Home app. But this lock not provide remote unlock function or they hide it.
It now can use like a sensor, report status to zigbe2mqtt, i will use it in hass.

dzungpv, 2018-09-10:
@kirovilya do you know what is in the start of the buffer: **1c 5f 11 0b 0a** So i can push a fix for this bug ?

kirovilya, 2018-09-10:
This is header of message https://github.com/zigbeer/zcl-packet/blob/master/lib/zcl.js#L69
1c - frameCntl
5f 11 - manufCode
0b - seqNum
0a - cmdId (10 = report)

dzungpv, 2018-09-10:
@kirovilya I have fixed this and waiting for merge

dzungpv, 2018-09-10:
The "heartbeat" signal not occur with door lock for the xiaoMiStruct type

dzungpv, 2018-09-17:
@kirovilya It produce more crash with basic report from Aqara Curtain and Aqua Lock S2:
S2 lock: https://pastebin.com/D8J2N0KN
Curtain: https://pastebin.com/6rbsthgQ

```
  serialport:bindings read +17s
  serialport:unixRead Starting read +17s
  serialport:unixRead Finished read 78 bytes +1ms
  serialport:main binding.read finished +17s
  cc-znp { sof: 254,
  cc-znp   len: 73,
  cc-znp   type: 'AREQ',
  cc-znp   subsys: 'AF',
  cc-znp   cmd: 'incomingMsg',
  cc-znp   payload:
  cc-znp    { groupid: 0,
  cc-znp      clusterid: 0,
  cc-znp      srcaddr: 29590,
  cc-znp      srcendpoint: 1,
  cc-znp      dstendpoint: 1,
  cc-znp      wasbroadcast: 0,
  cc-znp      linkquality: 52,
  cc-znp      securityuse: 0,
  cc-znp      timestamp: 5750579,
  cc-znp      transseqnumber: 0,
  cc-znp      len: 53,
  cc-znp      data: <Buffer 1c 5f 11 fb 0a 01 ff 42 2c 01 21 29 0d 03 28 23 05 21 0e 00 08 21 09 41 0a 21 4f                                                                                 87 64 20 49 65 20 60 66 20 64 67 20 16 68 23 00 00 00 81 69 20 15 9b ... > },
  cc-znp   fcs: 135,
  cc-znp   csum: 135 } +17s
  serialport:main _read reading +8ms
  serialport:bindings read +9ms
  serialport:unixRead Starting read +9ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 0, srcaddr: 29590, srcendpoint: 1, dstendpoint: 1,                                                                                 wasbroadcast: 0, linkquality: 52, securityuse: 0, timestamp: 5750579, transseqnumber: 0, len: 53, data: <Buf                                                                                fer 1c 5f 11 fb 0a 01 ff 42 2c 01 21 29 0d 03 28 23 05 21 0e 00 08 21 09 41 0a 21 4f 87 64 20 49 65 20 60 66                                                                                 20 64 67 20 16 68 23 00 00 00 81 69 20 15 9b ... > } +4ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +2ms
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:301
        dataTypeStr = zclId.dataType(dataType).key;
                                              ^

TypeError: Cannot read property 'key' of undefined
    at ensureDataTypeString (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:301:47)
    at getDataType (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:311:16)
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:710:28)
    at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_str                                                                                eam_transform.js:184:10)
    at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_st                                                                                ream_transform.js:172:12)
    at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:2                                                                                79:12)
    at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writabl                                                                                e.js:266:5)
    at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stre                                                                                am_writable.js:211:11)
    at Dissolve.Writable.end (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream                                                                                _writable.js:426:10)
```

Do you have any idea?

kirovilya, 2018-09-17:
< 1c 5f 11 fb 0a 
01 ff 
42 - charStr
2c - 44
01 - index 1
21 - 33 uint16
29 0d - value
03 - index 3
28 - 40 int8
23 - value
05 - index 5
21 - uint16
0e 00 - value
08 - index 8
21 - uint16
09 41 
0a - index 10
21 - uint16
4f 87 - value
64 - index 100
20 - uint8
49 - value
65 - index 101
20 - uint8
60 - value
66 - index 102
20 - uint 8
64 - value
67 - index 103
20 - uint 8
16 - value
68 - index 104
23 - uint32
00 00 00 81 - value
69 - index 105
20 - uint 8
15 - value
9b - index 155
... >
all data are normal. need more data! :) add console.log for `dataTypeStr = zclId.dataType(dataType).key` to know what is dataType.

Globally, we need to remove this part (xiaomistruct) of message parsing and leave the data buffer "as is". This will eliminate such falls. Like this commit https://github.com/open-zigbee/zigbee-bridge-packet/commit/dc2e8711a336020c18da0545144e666de7903718

But already outside the zcl-packet it is necessary to parse this buffer depending on the device (for S2, one option, and for the others - according to the old one).

dzungpv, 2018-09-18:
@kirovilya It is "charStr" so you decode method not perfect, we need a new decode method.

kirovilya, 2018-09-18:
@dzungpv What method are you talking about? About the current or about that which is under the link above?

dzungpv, 2018-09-18:
@kirovilya About the current 'xiaoMiStruct' decode function. I can temporary fixed it by check the 'len' ò the buffer, but it is not optimal solution

dzungpv, 2018-09-18:
And bufToAscii not best for decode, there are many data not readable after decode. Better use function buf2Str(buf) to get hexa string

closed 2018-12-03 by stale[bot]