Koenkk/zigbee2mqtt #350: Reconnect device

kaelsaes, 2018-09-09:
Hi,

sometimes a Xiaomi device (temp sensor) doesn't send values anymore although it worked for many days without a problem.

How can i reconnect this device?
Do i have to bind it again? With coordinator or router?

Thanks.

57k, 2018-09-09:
Hi all,
I have the same problem. Would be interested in any advice how to proceed.
Thanks.

kaelsaes, 2018-09-11:
When I pair the device a second time following message appears:
spinlock: false
device already in network

ciotlosm, 2018-09-27:
I found this to be usually a signal problem. Usually triggering a device wakeup (like blow hot hair in it) will send new reports, but I had to adjust position sometimes for better signal. If nothing works you can try adding routers to get around this.

basmeerman, 2018-09-30:
I've the same issue; I think. 

The device wasn't sending messages anymore. 

1. Replaced the battery. To no avail. 
2. Then removed it through MQTT option by sending ID to topic to remove device
3. restart.
4. re-pair. Sheppard says: "device allready in network"
5. checked: device is in configuration.yaml and in database.db
6. Nothing seems to help to receive messages from the sensor.

Please advice what to try next (,not looking for reflashing the cc2531 since It would let me re-pair 30+ devices).

Thanks.

lolorc, 2018-10-01:
the pairing process is such an hassle with zigbee2mqtt, It tooks me all afternoon yesterday to try to repair 30ish devices. And I still have some I can't pair for no reason...
There must be something very wrong with the hardware/firmware/software/dunno, cause the pairing works like a charm with another zigbee coordinator (JN5168) even with the device is far from the coordinator.

basmeerman, 2018-10-01:
Update: It seemed the pairing was incomplete and the network address (int) of the device changed on re-pairing (due to that I removed it before).

I edited the database.db file and made the following changes:
- device information updated to fix the incomplete binding. Example: the field model was not set. I copied the values from another sensor on a different line in the file.

- To have messages processed by zigbee2mqtt/zigbee sheppard I changed the nwkaddr field of the sensor (based on id) to the srcaddr that showed up in the Zigbee Sheppard logging (messages received where not recognized, transformed, and dispatched). 

This all took me about 4/5 hours of analyzing logs, code, and debug info. Finally fixed it within 5 minutes argggg. The great part is I gained a lot of insight into the internal workings of it all.

@lolorc question: with regards to your zigbee coordinator (JN5168) does this work with Xiaomi/Aqara sensors (I have about 30 sensors installed and I'm looking to extend this to about 60, but are hesitant with regards to the pairing difficulties)? Why did you switch zigbee coordinator? Thanks.

basmeerman, 2018-10-01:
@lolorc 

_ad pairing issues_) From what I've read and experienced is the pairing a real headache, but once paired devices seem to be running stable. 

_ad z stack 3_) From what I understand this would only work for new sensors that support  z stack 3.0. This support could come either by buying new sensors (With build in support) or somehow having a firmware upgrade on all devices (which I hope for, but deem unlikely). Furthermore from what I've read is that the amount of sensors/devices supported by CC2531 is limited due to memory limits on the cc2531. So unless a new stack would mean either efficiency or 'sourcing' more work from the chip to the software bridge (I haven't read the z stack 3.0 specs) I wouldn't ge my hopes up. But then again I've been mistaken before :-)

Thanks sofar.

blakeblackshear, 2018-10-01:
@basmeerman what did you use to edit the database by hand?

lolorc, 2018-10-01:
on my side I did use vim :-)
I also used a script like this:
```
a=$(date +"%Y%m%d%H%M%S")
cp data/database.db data/database.db-$a
grep -i -v -e 0xieeeAddr1 -e 0xieeeAddr2 -e 0xieeeAddr3 -e 0xieeeAddr4 data/database.db-$a > data/database.db
```
to remove the devices I couldn't pair

blakeblackshear, 2018-10-01:
Thanks. I didn't realize it was plain text.

basmeerman, 2018-10-01:
> @basmeerman I can’t find it now but I’m sure someone knowledgeable said in one issue here that on 3.0 the ‘limit’ is removed as it can be done in software. I thought I would find it in #211 but it’s not in there.
> 
> Sent with [GitHawk](http://githawk.com)

Thanks, If you find it let me know.

And for what you mentioned it's about the only possible explanation (offloading of the chip) for Z stack 3.0 to fullfill the aforementioned expectation on the same hardware. Or the current firmware is heavily inefficient (what I would not expect since a lot of knowledgeable people worked on it).

Hope to read more about it soon :-)

lolorc, 2018-10-01:
https://github.com/HalleyAssist/zigbee-shepherd is also using Z-Stack 3.
their forks (zigbee-shepertd/cc-znp/zcl*) are getting a lot updates.
But I don't know what do they use them with. (the software using their zigbeer forks)

ryanbeaton, 2018-10-01:
Yeah @splitice and his crew are doing some cool things
https://halleyassist.com
HalleyAssist supports safe, independent and healthy living in the home.

lolorc, 2018-10-01:
yep looks like it's not something I could try at home to see if all the pairing issues and weird behaviors go away with Z-Stack 3 :-)

basmeerman, 2018-10-01:
Could anyone please help me work item a quick breakdown to test this with zigbee2mqtt... 
(optimistic_mode=on; maybe I find some time next couple of weeks to work this out in between things)

Few steps to take:
1) flash new firmware to cc2531 (firmware is located on hallyassist git repo? or do I need to build it? and why?)
2) run latest version of stack @ halleyassist/zigbee-shepherd  // demo should at least recognize messages received. Try to bind a bunch of sensors >20 to the newly flashed cc2531 (see if it overcomes the max device boundry - hypotheses is with Zstack3 this would be solved outside the firmware ).
3) workout diff between koenk/zigbee-shepherd and halleyassist/zigbee-shepherd to see if any API breaks, fix these.
4) run and test the new stack.

Am I missing important details? Thanks.

ryanbeaton, 2018-10-02:
@basmeerman that is very close to what @kirovilya did in https://github.com/Koenkk/zigbee2mqtt/issues/211 . Maybe comment in there and work together.

lolorc, 2018-10-02:
hi there,

What I've been getting a lot as well are "Recieved zigbee message of type 'devStatus' with data '"online"' of device 'undefined' (0x)"
```
Oct 02 08:05:30   cc-znp:SREQ --> ZDO:simpleDescReq, { dstaddr: 0, nwkaddrofinterest: 0, endpoint: 6 } +1ms                                                                                                         
Oct 02 08:05:30   cc-znp { sof: 254,                                                           
Oct 02 08:05:30   cc-znp   len: 1,                                                                                                                                                                        
Oct 02 08:05:30   cc-znp   type: 'SRSP',                                                                                                                                                                            
Oct 02 08:05:30   cc-znp   subsys: 'ZDO',                                                                                                                                                                 
Oct 02 08:05:30   cc-znp   cmd: 'simpleDescReq',                                                                                                                                                                    
Oct 02 08:05:30   cc-znp   payload: { status: 0 },                                                                                                                                           
Oct 02 08:05:30   cc-znp   fcs: 96,                                                                                                                                                                                 
Oct 02 08:05:30   cc-znp   csum: 96 } +13ms                                                                                                                                                  
Oct 02 08:05:30   cc-znp:SRSP <-- ZDO:simpleDescReq, { status: 0 } +2ms                                                                                                                                             
Oct 02 08:05:30   cc-znp { sof: 254,                                                                                                    
Oct 02 08:05:30   cc-znp   len: 14,                                                                                                                                                                                 
Oct 02 08:05:30   cc-znp   type: 'AREQ',                                                                                                                                                                
Oct 02 08:05:30   cc-znp   subsys: 'ZDO',                                                                                                                                                                          
Oct 02 08:05:30   cc-znp   cmd: 'simpleDescRsp',                                                        
Oct 02 08:05:30   cc-znp   payload:                                                                                                                                                                                 Oct 02 08:05:30   cc-znp    { srcaddr: 0,                                                                                                                                                                           
Oct 02 08:05:30   cc-znp      status: 0,                                                                 
Oct 02 08:05:30   cc-znp      nwkaddr: 0,                                                                                                                                                               
Oct 02 08:05:30   cc-znp      len: 8,                                                         
Oct 02 08:05:30   cc-znp      endpoint: 6,                                                                                                                                                                       
Oct 02 08:05:30   cc-znp      profileid: 265,                                                       
Oct 02 08:05:30   cc-znp      deviceid: 5,                                                              
Oct 02 08:05:30   cc-znp      deviceversion: 0,                                                                                                                                                                     
Oct 02 08:05:30   cc-znp      numinclusters: 0,                                                                                                                                                                     
Oct 02 08:05:30   cc-znp      inclusterlist: <Buffer >,                           
Oct 02 08:05:30   cc-znp      numoutclusters: 0,                                                        
Oct 02 08:05:30   cc-znp      outclusterlist: <Buffer > },                                                                                                                                                          
Oct 02 08:05:30   cc-znp   fcs: 204,                                                                       
Oct 02 08:05:30   cc-znp   csum: 204 } +7ms                                                                                                                                                                         
[... cut ...]
Oct 02 08:05:30   zigbee2mqtt:info 2018-10-2 08:05:30 Device incoming...                                                              
Oct 02 08:05:30   zigbee2mqtt:info 2018-10-2 08:05:30 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'                                                      
Oct 02 08:05:30   zigbee2mqtt:debug 2018-10-2 08:05:30 Recieved zigbee message of type 'devStatus' with data '"online"' of device 'undefined' (0xcc2531_2)
```

closed 2018-10-17 by kaelsaes

bbrendon, 2018-12-31:
So what was the answer to the original issue? I have a sensor that disconnects and never reconnects unless i fiddle with it.

zen2, 2019-01-02:
> So what was the answer to the original issue? I have a sensor that disconnects and never reconnects unless i fiddle with it.

I don't know if that can help you but I got actually 12 routers and 12 end devices here and I lost connection with one sensor EndDevice after I have switch off (main power) the bulb router in the same room and to have move in the room this sensor EndDevice.
Before I realize the bulb was not powered at all, I've tried to pair again the sensor but with the unknown device result.

Then I realize the bulb was switched off and I switch on it then I retry to pair the sensor again with the same procedure and the sensor pairs perfectly this time.

In my case I suppose the sensor was using the nearby bulb as router to send its data and so coudn't do it anymore since the bulb was switched off at main power level so moving a EndDevice can bring a disconnect if It can reach his affected router anymore. I don't know actually how really cached routes are refreshed

re-pairing procedure:
```
service zigbee2mqtt stop
cp database.db database.db.bak
grep -v '0x00158d000272112a' database.db.bak > database.db
service zigbee2mqtt start
# reset device sensor to apair it
```