Koenkk/zigbee2mqtt #352: Retrieve device informations/data

nibi79, 2018-09-10:
hi guys,

I need a function to get to all device data / information of a specific device? 

Importent data are the attributes which could be send by the device like temperature,humidity ...
(It would be nice when additional informations can be send like like manufacture/vendor too)

For example the WSDCGQ11LM can send data for temperature,humidity, pressure. The response for zigbee2mqtt/bridge/log type 'devices' could be similar to this with all supported attributes:
```
{
   "type":"devices",
   "message":[
      {
         "ieeeAddr":"0x00158d00021347f4",
         "type":"EndDevice",
         "model":"WSDCGQ11LM",
         "friendly_name":"0x00158d00021347f4",
         "vendor":"Xiaomi",
         "attributes":[
            "temperature",
            "humidity",
            "pressure",
            "battery",
            "voltage",
            "linkquality"
         ]
      }
   ]
}
```

Another possibility could be a function to request data  for a specific device by sending a message to zigbee2mqtt/bridge/config/devices with payload=ieeeAddr. 

Is such a enhancement possible?

regards
Nils

Koenkk, 2018-09-10:
Definitely possible!

nibi79, 2018-09-10:
Ok I'm asking because I implemting a binding for openhab and with these information I am able to make it generic. I believe these informations are interesting for any other smarthome solutions and systems too.

Could you implement such an enhancement or is it possibly unwanted?

Koenkk, 2018-09-10:
Yes I can implement this, however I'm very limited in time due to moving houses so don't expect this soon from my side.

nibi79, 2018-09-11:
Thank you. 

What du you think is your prefered implemantation? 
1.) extending zigbee2mqtt/bridge/log type 'devices' with attibutes, vendor etc.
2.) sending a message to zigbee2mqtt/bridge/config/devices with payload=ieeeAddr
3.) none of these suggestions

So I would be able to mockup the payload and implement a generic binding until then.

nibi79, 2018-09-19:
@stas-demydiuk when I'm right you send the following information:
`           ieeeAddr,
            friendlyName,
            type,
            nwkAddr,
            manufId,
            manufName,
            powerSource,
            modelId,
            status,`

what I really need is what functionality is supported and which information is send by the device like battery, temperature,humidity etc.

so I need something like that:
`         "attributes":[
            "temperature",
            "humidity",
            "pressure",
            "battery",
            "voltage",
            "linkquality"
         ]`

closed 2018-12-22 by stale[bot]