Koenkk/zigbee2mqtt #353: Can not restart docker image

leeiberry, 2018-09-10:
System : Synology DS918+ 
Device 

Docker Container was working fine , I only had only one xioma device register and was working fine.

Tried to pair a second device , without any luck .
Decided to restart the container , just incase !

Docker container will not start
l have not idea how to fix this error  , refer below

The docker log is a follows (decending time order):

**2018-09-10 08:12:10 | stderr | npm ERR!     /root/.npm/_logs/2018-09-10T08_12_10_763Z-debug.log
-- | -- | --
2018-09-10 08:12:10 | stderr | npm ERR! A complete log of this run can be found in:
2018-09-10 08:12:10 | stderr |  
2018-09-10 08:12:10 | stderr | **npm ERR! This is probably no**t a problem with npm. There is likely additional logging output above.**
2018-09-10 08:12:10 | stderr | npm ERR! Failed at the zigbee2mqtt@0.1.3 start script.
2018-09-10 08:12:10 | stderr | npm ERR!
2018-09-10 08:12:10 | stderr | npm ERR! Exit status 1
2018-09-10 08:12:10 | stderr | npm ERR! zigbee2mqtt@0.1.3 start: `node index.js`
2018-09-10 08:12:10 | stderr | npm ERR! errno 1
2018-09-10 08:12:10 | stderr | npm ERR! code ELIFECYCLE
2018-09-10 08:12:10 | stderr | at MqttClient._handleConnack (/app/node_modules/mqtt/lib/client.js:894:10)
2018-09-10 08:12:10 | stderr | at MqttClient.emit (events.js:211:7)
2018-09-10 08:12:10 | stderr | at emitOne (events.js:121:20)
2018-09-10 08:12:10 | stderr | at MqttClient.client.on (/app/lib/mqtt.js:38:13)
2018-09-10 08:12:10 | stderr | at mqtt.connect (/app/lib/controller.js:79:89)
2018-09-10 08:12:10 | stderr | at Controller.handleMQTTConnected (/app/lib/controller.js:98:14)
2018-09-10 08:12:10 | stderr | at Controller.sendAllCachedStates (/app/lib/controller.js:102:37)
2018-09-10 08:12:10 | stderr | at Array.forEach (<anonymous>)
2018-09-10 08:12:10 | stderr | at zigbee.getAllClients.forEach (/app/lib/controller.js:104:22)
2018-09-10 08:12:10 | stderr | at Controller.mqttPublishDeviceState (/app/lib/controller.js:573:36)
2018-09-10 08:12:10 | stderr | TypeError: Cannot read property 'retain' of null
2018-09-10 08:12:10 | stderr |  
2018-09-10 08:12:10 | stderr | ^
2018-09-10 08:12:10 | stderr | retain: deviceSettings.retain,
2018-09-10 08:12:10 | stderr | /app/lib/controller.js:573
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 Connected to MQTT server
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 zigbee-shepherd ready
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 Connecting to MQTT server at mqtt://10.0.0.180
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 Zigbee: allowing new devices to join.
2018-09-10 08:12:10 | stdout | [33m  zigbee2mqtt:warn[39m 2018-9-10 08:12:10 Set `permit_join` to `false` once you joined all devices.
2018-09-10 08:12:10 | stdout | [33m  zigbee2mqtt:warn[39m 2018-9-10 08:12:10 Allowing new devices to join.
2018-09-10 08:12:10 | stdout | [33m  zigbee2mqtt:warn[39m 2018-9-10 08:12:10 `permit_join` set to  `true` in configuration.yaml.
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 unknown (0x00158d00015e95e1): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 Currently 1 devices are joined:
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 Coordinator firmware version: '20180507'
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 zigbee-shepherd started
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 Starting zigbee-shepherd
2018-09-10 08:12:10 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:10 Starting zigbee2mqtt version 0.1.3 (commit #c8c8417)
2018-09-10 08:12:09 | stdout | [32m  zigbee2mqtt:info[39m 2018-9-10 08:12:09 Logging to directory: '/app/data/log/2018-09-10.08:12:09'

Regards Leeb

leeiberry, 2018-09-11:
just delete the JSON file and it restarts !!

closed 2018-09-27 by ciotlosm

aleincsharp, 2018-10-02:
> just delete the JSON file and it restarts !!

Hi, could you specify what JSON file you deleted? Have the same problem./

ius, 2018-10-14:
Deleting `/app/data/state.json` fixed it for me.

Not sure why this is closed - seems to be a bug.