Koenkk/zigbee2mqtt #354: Support for Hue Lightstrip First generation (LST001)

akr1983, 2018-09-10:
Hi,

I'm new to this, so sorry if this is the wrong place. I own a Lightstrip first generation and it hasn't worked out of the box. So i checked the "unsupported devices" guide and just copied the config for the lightstrip V2 (LST002) and changed the modelid. It worked like a charm, could you please add the following modelid to the converters?

7299355PH 
And this is the definition within devices.js
`    {
        zigbeeModel: ['LST001'],
        model: '7299355PH',
        vendor: 'Philips',
        description: 'Hue white and color ambiance LightStrip',
        supports: generic.light_onoff_brightness_colortemp_colorxy().supports,
        fromZigbee: generic.light_onoff_brightness_colortemp_colorxy().fromZigbee,
        toZigbee: generic.light_onoff_brightness_colortemp_colorxy().toZigbee,
    },`

Thanks in advance :-)

Best regards,
Arne

Koenkk, 2018-09-11:
Added, thanks!

closed 2018-09-11 by Koenkk

WagnerBeccard, 2018-09-23:
Trying to call the colortemp dont work yet - Feel free to close 

https://pastebin.com/nxc8JMAz
Not working 
```
Recieved mqtt message on topic 'zigbee2mqtt/Vorraum lightstrip MQ/set' with data '{"color_temp": 386, "state": "ON"}'
e publish to '0x0017880100cbb469', lightingColorCtrl - moveToColorTemp - {"colortemp":386,"transtime":0} - null
 publish to '0x0017880100cbb469', genOnOff - on - {} - null
```
but working
```
MQTT publish, topic: 'zigbee2mqtt/Vorraum lightstrip MQ', payload: '{"state":"ON","color":{"x":0.58,"y":0.399},"brightness":254}'
```

Koenkk, 2018-09-24:
Does the strip support color temperature?

Koenkk, 2018-09-25:
That could be, but maybe it sets the x/y, I think the best thing to do is to remove the color temperature support for now.

WagnerBeccard, 2018-09-25:
As long the colour temp payload isn't in the message its working. 
I understand to little from the Sheppard part, but can't the convertor change the colour temp payload  to an xy? Then it would work like a charm

Koenkk, 2018-09-25:
I think that is the responsibility of the home automation software.