Koenkk/zigbee2mqtt #357: Mentioning device limitations

janmannl, 2018-09-11:
I don't know how else to ask my question
How many connected devices does the CC2530 CC2591 sniffer support?

It might be nice to mention this per device in the supported usb zigbee sniffer page

tb-killa, 2018-09-11:
Look at this Information: https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network#network-size

janmannl, 2018-09-11:
Thanks, although I get your point I'd expect it at least somewhere in getting started or the device listing. Out of the box it supports less devices then all alternative gateways (hue, xiaomi) afaik. But next time I'll read up more before ordering. 

Ssomehow I found Oskarn97's repository which seems to have firmware that supports 48 devices, could that be useful? https://github.com/oskarn97/zigbee2mqtt-firmware/tree/master/src/coordinator/CC2531

Koenkk, 2018-09-11:
@janmannl we've tried this firmware earlier, but it doesn't scale due to limited stack memory (which is instead allocated the the device registry).

ciotlosm, 2018-09-27:
@janmannl do you need more information or are you happy to close this issue down?

closed 2018-12-03 by stale[bot]