Koenkk/zigbee2mqtt #359: After pressing reset button on CC2531 USB Stick some parameters disappear from sensor's data

MikeVRT, 2018-09-11:
Coordinator firmware version: '20180507'
Zigbee2mqtt: 0.1.3 (commit #c8c8417)

After some reboots zigbee2mqtt ask me (in status) for press reset (near USB button) on USB stick. In that case I always reboots system, but now decides try press that button (it's Button S2 in cc2531 USB stick guide).
After pressing reset button most binary sensors starts translating only one (base) parameter, for example (2 binary sensors - one ok, second not):

`
Sep 11 19:38:27 hass npm[2612]:   zigbee2mqtt:info 2018-9-11 19:38:27 MQTT publish, topic: 'zigbee2mqtt/hallway_smoke', payload: '{"battery":"100.00","voltage":3185,"linkquality":78}'
Sep 11 19:40:06 hass npm[2612]:   zigbee2mqtt:info 2018-9-11 19:40:06 MQTT publish, topic: 'zigbee2mqtt/home_door', payload: '{"contact":false}'
Sep 11 19:40:16 hass npm[2612]:   zigbee2mqtt:info 2018-9-11 19:40:16 MQTT publish, topic: 'zigbee2mqtt/home_door', payload: '{"contact":true}'
Sep 11 19:40:16 hass npm[2612]:   zigbee2mqtt:info 2018-9-11 19:40:16 MQTT publish, topic: 'zigbee2mqtt/home_door', payload: '{"contact":true}'
`

So "home_door" sensor in sample translating only "contact" value now. Voltage, linkquality - dissapeared.
As result, we have errors in Home Assistant that uses other parameters in my configuration: 
`"Error parsing value: 'dict object' has no attribute 'battery'..."`
Hardware reboots not helps, short press or reconnect on sensors - not helps.

Koenkk, 2018-09-17:
Could you provide your log with debug enabled? https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration

MikeVRT, 2018-09-18:
Unfortunately no - I already clean all data manually and re pair all devices.

ciotlosm, 2018-09-27:
I'll get a test setup and try to reproduce this. 
@Koenkk I'll ping you when ready with some debug data.

closed 2018-12-22 by stale[bot]