Koenkk/zigbee2mqtt #361: Not able to run zigbee2mqtt as a service

frostnatt, 2018-09-11:
Hello,

i'm trying to run zigbee2mqtt as a deamon via systemctl - without success so far.

Here is what i get, when i check it's status:
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-09-11 21:50:33 CEST; 8s ago
  Process: 25016 ExecStart=/usr/bin/npm start (code=exited, status=203/EXEC)
 Main PID: 25016 (code=exited, status=203/EXEC)

Sep 11 21:50:33 raspberrypi systemd[1]: zigbee2mqtt.service: Changed auto-restart -> dead
Sep 11 21:50:33 raspberrypi systemd[1]: zigbee2mqtt.service: Job zigbee2mqtt.service/restart finished, result=done
Sep 11 21:50:33 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Sep 11 21:50:33 raspberrypi systemd[1]: zigbee2mqtt.service: Converting job zigbee2mqtt.service/restart -> zigbee2mqtt.service/start
Sep 11 21:50:33 raspberrypi systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Sep 11 21:50:33 raspberrypi systemd[1]: zigbee2mqtt.service: Changed dead -> failed
Sep 11 21:50:33 raspberrypi systemd[1]: zigbee2mqtt.service: Job zigbee2mqtt.service/start finished, result=failed
Sep 11 21:50:33 raspberrypi systemd[1]: Failed to start zigbee2mqtt.
Sep 11 21:50:33 raspberrypi systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Sep 11 21:50:33 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.

However, npm start ist working without any issues.
Here are the logs:
https://pastebin.com/A0pUaswM

Does anyone has any ideas? 
Looks like the service is being restarted several times before error occurs. But what is the cause for those restarts?

I appreciate every hint or useful information, thanks!

Koenkk, 2018-09-11:
- How does your `/etc/systemd/system/zigbee2mqtt.service`?
- Whats the place of zigbee2mqtt on your system?

frostnatt, 2018-09-11:
Thanks for the fast reply!

/etc/systemd/system/zigbee2mqtt.service content:

[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/usr/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target

What do you mean by "place of zigbee2mqtt"? It is located in /opt/zigbee2mqtt..

Koenkk, 2018-09-11:
What is the output of `cat /etc/os-release`?

frostnatt, 2018-09-11:
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

subzero79, 2018-09-13:
I cannot recall why exactly but it was due to systemd service issues, but when i started i decided to have an exclusive node bin for zigbee2mqtt. If you have python virtualenv you can try

```
virtualenv /tmp/nodeenv
. /tmp/nodeenv/bin/activate
pip install nodeenv
nodeenv -n 8.11.2 /opt/zigbee2mqtt/node_8.11.2
deactivate
. /opt/zigbee2mqtt/node_8.11.2/bin/activate
cd /opt/zigbee2mqtt
git clone https://github.com/Koenkk/zigbee2mqtt
```

Then follow the npm install

Now the service unit for this daemon should be then 

```
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
Environment=PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/opt/zigbee2mqtt/node_8.11.2/bin
ExecStart=/opt/zigbee2mqtt/node_8.11.2/bin/node index.js
WorkingDirectory=/opt/zigbee2mqtt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=homeassistant

[Install]
WantedBy=multi-user.target
```
My Os is debian stretch x86 as a proxmox guest.

ciotlosm, 2018-09-27:
@frostnatt did you fix the service start? Does your setup start without being a service? How do you start it?

greggotcher, 2018-10-02:
Was there a solution to this?  I am having the same issue.

frostnatt, 2018-10-02:
I think the issue was systemd related. Some other services didn't start up as well. I solved the problem in an old-fashioned way - by clean system installation...

greggotcher, 2018-10-03:
Well I wiped the SD card and completely reloaded Raspbian GNU/Linux 9 (stretch) onto the card and followed all the steps again to get it setup.  When I run npm start everything works great.  However when I try to run it using systemd it fails to load and I see the following:

pi@CONTROLLER-01:/opt/zigbee2mqtt$ systemctl status zigbee2mqtt.service
â— zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-10-03 03:59:46 UTC; 4s ago
  Process: 25660 ExecStart=/usr/bin/npm start (code=exited, status=203/EXEC)
 Main PID: 25660 (code=exited, status=203/EXEC)

Oct 03 03:59:45 CONTROLLER-01 systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Oct 03 03:59:45 CONTROLLER-01 systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Oct 03 03:59:46 CONTROLLER-01 systemd[1]: zigbee2mqtt.service: Service hold-off time over, sche
Oct 03 03:59:46 CONTROLLER-01 systemd[1]: Stopped zigbee2mqtt.
Oct 03 03:59:46 CONTROLLER-01 systemd[1]: zigbee2mqtt.service: Start request repeated too quick
Oct 03 03:59:46 CONTROLLER-01 systemd[1]: Failed to start zigbee2mqtt.
Oct 03 03:59:46 CONTROLLER-01 systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Oct 03 03:59:46 CONTROLLER-01 systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.

Any other ideas?  Different SD card?  Different Pi Zero W?  I am up to try anything.  I am so close to have this running and the last step seems to bee tripping me up.

donatelloth, 2018-10-06:
Just changed 
`/usr/bin/npm start `
to 
`/usr/local/bin/npm start `
and it worked

kennymc-c, 2018-10-10:
Since yesterday when I added 3 Xiaomi Doorsensors with a CC2531 I am also not able to start zigbee2mqtt as a service anymore. Manual start works as expected. Status says its active (running) but with this log:

> Okt 10 19:53:55 raspberrypi3 npm[12094]:     at Array.forEach (<anonymous>)
> Okt 10 19:53:55 raspberrypi3 npm[12094]:     at Object.<anonymous> (/opt/zigbee2mqtt/lib/util/logger.js:64:13)
> Okt 10 19:53:55 raspberrypi3 npm[12094]: npm ERR! code ELIFECYCLE
> Okt 10 19:53:55 raspberrypi3 npm[12094]: npm ERR! errno 1
> Okt 10 19:53:55 raspberrypi3 npm[12094]: npm ERR! zigbee2mqtt@0.1.3 start: `node index.js`
> Okt 10 19:53:55 raspberrypi3 npm[12094]: npm ERR! Exit status 1
> Okt 10 19:53:55 raspberrypi3 npm[12094]: npm ERR!
> Okt 10 19:53:55 raspberrypi3 npm[12094]: npm ERR! Failed at the zigbee2mqtt@0.1.3 start script.
> Okt 10 19:53:55 raspberrypi3 npm[12094]: npm ERR! This is probably not a problem with npm. There is likely additional logging output
> Okt 10 19:53:55 raspberrypi3 npm[12094]: npm ERR! A complete log of this run can be found 

Strangley there is not even a logfile inside /data/log. Only manual starts seem to have one. Also journalctl gives me only the entries from the last manual start.

greggotcher, 2018-10-11:
Thanks @donatelloth 
Changing
/usr/bin/npm start
to
/usr/local/bin/npm start
worked for me as well

kennymc-c, 2018-10-13:
I think i fixed my problem: I noticed that zigbee2mqtt was constantly creating empty log folders and also /data/database.db was owed by root:root and all other files by pi:pi. So I changed the rights to pi:pi. Don't know why this happened or wether database.db should be owned by root by default but now everything is working again.

closed 2018-12-19 by stale[bot]

adamsonr, 2021-02-06:
I had the same issue with openhabian and fixed it by changing

User=pi

to 

User=openhabian

since openhabian was the account I associated with mqtt