Koenkk/zigbee2mqtt #363: Failed to get D-Bus connection: Unbekannter Fehler -1

thomas1977mg, 2018-09-11:
Hello,

I installed the zigbee2mqtt software and can so far use it with FHEM by starting it via "npm start" via WinSCP. If the sesseion is closed the service is also closed.

When I try to use it with autostart - exactly following your instructions I get the below failure message. I can do what I want and nothing changes. 

**sudo systemctl start zigbee2mqtt
Failed to get D-Bus connection: Unbekannter Fehler -1**

Could you or someone else let me have a proposal what I can do now.

Many thanks in advance.

BR,
Thomas

Koenkk, 2018-09-12:
What is your setup (hardware/os)? it seems that systemd is not running: https://superuser.com/questions/1173756/cant-start-systemd-failed-to-get-d-bus-connection

ciotlosm, 2018-09-27:
@thomas1977mg you could also send the process in the background using & at the end of your command. If you'd have container support that would make it even easier to manage.

closed 2018-12-03 by stale[bot]