Koenkk/zigbee2mqtt #365: Support for new TRADFRI control outlet

UnrealKazu, 2018-09-12:
I've been able to get my hands on the new TRADFRI control outlets. I'm trying to compose a new entry we can use in the `devices.js` file for the shepherd, but I'm currently stuck.

When trying to link the device, `zigbee2mqtt` correctly discovers the device with the following entry:
`Device with modelID 'TRADFRI control outlet' is not supported.`

So, in accordance with the wiki I've created a new device entry, with the most basic config in order to determine what messages are sent on the message bus:
```
{
    zigbeeModel: ['TRADFRI control outlet'],
    model: 'E1603',
    vendor: 'IKEA',
    description: 'TRADFRI control outlet',
    supports: 'outlet control',
    fromZigbee: [],
    toZigbee: [],
},
```

But this is where I get stuck. After restarting `zigbee2mqtt` (in debug mode), I don't see any messages regarding missing converters like the wiki mentions. The only message I get intermittently is:
`Check online 0xd0cf5efffeb5706a 0xd0cf5efffeb5706a`, which is the ID of my outlet. After this I've linked the outlet to one of the Ikea buttons, and with this button I can correctly turn on the outlet, and was visible in my TRADFI gateway. So it is working correctly. I also tried it with another new outlet, which I tried to pair whilst my Tradfri gateway was turned off (to prevent it accidentally linking with the gateway), but that one wasn't being picked up either.

It seems like the outlet either does not send any messages, or they are not properly picked up. I've only been using `zigbee2mqtt` for a few weeks now, only with sensors so far, so my knowledge on how to solve this is still lacking.

I tried to skip ahead by just assuming that, because it is a power outlet, the `toZigbee` property should be set to `tz.onoff`. Trying to send a message on the bus results in the following message:
`Zigbee publish to '0xd0cf5efffeb5706a', genOnOff - off - {} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.`. Not sure if this helps, but it looks like the device is no longer paired to the Zigbee stick?

This is quite a lot of information, but I'm trying to be as complete as possible, as this is a new device with functionality that is also new for the TRADFRI line.

Koenkk, 2018-09-12:
Thanks for the info, I'm also very interested in this product, as soon as it is available in the IKEA store I will buy it and help supporting it.

tb-killa, 2018-09-12:
Look at this Information :

https://amp.reddit.com/r/tradfri/comments/9cl0kd/control_outlet_and_the_trådfri_app/

It seems that there will come some sort of update too for get them complete Integrate.

But nice to See this great product.

Would you please share your database.db content.?

tb-killa, 2018-09-18:
@UnrealKazu from what i read it support ``genOnOff``.
Please test this ``devices.js`` part.

```
{
        zigbeeModel: ['TRADFRI control outlet'],
        model: 'E1603',
        description: 'TRADFRI control outlet',
        supports: 'on/off',
        vendor: 'IKEA',
        fromZigbee: [fz.ignore_onoff_change, fz.generic_state],
        toZigbee: [tz.onoff],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
    },
```

it would be great if you would provide ``database.db`` content to because after that we could check if what endpoint id we have to use :)

please provide log too.

UnrealKazu, 2018-09-24:
Apologies for the delayed response, I've been quite busy after I posted the initial issue.

@tb-killa I've tried your config for `devices.js` and am pleased to say that it indeed works! One of my outlets configured itself correctly:

```
zigbee2mqtt:info 2018-9-24 18:42:50 Succesfully configured 0xd0cf5efffeb4572b 0xd0cf5efffeb4572b
zigbee2mqtt:debug 2018-9-24 18:43:11 Recieved mqtt message on topic 'zigbee2mqtt/0xd0cf5efffeb4572b/set' with data '{
zigbee    | "state": "OFF"
zigbee    | }'
zigbee2mqtt:info 2018-9-24 18:43:11 Zigbee publish to '0xd0cf5efffeb4572b', genOnOff - off - {} - null
zigbee2mqtt:info 2018-9-24 18:43:11 MQTT publish, topic: 'zigbee2mqtt/0xd0cf5efffeb4572b', payload: '{"state":"OFF","linkquality":92}'
zigbee2mqtt:debug 2018-9-24 18:43:11 Recieved zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'TRADFRI control outlet' (0xd0cf5efffeb4572b)
zigbee2mqtt:debug 2018-9-24 18:43:11 Recieved zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'TRADFRI control outlet' (0xd0cf5efffeb4572b)
zigbee2mqtt:info 2018-9-24 18:43:11 MQTT publish, topic: 'zigbee2mqtt/0xd0cf5efffeb4572b', payload: '{"state":"OFF","linkquality":99}'
zigbee2mqtt:debug 2018-9-24 18:43:15 Recieved mqtt message on topic 'zigbee2mqtt/0xd0cf5efffeb4572b/set' with data '{
zigbee    | "state": "ON"
zigbee    | }'
zigbee2mqtt:info 2018-9-24 18:43:15 Zigbee publish to '0xd0cf5efffeb4572b', genOnOff - on - {} - null
zigbee2mqtt:info 2018-9-24 18:43:15 MQTT publish, topic: 'zigbee2mqtt/0xd0cf5efffeb4572b', payload: '{"state":"ON","linkquality":99}'
zigbee2mqtt:debug 2018-9-24 18:43:15 Recieved zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI control outlet' (0xd0cf5efffeb4572b)
zigbee2mqtt:info 2018-9-24 18:43:15 MQTT publish, topic: 'zigbee2mqtt/0xd0cf5efffeb4572b', payload: '{"state":"ON","linkquality":102}'
zigbee2mqtt:debug 2018-9-24 18:43:15 Recieved zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI control outlet' (0xd0cf5efffeb4572b)
zigbee2mqtt:debug 2018-9-24 18:43:15 Recieved zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI control outlet' (0xd0cf5efffeb4572b)
zigbee2mqtt:error 2018-9-24 18:43:19 Failed to configure 0xd0cf5efffeb5706a 0xd0cf5efffeb5706a
zigbee2mqtt:debug 2018-9-24 18:43:29 Check online 0xd0cf5efffeb5706a 0xd0cf5efffeb5706a
zigbee2mqtt:debug 2018-9-24 18:43:29 Check online 0xd0cf5efffeb4572b 0xd0cf5efffeb4572b
zigbee2mqtt:debug 2018-9-24 18:43:30 Recieved zigbee message of type 'devInterview' with data '"0xd0cf5efffeb4572b"'
```

As you can see, after publishing a state change message the device correctly registered and the outlet turned itself off.

However, it seems like my other TRADFRI outlet is in some sort of fugue state. The entity name appears often in the debug log (when checking if it is online or not). However, it is not responsive to anything. When I plug in the outlet I get an error message:
`zigbee2mqtt:error 2018-9-24 18:43:19 Failed to configure 0xd0cf5efffeb5706a 0xd0cf5efffeb5706a`

And when trying to switch states I get the following error:
```
zigbee2mqtt:debug 2018-9-24 18:44:29 Check online 0xd0cf5efffeb5706a 0xd0cf5efffeb5706a
zigbee2mqtt:debug 2018-9-24 18:44:29 Check online 0xd0cf5efffeb4572b 0xd0cf5efffeb4572b
zigbee2mqtt:debug 2018-9-24 18:44:54 Recieved mqtt message on topic 'zigbee2mqtt/0xd0cf5efffeb5706a/set' with data '{
zigbee    | "state": "OFF"
zigbee    | }'
zigbee2mqtt:info 2018-9-24 18:44:54 Zigbee publish to '0xd0cf5efffeb5706a', genOnOff - off - {} - null
zigbee2mqtt:error 2018-9-24 18:44:54 Zigbee publish to '0xd0cf5efffeb5706a', genOnOff - off - {} - null failed with error Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.
```

Even removing the device does not work properly, as I get a successful remove response, but later on it still tries to check its online state:
```
zigbee2mqtt:debug 2018-9-24 18:48:34 Recieved mqtt message on topic 'zigbee2mqtt/bridge/config/remove' with data '{
zigbee    | "0xd0cf5efffeb5706a"
zigbee    | }'
zigbee2mqtt:info 2018-9-24 18:48:34 Successfully removed {
zigbee    | "0xd0cf5efffeb5706a"
zigbee    | }
zigbee2mqtt:info 2018-9-24 18:48:34 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_removed","message":"{\n\"0xd0cf5efffeb5706a\"\n}"}'
zigbee2mqtt:debug 2018-9-24 18:49:29 Check online 0xd0cf5efffeb5706a 0xd0cf5efffeb5706a
```
Resetting the outlet does not work (tried holding the reset button for 5 seconds, after which it starts blinking rapidly and then slows down again, indicating a reset).

So of the two TRADFRI outlets I have, one (0xd0cf5efffeb4572b) works perfectly with your config, but the other is not responding to anything (0xd0cf5efffeb5706a). I'm running the latest version as of this time of writing.

Also, I've attached the `database.db` you requested. Hopefully this will give you some insight.
[zigbee_database.zip](https://github.com/Koenkk/zigbee2mqtt/files/2412459/zigbee_database.zip)

tb-killa, 2018-10-02:
Hi @UnrealKazu 
Do you buy this control outlet as a set with the remote ?

I´m interessting in get them also integrate so if you have some time please pair them with your zigbee2mqtt and provide debug output, the database.db entry(s) for this device would be also great to see.

I think this remote control device could support on/off , hold and release informations.

But first lets show what we got :)

UnrealKazu, 2018-10-03:
@tb-killa Unfortunately, when I went to buy the outlets they only had the version in stock that did not include the remote. So I only have the `database.db` I already supplied in https://github.com/Koenkk/zigbee2mqtt/issues/365#issuecomment-424088529, with one outlet paired without the remote.

I'm not sure if they sell the switch remotes separately. According to the IKEA website (at least the Dutch one) they only sell them coupled with the outlet.

tb-killa, 2018-10-03:
@UnrealKazu .. thanks for the answer.
Sadly in germany this type of device isn´t available .. maybe in the netherlands (near of venlo) .. have to check .. it should be official in 2019 ..

Skeletorjus, 2018-10-07:
I bought some of these, and while the first one paired perfectly, I have trouble paring the two others. All of them are reported by the Trådfri-app to be on the latest firmware.
Have any of you sucessfully connected 2 or more of these devices to the network?

SuperTux101, 2018-10-13:
Pairing the device was a little bit confusing. No documentation found on this subject. It is however very simple. Near the light is a pinhole. Put zigbee2mqtt in pairing mode. Stick a small pin in the pinhole. Inside is a microswitch which must be pressed. Keep the switch pressed until the light starts to change its brightness. The logging of zigbee2mqtt shows that a new device is added:
  E1603 - IKEA TRADFRI control outlet (Router)

The following configuration can be used to control the outlet from HA.

switch:
  - platform: mqtt
    name: "IKEA outlet"
    state_topic: "zigbee2mqtt/0xd0cf5efffeef33ac"
    command_topic: "zigbee2mqtt/0xd0cf5efffeef33ac/set"
    payload_on: "ON"
    payload_off: "OFF"
    state_on: "ON"
    state_off: "OFF"
    value_template: "{{ value_json.state }}"

UnrealKazu, 2018-11-02:
I've added the reset instructions to the wiki. Now that support for this device has been released, and people mentioning that it works fine, I will close this issue.

closed 2018-11-02 by UnrealKazu

Koenkk, 2018-11-02:
@UnrealKazu thanks!