Koenkk/zigbee2mqtt #366: Why is IKEA Tradfri LED1623G12 identified as type Router

tobang, 2018-09-12:
When I pair the LED1623G12, it is added with type Router?
{"ieeeAddr":"0x000b57fffec2b95a","type":"Router","model":"LED1623G12","friendly_name":"PhLampe"}

Thank you

Koenkk, 2018-09-12:
This is correct, all/most AC powered devices are routers: https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network#router

closed 2018-09-27 by ciotlosm

alinelena, 2018-10-01:
sorry to comment on a closed issue..I am in the same situation all the ikea lights listed as routers.
is there a way to use them or disable them on will. (the router capability)

Koenkk, 2018-10-01:
No that's not possible. But why would you want that?

dirkclemens, 2018-10-19:
Dear @Koenkk but how can I switch a bulb, when it is recognized as a router? I would expect to send a mqtt message with a message like "on" to switch that bulb on (and off). Did I miss something?